espressif/esp-idf #2041: ledc_fade_func_uninstall breaks deep sleep

OtherCrashOverride, 2018-06-08:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [none]
- Core (if using chip or module): [ESP32-WROVER]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    34816ff0080726678ad547790f8e069a138bc9ca (reverted 0d65f3b7f9a1dace8853719871d920d80aa3604b)
- Development Env:		[other]
- Operating System:		[Ubuntu]
- Power Supply:         [USB + Battery]

## Problem Description

if ```ledc_fade_func_uninstall()``` is called prior to entering deep sleep, the device can not wake.  Commenting out the above call restores normal operation.

### Expected Behavior
Deep sleep should wake regardless of ```ledc_fade_func_uninstall()```.

### Actual Behavior
Device does not wake from deep sleep.

### Steps to repropduce

1. Call ```ledc_fade_func_uninstall()```
2. Call ```esp_sleep_enable_ext0_wakeup(GPIO_NUM_13, 0);
rtc_gpio_pullup_en(GPIO_NUM_13);
rtc_gpio_isolate(GPIO_NUM_12);
esp_deep_sleep_start();```

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
Not available.

## Debug Logs
Not available.

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

xiongyumail, 2018-06-13:
Hi,@OtherCrashOverride
The problem you should encounter is not related to LEDC. It is related to deep_sleep ext0. We are working hard to repair this bug.

igrr, 2018-06-20:
Deep sleep EXT0 wakeup issue has been resolved in https://github.com/espressif/esp-idf/commit/4359d5985eba8ad9220fa7a4b3459197024f6e8c.

xiongyumail, 2018-06-22:
Hi,@OtherCrashOverride 
The issue of the related deep_sleep has been repaired, you can use the latest version of IDF to test.

closed 2018-06-22 by FayeY