espressif/esp-idf #2043: esp_sleep_enable_ext0_wakeup() always wakes when level set to 0, ignoring pin state

MrSurly, 2018-06-08:
## Environment

- Development Kit:      Custom/ bare WROOM-32
- Core: ESP-WROOM32
- IDF version: 1f7b41e2
- Development Env: Make (Micropython)
- Operating System: Ubuntu 18.04
- Power Supply: USB

## Problem Description

Downstream issue: https://github.com/micropython/micropython/issues/3835

The apparent cause of the above issue (GPIO pin always reading 0 after deep sleep) was fixed by calling `gpio_config(&config);`, per a [comment] by @negativekelvin.

Now after deep sleep wakeup, the GPIO pin is read correctly.  

**However**, after calling `esp_sleep_enable_ext0_wakeup()` with `level = 0`, the device always wakes immediately, even with the pin tied directly to the 3.3v supply.  

This _does not occur_ when the pin is tied to ground, with `level = 1`.

### Expected Behavior

Device remains in deep sleep when configuring ext0 to wake on "low", and the pin is tied "high".

### Actual Behavior

Device wakes from deep sleep immediately, with the wake_cause being EXT0

### Steps to reproduce

Build & run code below. 

### Code to reproduce this issue
```c

#include <stdio.h>
#include <unistd.h>
#include "esp_sleep.h"
#include "driver/gpio.h"

void app_main()
{

    printf("Wake cause: %d\n", esp_sleep_get_wakeup_cause());

    gpio_num_t pin = GPIO_NUM_34;

    gpio_config_t config;
    config.pin_bit_mask = 1LL << pin;
    config.mode = GPIO_MODE_DEF_INPUT;
    config.pull_up_en = GPIO_PULLUP_DISABLE;
    config.pull_down_en = GPIO_PULLDOWN_DISABLE;
    config.intr_type = GPIO.pin[pin].int_type; // don't change
    
    gpio_config(&config);
   
    int value = gpio_get_level(pin);
    int level = value ? 0 : 1;
   
    esp_err_t err = esp_sleep_enable_ext0_wakeup(pin, level);
    printf("pin: %d value: %d ext0 wake level: %d error: %d\n", pin, value, level, err);
    sleep(1);
    esp_deep_sleep_start();
}  
```
## Debug Logs

### Bad, with pin 34 tied to 3.3v
```
[... 145 lines cut ...]
I (180) boot: Disabling RNG early entropy source...
I (181) cpu_start: Pro cpu up.
I (184) cpu_start: Starting app cpu, entry point is 0x40080fb8
0x40080fb8: call_start_cpu1 at /home/epoulsen/workspaces/esp-idf/components/esp32/cpu_start.c:224

I (0) cpu_start: App cpu up.
I (195) heap_init: Initializing. RAM available for dynamic allocation:
I (201) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (208) heap_init: At 3FFB2830 len 0002D7D0 (181 KiB): DRAM
I (214) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (220) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (227) heap_init: At 40088798 len 00017868 (94 KiB): IRAM
I (233) cpu_start: Pro cpu start user code
I (251) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Wake cause: 0
I (254) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
pin: 34 value: 0 ext0 wake level: 1 error: 0
< device is asleep >
```

## Other items if possible

* This occurs on other pins
* The SDK config is the default one written by doing a `make menuconfig`, changing nothing, and saving in the `examples/system/deep_sleep` directory.

OtherCrashOverride, 2018-06-09:
Check if issue #1931 is relevant.

igrr, 2018-06-13:
Hi @MrSurly, thank you for the detailed report. We were able to reproduce the issue and have a fix coming up. In the meantime, you can work around the issue by adding the following lines before `esp_deep_sleep_start` call.

```
rtc_gpio_init(pin);
rtc_gpio_set_direction(pin, RTC_GPIO_MODE_INPUT_ONLY);
```

closed 2018-06-15 by igrr

MrSurly, 2018-06-24:
@igrr Thanks!