espressif/esp-idf #2046: Any socket examples for PPPoS?

sarpdaltaban, 2018-06-10:
PPPoS example is clearly working; however, 
     when I try to open socket as a SERVER, clients return incorrect ACK number,
     when I try to act as a CLIENT, I can send data till ppp_output_callback buffer stucks at the size 584,

Do you have any PPPoS socket example that works with/without RTOS?

xiongyumail, 2018-06-11:
Hi,
There's a good project here!<https://github.com/loboris/ESP32-PPPOS-EXAMPLE.git>

sarpdaltaban, 2018-06-11:
Sir, thank you very much, I am working on the example.

From: XiongYu
Sent: Monday, June 11, 2018 11:34 AM
To: espressif/esp-idf
Cc: sarpdaltaban; Author
Subject: Re: [espressif/esp-idf] Any socket examples for PPPoS? (#2046)

Hi,
There's a good project here!https://github.com/loboris/ESP32-PPPOS-EXAMPLE.git

closed 2018-06-13 by FayeY