espressif/esp-idf #2048: About relesaes and submodule

Shaco-Ma, 2018-06-11:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Core (if using chip or module): [ESP32-Wrover-I|ESP32-Wrover|ESP32-PICO-D4|ESP32-SOLO-I|ESP32-WROOM-32U|ESP-WROOM-32D|ESP-WROOM32|ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
       ESP-IDF Release v3.0
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
Hello:
        Why the esp-idf releases of esp-idf didn't include submodule? Like BT, the lib of [https://github.com/espressif/esp32-bt-lib](url), If i want use the code of realeases, but it has no all of submodules.
        I think the releases of github must have all of this should include, And I can't unstand why you releases of esp-idf didn't include submodules.

negativekelvin, 2018-06-11:
> Download an archive with submodules included
> 
> Attached to this release is an esp-idf-v3.0.zip archive. It includes .git directory and all the submodules, so can be used out of the box. This archive is provided for users who have connectivity issues preventing them from cloning from Github.

Shaco-Ma, 2018-06-12:
@negativekelvin Only get the archive from git address? Can't git all submodules in realeases version?

negativekelvin, 2018-06-12:
GitHub doesn't support packing submodules in auto generated archive but espressif does make a separate download for major releases that includes them

https://github.com/espressif/esp-idf/releases/download/v3.0/esp-idf-v3.0.zip

closed 2018-06-14 by Shaco-Ma