espressif/esp-idf #2050: Proper way to deinit Wi-Fi on esp32

KryvosheyaIvan, 2018-06-11:
Hello!
There are many issues here and posts on [esp32.com](https://esp32.com/index.php) forum about problem that  in case of esp32 was disconnected from router with log:
`wifi: bcn_timout,ap_probe_send_start`
it can`t connect again!
Those people, who tried to fully reinit wi-fi in esp32 got esp32 restarted.
So my question is: what is the proper way to deinit wi-f- in esp32?
`            esp_wifi_stop();
            esp_wifi_deinit();
            initialise_wifi()

void initialise_wifi(void)
    {
        tcpip_adapter_init();
        wifi_event_group = xEventGroupCreate();
        ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
        ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
        wifi_config_t wifi_config = {
			.sta = {
                .ssid = "MySSID",
                .password = "MyPassword",
              },
          };
        ESP_LOGI(WiFi_s, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
        ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
        ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
        ESP_ERROR_CHECK( esp_wifi_start() );
    }`
that way makes esp32 to unexpectedly restart as well

liuzfesp, 2018-06-22:
HI ＠KryvosheyaIvan， you can call esp_wifi_stop() first, then call esp_wifi_deinit()

KryvosheyaIvan, 2018-06-26:
@liuzfesp , that doesn`t work!
    void myDeinit(void)
    {
    	esp_wifi_stop();
    	vTaskDelay(200 / portTICK_PERIOD_MS);
    
    	esp_wifi_deinit();
    	vTaskDelay(200 / portTICK_PERIOD_MS);
    
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    
        wifi_config_t wifi_config = {
                .sta = {
                    .ssid = "SenseSystems",
                    .password = "sense2016",
                },
            };
        //ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
        ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
        ESP_ERROR_CHECK( esp_wifi_start() );
    }

KryvosheyaIvan, 2018-06-26:
@liuzfesp before I opened this issue I had tried a lot of ways to deinit WiFi and every time it reboots.

liuzfesp, 2018-06-26:
@KryvosheyaIvan got it, so I will write a demo for you.

liuzfesp, 2018-06-26:
HI @KryvosheyaIvan just replace following code with idf/examples/wifi/simple_wifi/main/simple_wifi.c, the esp_wifi_stop/esp_wifi_deinit work correctly for me, please let me know if can't work for you:
`
/* Simple WiFi Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use simple WiFi configuration that you can set via
   'make menuconfig'.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_MODE_AP   CONFIG_ESP_WIFI_MODE_AP //TRUE:AP FALSE:STA
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_MAX_STA_CONN       CONFIG_MAX_STA_CONN

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int WIFI_CONNECTED_BIT = BIT0;

static const char *TAG = "simple wifi";
static bool g_wifi_reconnect_flag = true;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        xEventGroupSetBits(wifi_event_group, WIFI_CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_AP_STACONNECTED:
        ESP_LOGI(TAG, "station:"MACSTR" join, AID=%d",
                 MAC2STR(event->event_info.sta_connected.mac),
                 event->event_info.sta_connected.aid);
        break;
    case SYSTEM_EVENT_AP_STADISCONNECTED:
        ESP_LOGI(TAG, "station:"MACSTR"leave, AID=%d",
                 MAC2STR(event->event_info.sta_disconnected.mac),
                 event->event_info.sta_disconnected.aid);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        ESP_LOGI(TAG, "disconnected");
        if (g_wifi_reconnect_flag) {
            ESP_LOGI(TAG, "reconnect again");
            esp_wifi_connect();
        }
        xEventGroupClearBits(wifi_event_group, WIFI_CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

void wifi_init_softap()
{
    wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .ssid_len = strlen(EXAMPLE_ESP_WIFI_SSID),
            .password = EXAMPLE_ESP_WIFI_PASS,
            .max_connection = EXAMPLE_MAX_STA_CONN,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };
    if (strlen(EXAMPLE_ESP_WIFI_PASS) == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "wifi_init_softap finished.SSID:%s password:%s",
             EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
}

void wifi_init_sta(void *param)
{
    wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

_again: 
    g_wifi_reconnect_flag = true;
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS
        },
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    vTaskDelay(5000/portTICK_PERIOD_MS);

    g_wifi_reconnect_flag = false;
    ESP_LOGI(TAG, "wifi_init_sta stop sta");
    ESP_ERROR_CHECK(esp_wifi_stop() );
    ESP_LOGI(TAG, "wifi_init_sta deinit sta");
    ESP_ERROR_CHECK(esp_wifi_deinit() );
    goto _again;
    ESP_LOGI(TAG, "wifi_init_sta finished.");
    ESP_LOGI(TAG, "connect to ap SSID:%s password:%s",
             EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
}

void app_main()
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    
#if EXAMPLE_ESP_WIFI_MODE_AP
    ESP_LOGI(TAG, "ESP_WIFI_MODE_AP");
    wifi_init_softap();
#else
    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    xTaskCreate(wifi_init_sta, "testTask", 4096, NULL, 10, NULL);
#endif /*EXAMPLE_ESP_WIFI_MODE_AP*/

}
`

closed 2018-07-17 by FayeY

PriyankPa, 2018-12-17:
```
goto _again;
    ESP_LOGI(TAG, "wifi_init_sta finished.");
    ESP_LOGI(TAG, "connect to ap SSID:%s password:%s",
             EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
```
The part after `goto ` will never execute.

PriyankPa, 2018-12-17:
@KryvosheyaIvan Did you find solution to your problem?

rochamorelli, 2019-01-15:
@KryvosheyaIvan Did you try this?

> 8. Wi-Fi Deinit Phase
        s8.1: Call esp_wifi_disconnect() to disconnect the Wi-Fi connectivity.
        s8.2: Call esp_wifi_stop() to stop the Wi-Fi driver.
        s8.3: Call esp_wifi_deinit() to unload the Wi-Fi driver.

https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-deinit-phase

bill1davis, 2019-08-20:
Version 3.2 shuts down better by waiting for events in lock step
disconnect, then wait for disconnected, then stop and wait for stopped, then deinit.