espressif/esp-idf #2052: cannot wake up using touch pad

infotrino, 2018-06-12:
## Environment

- Development Kit:      ESP32-Wrover : personal board from infotron
- Kit version : v1
- Core (if using chip or module): ESP32-Wrover
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): d6ab395
- Development Env:		Make
- Operating System:		Windows
- Power Supply:         USB

## Problem Description
I cannot wake up from light sleep using touch pad (and also from deep sleep).
If I use esp-idf (20180505 version), there is no problem.
So I tried to replace two files to previous version, and problem was solved!
    (1. esp-idf/components/driver/rtc_module.c, 
     2. esp-idf/components/driver/include/driver/touch_pad.h)

### Steps to repropduce
1. copy deep_sleep folder from ''esp-idf/examples/system/deep_sleep''
2. enable touch pad wake up from menu config
3. make flash monitor
4. esp32 goes to deep sleep.  and you cannot wake up from deep sleep using touch pad.

5. roll back two files to previous version (1. esp-idf/components/driver/rtc_module.c, 
     2. esp-idf/components/driver/include/driver/touch_pad.h)
6. recomplie and make flash monitor
7. You can see that problem is solved, so wake up using touch pad.

### Code to reproduce this issue
almost same as esp-idf/examples/system/deep_sleep

negativekelvin, 2018-06-12:
Try calling` touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);` after  `touch_pad_init();`

infotrino, 2018-06-13:
That works well. 
Thank you ~

negativekelvin, 2018-06-20:
Usage note is ok but how about check mode in esp_sleep_enable_touchpad_wakeup and either set or print error?

closed 2018-10-09 by infotrino