espressif/esp-idf #2054: Console performance problems on latest master

vonnieda, 2018-06-12:
On 7aa7b35a951b67ae1388cc22a4148e88d92c1579, running the system/console example, or my own console code based on the example, there is a performance issue where pressing enter causes a very long delay before the command prompt returns.

Here's a short screen capture video showing the problem:

[esp32-slow-console.mov.zip](https://github.com/espressif/esp-idf/files/2095526/esp32-slow-console.mov.zip)

If you look closely, you can see the cursor move to the right after about 1 second, then to the next line after another second. 

I first noticed this issue on our custom hardware using our custom console code. I then reproduced it using a WROVER-KIT-V3 and the stock system/console example.

vonnieda, 2018-06-12:
A few more thoughts:
* Performance of everything else seems fine. All my other tasks appear to be running at full speed.
* Changing the priority of the console task to highest performance seems to make no difference.

dobairoland, 2018-06-13:
Thank you @vonnieda for reporting this issue. You are right about this problem. The one second delay is artificial which is caused by a new feature of the IDF monitor. I will fix this ASAP.

closed 2018-06-15 by igrr

vonnieda, 2018-06-15:
Hi @dobairoland, I just updated to latest master and while the performance is better, it's still very bad compared to before the filter change was added. I'd ask you to run the console example and try typing some commands, using command completion, arrow keys to go back, etc. There is a noticeable delay between characters now and it makes the console feel very sluggish.

dobairoland, 2018-06-15:
Ok, I'll take a look into this issue again and will improve it.

markwj, 2018-07-30:
This has broken 'make monitor' for us. The simple monitor still works fine.

Unfortunately, our console is quite complex, so hard to reproduce this. Rolling back this change fixes it for us, though.

vonnieda, 2018-07-30:
Agreed @markwj - I don't think this issue should be closed. The performance is still very bad compared to before this change.

Perhaps @dobairoland could take a second look?

For a simple example, try using the console example with history enabled and page back through the history. It feels very sluggish.

This past week I've been switching between pre 3.0, 3.0.2, and 3.1 quite a bit as I work through other problems and this issue is very evident when I switch to >= 3.0.

Thanks,
Jason

dobairoland, 2018-07-30:
Hello @markwj & @vonnieda. Sorry, but I think that the fix was pushed only now. At least that is what the previous line shows me. Please re-test it with https://github.com/espressif/esp-idf/commit/b8937e55224c8bca48ed81d77553d2c8be86cfe8 or check if this commit was already included in your IDF.

vonnieda, 2018-07-30:
This does look to be fixed in the latest code. Thanks!