espressif/esp-idf #2056: Eclipse can't find includes for c++ file

jeremyherbert, 2018-06-13:
## Environment
- IDF version: 7aa7b35a 
- Development Env:		Eclipse
- Operating System:		Xubuntu 18.04

- Eclipse version: Oxygen.3a Release (4.7.3a)

## Problem Description

If I open the "hello_world" example in eclipse, everything works correctly after setting up as per the documentation. If I then rename the `hello_world_main.c` file to `hello_world_main.cpp` and add the following code in:

```extern "C" {
	void app_main(void);
}```

the build will still succeed, but eclipse will no longer be able to resolve the include directories at the top of the file. For example, `#include "freertos/FreeRTOS.h"` has the orange squiggle underline.

### Expected Behavior

Eclipse should resolve the include paths for both C and C++

### Actual Behavior

It only works for C.

## Debug Logs

Eclipse indexer log for C++:
```
Project:               hello_world
File:                  file:/home/jeremy/esp/hello_world/main/hello_world_main.cpp
Language:              GNU C++
Index Version:         211.0
Build Configuration:   Default
Context:               file:/home/jeremy/esp/hello_world/main/hello_world_main.cpp
   C++, {}
Versions in Index:     1
   C++: {}; 0 macros, 5 includes, 13 names;

Include Search Path (option -I):
   /home/jeremy/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0
   /home/jeremy/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/xtensa-esp32-elf
   /home/jeremy/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/backward
   /home/jeremy/esp/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/5.2.0/include
   /home/jeremy/esp/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/5.2.0/include-fixed
   /home/jeremy/esp/xtensa-esp32-elf/xtensa-esp32-elf/include
   /home/jeremy/esp/xtensa-esp32-elf/xtensa-esp32-elf/sysroot/usr/include
```

Eclipse indexer log for C:
```
Project:               hello_world
File:                  file:/home/jeremy/esp/hello_world/main/hello_world_main.c
Language:              GNU C
Index Version:         211.0
Build Configuration:   Default
Context:               file:/home/jeremy/esp/hello_world/main/hello_world_main.c
   C, {}
Versions in Index:     1
   C: {}; 0 macros, 5 includes, 32 names;

Include Search Path (option -I):
   /home/jeremy/esp/esp-idf/components/app_trace/include
   /home/jeremy/esp/esp-idf/components/app_update/include
   /home/jeremy/esp/esp-idf/components/bootloader_support/include
   /home/jeremy/esp/esp-idf/components/bt/include
   /home/jeremy/esp/esp-idf/components/coap/port/include
   /home/jeremy/esp/esp-idf/components/coap/port/include/coap
   /home/jeremy/esp/esp-idf/components/coap/libcoap/include
   /home/jeremy/esp/esp-idf/components/coap/libcoap/include/coap
   /home/jeremy/esp/esp-idf/components/console
   /home/jeremy/esp/esp-idf/components/driver/include
   /home/jeremy/esp/esp-idf/components/esp-tls
   /home/jeremy/esp/esp-idf/components/esp32/include
   /home/jeremy/esp/esp-idf/components/esp_adc_cal/include
   /home/jeremy/esp/esp-idf/components/esp_http_client/include
   /home/jeremy/esp/esp-idf/components/ethernet/include
   /home/jeremy/esp/esp-idf/components/expat/port/include
   /home/jeremy/esp/esp-idf/components/expat/include/expat
   /home/jeremy/esp/esp-idf/components/fatfs/src
   /home/jeremy/esp/esp-idf/components/freertos/include
   /home/jeremy/esp/esp-idf/components/heap/include
   /home/jeremy/esp/esp-idf/components/idf_test/include
   /home/jeremy/esp/esp-idf/components/jsmn/include
   /home/jeremy/esp/esp-idf/components/json/cJSON
   /home/jeremy/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include
   /home/jeremy/esp/esp-idf/components/libsodium/port_include
   /home/jeremy/esp/esp-idf/components/log/include
   /home/jeremy/esp/esp-idf/components/lwip/include/lwip
   /home/jeremy/esp/esp-idf/components/lwip/include/lwip/port
   /home/jeremy/esp/esp-idf/components/lwip/include/lwip/posix
   /home/jeremy/esp/esp-idf/components/lwip/apps/ping
   /home/jeremy/esp/esp-idf/components/mbedtls/port/include
   /home/jeremy/esp/esp-idf/components/mbedtls/mbedtls/include
   /home/jeremy/esp/esp-idf/components/mdns/include
   /home/jeremy/esp/esp-idf/components/micro-ecc/micro-ecc
   /home/jeremy/esp/esp-idf/components/newlib/platform_include
   /home/jeremy/esp/esp-idf/components/newlib/include
   /home/jeremy/esp/esp-idf/components/nghttp/port/include
   /home/jeremy/esp/esp-idf/components/nghttp/nghttp2/lib/includes
   /home/jeremy/esp/esp-idf/components/nvs_flash/include
   /home/jeremy/esp/esp-idf/components/openssl/include
   /home/jeremy/esp/esp-idf/components/pthread/include
   /home/jeremy/esp/esp-idf/components/sdmmc/include
   /home/jeremy/esp/esp-idf/components/smartconfig_ack/include
   /home/jeremy/esp/esp-idf/components/soc/esp32/include
   /home/jeremy/esp/esp-idf/components/soc/include
   /home/jeremy/esp/esp-idf/components/spi_flash/include
   /home/jeremy/esp/esp-idf/components/spiffs/include
   /home/jeremy/esp/esp-idf/components/tcpip_adapter/include
   /home/jeremy/esp/esp-idf/components/ulp/include
   /home/jeremy/esp/esp-idf/components/vfs/include
   /home/jeremy/esp/esp-idf/components/wear_levelling/include
   /home/jeremy/esp/esp-idf/components/wpa_supplicant/include
   /home/jeremy/esp/esp-idf/components/wpa_supplicant/port/include
   /home/jeremy/esp/esp-idf/components/xtensa-debug-module/include
   /home/jeremy/esp/hello_world/build/include
   /home/jeremy/esp/hello_world/build/main
   /home/jeremy/esp/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/5.2.0/include
   /home/jeremy/esp/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/5.2.0/include-fixed
   /home/jeremy/esp/xtensa-esp32-elf/xtensa-esp32-elf/include
   /home/jeremy/esp/xtensa-esp32-elf/xtensa-esp32-elf/sysroot/usr/include
```

jeremyherbert, 2018-06-13:
If I change the build output parser regex to `xtensa-esp32-elf-([gc]\+\+)` it works correctly. I'd rather not create a pull request with this though, as I am not fully aware of the broader scale implications of changing this.

projectgus, 2018-06-19:
Hi @jeremyherbert ,

I think you're right that the Build Output Parser regex is wrong. The one you're using will only match xtensa-esp32-elf-g++ not xtensa-esp32-elf-gcc, so it will break C.

I think a simpler regex which should work in all cases is `xtensa-esp32-elf-[gc]..` This drops "clang" support but we don't have a Clang toolchain yet anyhow. Could you please try this value and confirm it fixes the problem you're seeing?

jeremyherbert, 2018-06-26:
Yep, that regex works. See pull request.

closed 2018-06-27 by igrr

projectgus, 2018-06-27:
Hi @jeremyherbert,

Sorry, your PR appeared at the same time I'd independently verified the fix so we ended up doing our own patch for this.

Thanks for reporting it!

Angus

jeremyherbert, 2018-06-27:
no worries, glad to see it fixed