espressif/esp-idf #2057: example_ble_sec_gatts_demo.c has incorrect UUID in sec_service_uuid string

jcwren, 2018-06-13:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version: 98714f8
- Development Env:		[other]
- Operating System:		[MacOS]
- Power Supply:         [USB]

## Problem Description

The UUID for a BLE service or characteristic must be LSB first. In the code below, from the ```example_ble_sec_gatts_demo.c```, bytes 12 and 13 are the 16-bit service ID. For the heart rate monitor service, this should be ```0x180d```. It appears these two bytes are reversed.

```
static uint8_t  sec_service_uuid [16] =
{
  /* LSB <--------------------------------------------------------------------------------> MSB */
  //first uuid, 16bit, [12],[13] is the value
  0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0x18, 0x0D, 0x00, 0x00,
};
```

Should be:
```
  0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0x0D, 0x18, 0x00, 0x00,
```

chegewara, 2018-06-13:
I think it is on purpose. As you can see characteristic UUID is not for heart rate but for thermometer.
It is usage example, not complete solution. (just my opinion)

jcwren, 2018-06-14:
Yeah, and actually, that's a bug too. It should be for the heart rate monitor. A good example program should at least function correctly, even if it has a minimal feature set. When someone is attempting to use it as a basis to learn from, having incorrect code and/or poor documentation tends to defeat the purpose.

jcwren, 2018-06-14:
You're missing the point. I've got my code working, after beating the example into submission. Why the heck should anyone else need to, when the examples can be fixed?

Weijian-Espressif, 2018-06-25:
Thanks for your reminder, I will modify it in the v3.2.

FayeY, 2018-07-03:
Hi, this issue has been fixed, please have a try. If it is not, please feel free to reopen it. Thanks.

closed 2018-07-03 by FayeY