espressif/esp-idf #2059: Ethernet in esp32

KrrishnaShah, 2018-06-13:
I'm unable to initiate ethernet in esp32 using example code.
I'm using platformIO ESP-IDF sdk for testing Ethernet in esp32.

please guide me through this:
my test code is

	/* ethernet Example

	   This example code is in the Public Domain (or CC0 licensed, at your option.)

	   Unless required by applicable law or agreed to in writing, this
	   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
	   CONDITIONS OF ANY KIND, either express or implied.
	*/
	#include <stdio.h>
	#include <string.h>

	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"

	#include "esp_system.h"
	#include "esp_err.h"
	#include "esp_event_loop.h"
	#include "esp_event.h"
	#include "esp_attr.h"
	#include "esp_log.h"
	#include "esp_eth.h"

	#include "rom/ets_sys.h"
	#include "rom/gpio.h"

	#include "soc/dport_reg.h"
	#include "soc/io_mux_reg.h"
	#include "soc/rtc_cntl_reg.h"
	#include "soc/gpio_reg.h"
	#include "soc/gpio_sig_map.h"

	#include "tcpip_adapter.h"
	#include "nvs_flash.h"
	#include "driver/gpio.h"

	#include "soc/emac_ex_reg.h"
	#include "driver/periph_ctrl.h"

	#include "../../lib/debug/debug.h"
	#include "eth_phy/phy_lan8720.h"

	#define DEFAULT_ETHERNET_PHY_CONFIG phy_lan8720_default_ethernet_config

	#define CONFIG_PHY_ADDRESS  0
	#define PIN_SMI_MDC         23
	#define PIN_SMI_MDIO        18

	static void eth_task(void *pvParameter);
	static void eth_gpio_config_rmii(void);
	static void phy_device_power_enable_via_gpio(bool enable);

	void app_main(void)
	{
		esp_err_t ret = ESP_OK;
		tcpip_adapter_init();
		esp_event_loop_init(NULL, NULL);

		eth_config_t config; // = DEFAULT_ETHERNET_PHY_CONFIG;
		memcpy(&config, &DEFAULT_ETHERNET_PHY_CONFIG, sizeof(eth_config_t));
		/* Set the PHY address in the example configuration */
		config.phy_addr = CONFIG_PHY_ADDRESS;
		config.gpio_config = eth_gpio_config_rmii;
		config.tcpip_input = tcpip_adapter_eth_input;
		config.phy_power_enable = phy_device_power_enable_via_gpio;
[... 133 lines cut ...]
	16:42:16.339> [0;32mI (0) cpu_start: App cpu up.[0m
	16:42:16.339> [0;32mI (220) heap_init: Initializing. RAM available for dynamic allocation:[0m
	16:42:16.339> [0;32mI (227) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
	16:42:16.339> [0;32mI (233) heap_init: At 3FFCB790 len 00014870 (82 KiB): DRAM[0m
	16:42:16.339> [0;32mI (239) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
	16:42:16.339> [0;32mI (245) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
	16:42:16.339> [0;32mI (252) heap_init: At 40087DF8 len 00018208 (96 KiB): IRAM[0m
	16:42:16.339> [0;32mI (258) cpu_start: Pro cpu start user code[0m
	16:42:16.339> [0;32mI (276) cpu_start: Starting scheduler on PRO CPU.[0m
	16:42:16.339> [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
	16:42:16.339> [0;32mI (463) emac: mac version 1137a[0m
	16:42:16.339> [0;32mI (463) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
	16:42:16.339> 
	16:42:16.339> esp_eth_init success.
	16:42:16.339> 
	16:42:16.339> [0;32mI (473) emac: emac start !!!
	16:42:16.339> [0m
	16:42:16.339> [0;32mI (473) emac: emac reset done[0m
	16:42:17.092> [0;31mE (1473) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff[0m
	16:42:17.984> [0;31mE (2473) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff[0m
	16:42:18.983> [0;31mE (3473) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff[0m
	16:42:19.983> [0;31mE (4473) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff[0m
	16:42:20.983> [0;31mE (5473) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff[0m
	16:42:21.985> [0;31mE (6473) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff[0m
	16:42:22.983> [0;31mE (7473) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff[0m

negativekelvin, 2018-06-13:
You didn't say anything about hardware or if you verified phy address setting

KrrishnaShah, 2018-06-13:
My hardware is ESP32-WROM-32
Interfaced with LAN8720  (MDIO-18, MDC-23)
Phyter address is 0.

schematic:
[image]
[image]

sauttefk, 2018-06-14:
The problem could be the 74VC04 inverter in your clock signal.

KrrishnaShah, 2018-06-15:
Dear all,
The above issue has been solved (solution: phyter address = 1).
But Now my device is unable to get IP.
Is there any thing else I've to add to get IP like dhcp_start() or something ?

KrrishnaShah, 2018-06-15:
I removed 74VC04 and and changed R83 to 0 ohm and shorted. Also changed GPIO0 pull up from 330K to 4.7K.

Still no IP.

I'm unable to find what is the problem.
Please help me !!!

versamodule, 2018-06-18:
This is what I am using and it works just fine. Code is not cleaned up and used for something else to prototype with, but maybe there is something in there that may help. 
[image]
[main.zip](https://github.com/espressif/esp-idf/files/2112055/main.zip)

sauttefk, 2018-06-20:
@KrrishnaShah: Your schematic does not show a critical part: Is LED2/nINTSEL pulled low so that nINT/
REFCLKO is configured as clock output? (e.g. is the LED connected to GND or VDD?)

From the data sheet:
>  When nINTSEL is floated or pulled toVDD2A, nINT is selected for operation on the nINT/REFCLKO pin (default).
• When nINTSEL is pulled low to VSS, REFCLKO is selected for operation on the nINT/REFCLKO pin.

Matheus-Garbelini, 2018-07-08:
@sauttefk I wanted to know this also. It appears that REFCLKO or nINT is not used as the same clock from the crystal goes to the esp32 in the GPIO0 pin.

Another thing that I'm corious: for what porpoise there are 2 33Ohms resistors in series with the CLKIN pin? Is something to do with high speed, impedance matching or signal isolation?

KrrishnaShah, 2018-07-11:
@sauttefk My phyter LED schematic is as : 

[image]

KrrishnaShah, 2018-07-15:
Thanks everyone 
The issue has been solved.

Problem:  was the interference in clock and signals , 
Solution: Track redesign and done

closed 2018-07-15 by KrrishnaShah