espressif/esp-idf #2060: A2DP Source stream audio to two Sink device

rob-bits, 2018-06-13:
Hello all,

I am interesting in the functionality of stream data at the same time to multiple A2DP SINK devices. More precisely I want to stream audio from ESP32 to two bluetooth speaker. Is it possible with the current esp-idf?

Could be the A2DP_source example code modified to work with two sink devices?

I went through the current source example and I think it can be modificated to discover multiple devices and connect to them, but will it work in the end?

Can be the esp_a2d_source_connect() function called if we already connected to one sink device? Will the previous connection maintained?

If we can connect multiple sink devices already then what about the following code (in /a2dp_source/main/main.c line 277 - 280):

```
/* initialize A2DP source */
esp_a2d_register_callback(&bt_app_a2d_cb);
esp_a2d_source_register_data_callback(bt_app_a2d_data_cb); 
```

What will be the role of the data callback function? Will it stream the data to both of the connected device? Or how can we handle which data goes where?

Rob

yoelrc88, 2018-06-22:
Hi @rob-bits, I am also interested in this topic. Thanks for the white paper reference. Will be following this conversation. 
Thanks

rob-bits, 2018-06-22:
This is my current error log. It can be seen it get stuck at the av "**scb not available for avdt connection**" part. But before this error message there is an another "**BT_OSI: alarm_set failed to start timer, err 0x103**", I do not know how much is that relating to this topic.

@yoelrc88, have you had any history activity in this topic?

[debug_two_device_connection_02.pdf](https://github.com/espressif/esp-idf/files/2128921/debug_two_device_connection_02.pdf)

panafana, 2018-07-15:
Hi @rob-bits im also interested in working on this. Currently studying the A2DP standard if i find anything i will contact you

rob-bits, 2018-07-15:
Hi @panafana 

Thank you very much for your comment. Well I spent +20 hours to work out a proof-of-concept example with esp-idf for this application. But I gained no success. I could not connect two device in the same time and so I cannot test the multiple streaming.

The thing is the the bluedroid stack is more complicated as I can undenstand it and follow all of its layers. The current stack does not support any multi connection for the AVDTP or handling more SEPs as the application requires. So it requires too much work. And I did not get any support from the dev team so it is a dead end for me.

BUT, I found the btstack from Bluekitchen which is compatible with the ESP32. It is more transparent for me and more low level what I like. It already supports the multi connection. And I have already reach more success. I can connect two BT speaker to the ESP32 and I can stream music, seperatly for both of them. With a software switch I can immediatly change the sink device. Now I have some issues when I am trying to stream to both of the devices, but I think I am not too far from the final goal.

And the dev team of bluekitchen is very helpful.

Anyway here is my open issue at the btstack repository:
[https://github.com/bluekitchen/btstack/issues/143](https://github.com/bluekitchen/btstack/issues/143)

If you made any success in this topic feel free to share it. And I accept and acknowledge any help or suggestion in this topic.

Cheers,

Rob

panafana, 2018-07-21:
@rob-bits could you maybe share some code of your project so i can get up to speed faster?

rob-bits, 2018-07-21:
@panafana  Which one would you like to have? The ESP-IDF or the btstack related code?

panafana, 2018-07-21:
@rob-bits the btstack but is it java for android or is it in C?

rob-bits, 2018-07-21:
@panafana It is in c. I am on it. If I uploaded the code I will update this comment.

panafana, 2018-07-21:
@rob-bits Awesome man!
My main goal is to try and port this to be used in phones somehow (dont know if there is any OS restrictions in Android that prevent it).
If i am unsuccessful i will buy the bluetooth module for my arduino and work on that.
Again many thanks!

rob-bits, 2018-07-21:
@panafana Okay. If you reach any success keep me updated then :)

yunling-cn, 2019-06-23:
Hello, how do you implement the connection to the Bluetooth headset? I use ESP32 to connect to the Bluetooth headset and always stuck in "esp_a2d_source_connect". I just modified the official example a2dp_source. But connecting a computer can be, that is, connecting a Bluetooth headset does not work. Sometimes it suddenly crashes and then restarts and suddenly connects.

pad52, 2020-05-12:
Any updates in this?

jack0c, 2022-01-05:
This will not be implemented in any Espressif chips, we will develop BLE audio chips to replace this. Chips after ESP32C6 will support BLE audio(BLE5.2).

Ethernetz, 2022-09-03:
Are there any esp32 devices out right now that support BLE audio? If so, will those devices allow me to stream audio to multiple speakers?

jack0c, 2023-04-12:
Sorry, the BLE audio chip is not ready, still in R&D stage.

jack0c, 2024-02-07:
Next generation chips will support:
1. ESP32H4, BLE audio
2. ESP32S6, BLE audio + Classic BT TWS solution(A2DP source+2 A2DP sink)

Both of which will be taped out in 2024.

aximut, 2025-08-10:
What's the state on this @jack0c?