espressif/esp-idf #2061: Changing  number of losed beacons before AP kicks off the station

KryvosheyaIvan, 2018-06-13:
[image]

Hello!
Please tell me where can I change number of losed beacons before AP kicks off the station.

liuzfesp, 2018-06-22:
Hi @KryvosheyaIvan, we have no API to configure the number of losed beacons before AP kicks off the station. Could you describe the issue in detailed?

KryvosheyaIvan, 2018-06-22:
Hi @liuzfesp, there many issues on github and questions on esp32.com concerning esp32 specific error
`WIFI_REASON_BEACON_TIMEOUT = 200`
Although, developers of esp-idf are very active and interested in solving problems, there are no solution to that problem at all !
The problem loocks as follows: at first all is ok. WiFi is initialized (and BLE by the way). Then I periodically make request to some server and then, though WiFi signal is good, the 200 error arises followed by 201
`WIFI_REASON_NO_AP_FOUND = 201`. But what is of great concern is that after I get Disconnect event It cant just reconnect by means of `esp_wifi_connect()`. Also, as well as other users I cant reinit wifi because esp32 unexpectedly reboot every time. So in my product I am forced to reboot esp32 by means of `esp_restart()` after several times of unsuccessful attemts to just reconnect. Interesting fact - after rebooting esp32 connects perfectly at the very first attempt. So think that problem is somewhere at the lower level...

KryvosheyaIvan, 2018-06-23:
@heyinling , I switched off BLE and faced the same problem! It disconnects every 5 minutes. The result the same as it described previously.
How can I find out what IDF commit I use?

KryvosheyaIvan, 2018-06-25:
@heyinling, Here is IDF commit I use:
c3bec5b103888217d53527924e5185d49f9636ea

liuzfesp, 2018-07-08:
HI @KryvosheyaIvan just can't reproduce your issue with the commit c3bec5b, could you show me the detailed info of the AP to which the esp32 station is connected? Moreover, does this issue happen if there is no "periodically make request to some server"? Also does this issue happen on one specific board or on every board you have?

KryvosheyaIvan, 2018-07-09:
@liuzfesp , I founded where was the problem!
It was in using libcurl library: https://github.com/loboris/ESP32_curl_example .
I don`t know whether it was my fault, or it was in library, but when I rewrite the requests using sockets,
the issue gone.

closed 2018-07-09 by KryvosheyaIvan