espressif/esp-idf #2062: spi master management of byte order when the size of not byte aligned is crazy

Jegeva, 2018-06-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|]
- Kit version (for WroverKit/PicoKit/DevKitC): [NA]
- Core (if using chip or module): [NA]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 0e501e5e
- Development Env:		[Make]
- Operating System:		[Debian]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
while sending data with the spi master : 

void send_9(uint16_t data){
    esp_err_t ret;
    spi_transaction_t t;
    uint16_t workaround = data;
    memset(&t, 0, sizeof(t));        //Zero out the transaction
    t.length=9;                      //Command is >9< bits
    t.tx_buffer=&workaround;               //The data is the cmd itself
    ret=spi_device_transmit(spi, &t);//Transmit!
    assert(ret==ESP_OK);
}

### Expected Behavior
data_bit8, data_bit7,...,data_bit0 as expected
### Actual Behavior
data_bit7,...,data_bit0,data_bit15 is sent (check with a logic analyser)
### Steps to repropduce

hook an esp32 to a LA, 
set it up to use VSPI 

send 0x00ff with a size of 9 : 0b111111110 is sent
send 0x80ff with a size of 9 : 0b111111111 is sent

this makes no sense

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
-> use the spi master exemple, change the size, test the patterns i explained with a LA.

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

xiongyumail, 2018-06-15:
Hi,@Jegeva
Can you provide a completed test program for us to use? Let us solve your problem faster.

FayeY, 2018-06-22:
Any update? 

Hi @Jegeva,

Could you provide a test project so we can reproduce the problem and help solve it?

Thanks.

ginkgm, 2018-06-27:
Hi @Jegeva ,
ESP32 is a little endian system, and this is a common issue existing in little endien systems. 

In such systems, an unit32_t variable a=0x12345678 is placed like following in the memory.
 ``0x78, 0x56, 0x34, 0x12``

However the DMA don't know how your data is organized (as uint8_t, uint16_t or uint32_t), it always assume your data are ``uint_8``s. The DMA takes data from the memory from low to high address, byte by byte. i.e. ``0x78->0x56->0x34->0x12``, and send from the MSB to LSB, namely:
``01111000->01010110->00110100->00010010``

This is how ESP32 hardware works. 

To send uint16_t and uint32_t and even random length of data, please:

1. Move your data to start from the MSB of your variable (bit15 for uint16_t and bit31 for uint32_t).
2. use __builtin_bswap16 or __builtin_bswap32 to re-order the MSB byte to the lowest address

To resolve data received, please do in the reverse way.

Example:
```
#define SPI_SHIFT_DATA_8(data, len) (data<<(8-len))
#define SPI_SHIFT_DATA_16(data, len) __builtin_bswap16(data<<(16-len))
#define SPI_SHIFT_DATA_32(data, len) __builtin_bswap32(data<<(32-len))

#define SPI_GET_DATA_8(data, len) (data>>(8-len))
#define SPI_GET_DATA_16(data, len) (__builtin_bswap16(data)>>(16-len))
#define SPI_GET_DATA_32(data, len) (__builtin_bswap32(data)>>(32-len))

void send_9(uint16_t data){
esp_err_t ret;
spi_transaction_t t;
uint16_t workaround = SPI_SHIFT_DATA_16(data, 9);
memset(&t, 0, sizeof(t)); //Zero out the transaction
t.length=9; //Command is >9< bits
t.tx_buffer=&workaround; //The data is the cmd itself
ret=spi_device_transmit(spi, &t);//Transmit!
assert(ret==ESP_OK);
}
```

We plan to provide macros in the future version.

Another notice, please take care that your variable sent to the driver should be 32-bit aligned.

ginkgm, 2018-06-27:
Update: 
```
#define SPI_SHIFT_DATA(data, len) __builtin_bswap32((uint32_t)data<<(32-len))
#define SPI_REARRANGE_DATA(data, len) (__builtin_bswap32(data)>>(32-len))
```

closed 2018-07-04 by igrr

ginkgm, 2018-07-04:
Hi @Jegeva ,
The IDF is updated according to your problem. Please update and have a try.
If there's any issue, please come and feel free to reopen this.