espressif/esp-idf #2068: ota update fail

infotrino, 2018-06-15:
## Environment

- Development Kit:     custom board from infotron ([ESP32-Wrover)
- Kit version v1.0
- Core (if using chip or module): ESP32-Wrover
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): ff9a250
- Development Env:		Make
- Operating System:		Windows
- Power Supply:         USB

## Problem Description
When execute ota update,  esp_ota_end(...) gives Error ( esp_image: Image hash failed - image is corrupt)
This error is solved by roll back to previous ESP-IDF. (that is 20180612 version --> 20180505 version)

### Actual Behavior
I (58933) esp_image: segment 0: paddr=0x000c0020 vaddr=0x3f400020 size=0x06850 ( 26704) map
I (58953) esp_image: segment 1: paddr=0x000c6878 vaddr=0x3ffb0000 size=0x022d4 (  8916)
I (58963) esp_image: segment 2: paddr=0x000c8b54 vaddr=0x3ffb22d4 size=0x00000 (     0)
I (58973) esp_image: segment 3: paddr=0x000c8b5c vaddr=0x40080000 size=0x00400 (  1024)
0x40080000: _iram_start at D:/ghlee/18.esp32/esp-idf/components/freertos/xtensa_vectors.S:1685
I (58983) esp_image: segment 4: paddr=0x000c8f64 vaddr=0x40080400 size=0x070ac ( 28844)
I (59003) esp_image: segment 5: paddr=0x000d0018 vaddr=0x400d0018 size=0x135d4 ( 79316) map
0x400d0018: _stext at ??:?
I (59023) esp_image: segment 6: paddr=0x000e35f4 vaddr=0x400874ac size=0x02400 (  9216)
0x400874ac: ppSearchTxframe at ??:?
I (59033) esp_image: segment 7: paddr=0x000e59fc vaddr=0x400c0000 size=0x00064 (   100)
I (59033) esp_image: segment 8: paddr=0x000e5a68 vaddr=0x50000200 size=0x00008 (     8)
I (59033) esp_image: segment 9: paddr=0x000e5a78 vaddr=0x50000208 size=0x00000 (     0)
E (59033) esp_image: Image hash failed - image is corrupt
E (59033) avs_auth: esp_ota_end failed! (5379)

### Steps to repropduce

1.  after connect socket, call function ota_example_task()
2.  esp_ota_end(update_handl) gives error (error code=ESP_ERR_OTA_VALIDATE_FAILED)
3. If I use previous ESP-IDF(20180505), problem disappear.

### Code to reproduce this issue
int	write_ota(esp_ota_handle_t update_handle, char* p, int len)
{
    esp_err_t err = esp_ota_write( update_handle, (const void *)p, len);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Error: esp_ota_write failed! err=0x%x", err);
    }                    
	return len;
}

void ota_example_task(int socket_id)
{
    esp_err_t err;
    esp_ota_handle_t update_handle = 0 ;
    const esp_partition_t *update_partition = NULL;

    update_partition = esp_ota_get_next_update_partition(NULL);

    err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
    if (err != ESP_OK) { ESP_LOGE(TAG, "esp_ota_begin failed, error=%d", err);}
    ESP_LOGI(TAG, "esp_ota_begin succeeded");
    
    int32_t written_len=0;
    int32_t read_len=0;
    int32_t file_len=-1000;

    while (1) 
    {
        memset(text, 0, BUFFSIZE);
        read_len = recv(socket_id, text, BUFFSIZE, 0);
        if (read_len < 0) {
            ESP_LOGE(TAG, "Error: receive data error! errno=%d", errno);
            return;
        }
    	written_len+=write_ota(update_handle, text, read_len);
   		ESP_LOGI(TAG, "written len: %d", written_len);

        if (written_len==154272) //file length
        {
            ESP_LOGE(TAG, "eof");
            break;
        }
    }

	err=esp_ota_end(update_handle);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_ota_end failed! (%d)", err);
    }

    err = esp_ota_set_boot_partition(update_partition);
    if (err != ESP_OK) 
    {
        ESP_LOGE(TAG, "esp_ota_set_boot_partition failed! err=0x%x", err);
    }

    //OTA_flag = false;
    close(socket_id);
    esp_restart();
    
    return ;
}

iosix, 2018-06-16:
We are experiencing the same issue, reported directly to management team on May 24:

We are having an issue with latest SDK - images are being declared corrupt:

					esp_image_metadata_t data = {0};
					esp_partition_pos_t part_pos = {
						.offset = OtaPartition->address,
						.size = OtaPartition->size,
					};
					
					part_pos.offset = 0x10000;
					part_pos.size = 0xF0000;
					esp_image_load(ESP_IMAGE_VERIFY, &part_pos, &data);
					ESP_LOGE(tag, "IMAGE VERIFY: %u %u, %u %u, %u %u %u, %u, %u (%u %u %u) %u", data.start_addr, data.image_len,
						data.image.magic, data.image.segment_count, data.image.spi_mode, data.image.spi_speed, data.image.spi_size,
						data.image.entry_addr, data.image.wp_pin, data.image.spi_pin_drv[0], data.image.spi_pin_drv[1],
						data.image.spi_pin_drv[2], data.image.hash_appended);
					
					part_pos.offset = 0x100000;
					part_pos.size = 0xF0000;
					esp_image_load(ESP_IMAGE_VERIFY, &part_pos, &data);
					ESP_LOGE(tag, "IMAGE VERIFY: %u %u, %u %u, %u %u %u, %u, %u (%u %u %u) %u", data.start_addr, data.image_len,
						data.image.magic, data.image.segment_count, data.image.spi_mode, data.image.spi_speed, data.image.spi_size,
						data.image.entry_addr, data.image.wp_pin, data.image.spi_pin_drv[0], data.image.spi_pin_drv[1],
						data.image.spi_pin_drv[2], data.image.hash_appended);

We are seeing this:
Subtracted 48 from currcontentlength: buflen 1388

E (918793) esp_image: Image hash failed - image is corrupt

E (918794) upgrade: IMAGE VERIFY: 0 0, 0 0, 0 0 0, 0, 0 (0 0 0) 0

E (918945) esp_image: Image hash failed - image is corrupt

E (918945) upgrade: IMAGE VERIFY: 0 0, 0 0, 0 0 0, 0, 0 (0 0 0) 0

E (919098) esp_image: Image hash failed - image is corrupt

E (919099) upgrade: Failed to activate boot partition ota_1, error 5379

I used this version when I discovered the issue, and rolling back solved it:
git log -n 1
commit 358c822d6a29c8f881fd2fbbb100ec9fa9320596 (HEAD -> master, origin/master, origin/HEAD)
Merge: 21e085d2 feaa6e8a

iosix, 2018-06-16:
By the way, it works if I roll back to 595ddfd8:

E (132298) upgrade: IMAGE VERIFY: 65536 938432, 233 7, 2 0 1, 1074270224, 238 (0 0 0) 1

E (132841) upgrade: IMAGE VERIFY: 1048576 938432, 233 7, 2 0 1, 1074270224, 238 (0 0 0) 1

E (132841) upgrade: OTA END: 261

E (133452) upgrade: Boot partition activated: ota_1

iosix, 2018-06-16:
Ok, so it looks like the many recent changes to the SHA256 code have caused this problem.  Here is a log of OTAing the same image and then a slightly different image.  I made a code change to ignore the SHA256 error, as well.  The bootloader is calculating the SHA256 correctly, but not the runtime code:

E (272944) esp_image: Calculated hash: 00000000

E (272945) esp_image: Image hash failed - image is corrupt

E (272946) esp_image: Expected hash: cd818cf9

E (272946) upgrade: IMAGE VERIFY: 65536 949600, 233 9, 2 0 1, 1074270372, 238 (0 0 0) 1

E (273135) esp_image: Calculated hash: 00000000

E (273136) esp_image: Image hash failed - image is corrupt

E (273137) esp_image: Expected hash: cd818cf9

E (273137) upgrade: IMAGE VERIFY: 1048576 949600, 233 9, 2 0 1, 1074270372, 238 (0 0 0) 1

E (273137) upgrade: OTA END: 261

E (273326) esp_image: Calculated hash: 00000000

E (273327) esp_image: Image hash failed - image is corrupt

E (273328) esp_image: Expected hash: cd818cf9

E (273398) upgrade: Boot partition activated: ota_1

E (273591) esp_image: Calculated hash: 00000000

E (273592) esp_image: Image hash failed - image is corrupt

E (273593) esp_image: Expected hash: cd818cf9

E (273658) upgrade: Boot partition activated: ota_1

Flash success and closing socket

RESETTING

E (342) esp_image: Calculated hash: cd818cf9

--------------------------------------

E (360005) esp_image: Calculated hash: 00000000

E (360006) esp_image: Image hash failed - image is corrupt

E (360007) esp_image: Expected hash: 2bc866c1

E (360007) upgrade: IMAGE VERIFY: 65536 949600, 233 9, 2 0 1, 1074270372, 238 (0 0 0) 1

E (360196) esp_image: Calculated hash: 00000000

E (360197) esp_image: Image hash failed - image is corrupt

E (360198) esp_image: Expected hash: cd818cf9

E (360198) upgrade: IMAGE VERIFY: 1048576 949600, 233 9, 2 0 1, 1074270372, 238 (0 0 0) 1

E (360198) upgrade: OTA END: 261

E (360385) esp_image: Calculated hash: 00000000

E (360386) esp_image: Image hash failed - image is corrupt

E (360387) esp_image: Expected hash: 2bc866c1

E (360454) upgrade: Boot partition activated: ota_0

Flash success and closing socket

RESETTING

E (342) esp_image: Calculated hash: 2bc866c1

iosix, 2018-06-16:
This is my version of the end of esp_image_format.c in components/bootloader_support/src/:

static esp_err_t verify_simple_hash(bootloader_sha256_handle_t sha_handle, esp_image_metadata_t *data)
{
    uint8_t image_hash[HASH_LEN] = { 0 };
    bootloader_sha256_finish(sha_handle, image_hash);

    // Log the hash for debugging
    debug_log_hash(image_hash, "Calculated hash");

    // Simple hash for verification only
    const void *hash = bootloader_mmap(data->start_addr + data->image_len - HASH_LEN, HASH_LEN);
    if (memcmp(hash, image_hash, HASH_LEN) != 0) {
        ESP_LOGE(TAG, "Image hash failed - image is corrupt");
		debug_log_hash(hash, "Expected hash");
        //bootloader_munmap(hash);
        //return ESP_ERR_IMAGE_INVALID;
    }

    bootloader_munmap(hash);
    return ESP_OK;
}

// Log a hash as a hex string
static void debug_log_hash(const uint8_t *image_hash, const char *label)
{
//#if BOOT_LOG_LEVEL >= LOG_LEVEL_DEBUG
        char hash_print[sizeof(image_hash)*2 + 1];
        hash_print[sizeof(image_hash)*2] = 0;
        for (int i = 0; i < sizeof(image_hash); i++) {
            for (int shift = 0; shift < 2; shift++) {
                uint8_t nibble = (image_hash[i] >> (shift ? 0 : 4)) & 0x0F;
                if (nibble < 10) {
                    hash_print[i*2+shift] = '0' + nibble;
                } else {
                    hash_print[i*2+shift] = 'a' + nibble - 10;
                }
            }
        }
        ESP_LOGE(TAG, "%s: %s", label, hash_print);
//#endif
}

KonstantinKondrashov, 2018-06-19:
Hi, @infotrino, @iosix.  Could you send your `sdkconfig.h`? Thanks.

infotrino, 2018-06-19:
Hi, My sdkconfig.h is ..

----------------------------------------------------------------
/*
 *
 * Automatically generated file; DO NOT EDIT.
 * Espressif IoT Development Framework Configuration
 *
 */
#define CONFIG_ESP32_PHY_MAX_TX_POWER 20
#define CONFIG_TRACEMEM_RESERVE_DRAM 0x0
#define CONFIG_FREERTOS_MAX_TASK_NAME_LEN 16
#define CONFIG_FATFS_LFN_NONE 1
#define CONFIG_TCP_RECVMBOX_SIZE 18
#define CONFIG_FATFS_CODEPAGE_437 1
#define CONFIG_LWIP_ETHARP_TRUST_IP_MAC 1
#define CONFIG_TCP_WND_DEFAULT 22976
#define CONFIG_SPIFFS_USE_MAGIC_LENGTH 1
#define CONFIG_ESPTOOLPY_FLASHSIZE_4MB 1
#define CONFIG_IPC_TASK_STACK_SIZE 1024
#define CONFIG_FATFS_PER_FILE_CACHE 1
#define CONFIG_AP_BEACON_INTERVAL 100
#define CONFIG_ESPTOOLPY_FLASHFREQ "40m"
#define CONFIG_AP_SSID "avsatom-1"
#define CONFIG_MBEDTLS_KEY_EXCHANGE_RSA 1
#define CONFIG_UDP_RECVMBOX_SIZE 6
#define CONFIG_AP_CHANNEL 0
#define CONFIG_FREERTOS_QUEUE_REGISTRY_SIZE 0
#define CONFIG_MBEDTLS_AES_C 1
#define CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED 1
#define CONFIG_MBEDTLS_GCM_C 1
#define CONFIG_ALEXA_LOCALE_EN_US 1
#define CONFIG_ESPTOOLPY_FLASHSIZE "4MB"
#define CONFIG_HEAP_POISONING_DISABLED 1
#define CONFIG_SPIFFS_CACHE_WR 1
#define CONFIG_SPIRAM_CACHE_WORKAROUND 1
#define CONFIG_BROWNOUT_DET_LVL_SEL_0 1
#define CONFIG_SPIFFS_CACHE 1
#define CONFIG_INT_WDT 1
#define CONFIG_MBEDTLS_SSL_PROTO_TLS1 1
#define CONFIG_MBEDTLS_ECDSA_C 1
#define CONFIG_ESPTOOLPY_FLASHFREQ_40M 1
#define CONFIG_LOG_BOOTLOADER_LEVEL_INFO 1
#define CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE 0
#define CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS 1
#define CONFIG_MBEDTLS_ECDH_C 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE 1
#define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 25
#define CONFIG_MBEDTLS_SSL_ALPN 1
#define CONFIG_MBEDTLS_PEM_WRITE_C 1
#define CONFIG_LOG_DEFAULT_LEVEL_INFO 1
#define CONFIG_BT_RESERVE_DRAM 0x0
#define CONFIG_FATFS_FS_LOCK 0
#define CONFIG_IP_LOST_TIMER_INTERVAL 120
#define CONFIG_ALEXA_LOCALE "en-US"
#define CONFIG_SPIFFS_META_LENGTH 4
#define CONFIG_WIFI_AUTH_WPA2_PSK 1
#define CONFIG_ESP32_PANIC_PRINT_REBOOT 1
#define CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED 1
#define CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION 1
#define CONFIG_CONSOLE_UART_BAUDRATE 115200
#define CONFIG_SPIRAM_SUPPORT 1
#define CONFIG_LWIP_MAX_SOCKETS 4
#define CONFIG_LWIP_NETIF_LOOPBACK 1
#define CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL 12000
#define CONFIG_SPIRAM_TYPE_ESPPSRAM32 1
#define CONFIG_EMAC_TASK_PRIORITY 20
#define CONFIG_TIMER_TASK_STACK_DEPTH 2048
#define CONFIG_TCP_MSS 1436
[... 162 lines cut ...]
#define CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA 1
#define CONFIG_MBEDTLS_TLS_CLIENT_ONLY 1
#define CONFIG_WIFI_PASSWORD "qqqq4444"
#define CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM 64
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE 32
#define CONFIG_ESP32_WIFI_TX_BUFFER_TYPE 0
#define CONFIG_SUPPORT_STATIC_ALLOCATION 1
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 1
#define CONFIG_LWIP_LOOPBACK_MAX_PBUFS 8
#define CONFIG_APP_OFFSET 0x10000
#define CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP 3
#define CONFIG_SPI_FLASH_ROM_DRIVER_PATCH 1
#define CONFIG_SPIFFS_PAGE_SIZE 256
#define CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED 1
#define CONFIG_AUDIO_OUTPUT_MODE_I2S 1
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 1
#define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 2048
#define CONFIG_LWIP_SO_RCVBUF 1
#define CONFIG_MBEDTLS_HARDWARE_MPI 1
#define CONFIG_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_NEWLIB_STDOUT_LINE_ENDING_CRLF 1
#define CONFIG_ESPTOOLPY_PORT "COM3"
#define CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS 1
#define CONFIG_OPTIMIZATION_LEVEL_RELEASE 1

iosix, 2018-06-19:
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=y
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=1
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=

#
# Security features
#
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="COM15"
CONFIG_ESPTOOLPY_BAUD_115200B=
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=y
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=921600
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_ESPTOOLPY_AFTER="no_reset"
CONFIG_MONITOR_BAUD_9600B=
CONFIG_MONITOR_BAUD_57600B=
[... 670 lines cut ...]
#
# Debug Configuration
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

KonstantinKondrashov, 2018-06-21:
Hi, @infotrino, @iosix.  I tried to reproduce your error but I can't. We have an example `~/esp/esp-idf/examples/system/ota`, it works without error. I think this simple example, on your side, works also good, as well. Is it true? I think error depends on other tasks in FreeRTOS. Could you give a code which reproduces this error?

Could you confirm flash chip size by doing? 
1 `esptool.py --port COM4 flash_id`
2. partition table `make partition_table` 
3. size your image firmware.

KonstantinKondrashov, 2018-06-21:
@infotrino. I note one moment in your code. Why do you have constant `154272` for the length of firmware file?  `if (written_len==154272) //file length`. The length of firmware file will be really different each time.

iosix, 2018-06-21:
It’s very clear that the changes to the SHA256 function between the two commit versions I mentioned have resulted in SHA256 returning 0 when running live. This is why it isn’t working. Please investigate the code changes for correctness. Many of them have to do with timing and synchronization.  

Robert Vogt IV
CEO
IOSiX, LLC
389 Airport Industrial Drive
Ypsilanti, MI  48198
P: +1-855-OBD-1939
C: +1-734-730-9690
robert@iosix.com

infotrino, 2018-06-22:
Hi KonstantinKondrshov,
File length is variable in my local code.  
I fixed that size only for debugging.
Thanks.
I will try an example "~/esp/esp-idf/examples/system/ota" sooner or later.

KonstantinKondrashov, 2018-06-22:
Hi @infotrino, @iosix. I reproduced your error. It depends on the option `CONFIG_OPTIMIZATION_ASSERTIONS_DISABLED=y`. You can switch this option on `CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED=y` or wait for the update our fix, we will merge it soon.
Thanks.

infotrino, 2018-06-24:
Hi KonstantinKondrshov,
ota is normal with CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED=y.
Thank you.

closed 2018-06-26 by igrr