espressif/esp-idf #2070: BLE related asserts and crashes in recent SDKs

rojer, 2018-06-18:
after updating IDF to https://github.com/cesanta/esp-idf/commit/9a55b42f0841b3d38a61089b1dda4bf28135decd i am and seeing new crashes and ASSERTs related to BLE:

```
ASSERT_PARAM(0 0), in rwble.c at line 222
```
followed by ISR WDT crash

and a different one:
```
0x40086e3c in r_lld_evt_end_isr ()                    
#0  0x40086e3c in r_lld_evt_end_isr ()
#1  0x40087d9d in r_rwble_isr ()                              
#2  0x40088d16 in r_rwbtdm_isr_wrapper ()                           
#3  0x40081ae4 in _xt_lowint1 ()                                                                
    at /opt/Espressif/esp-idf/components/freertos/xtensa_vectors.S:1105                                     
```

there is a passive BLE scan ongoing, no other BLE activity. WiFi is enabled in station mode.

i'm updating IDF from https://github.com/cesanta/esp-idf/commit/441b4a974226c2deaa9ffb8e7308c3009985aba6 which did not have these problems.

rojer, 2018-06-18:
for the second one i can add that it's a NULL deref (EXCVADDR = 0x00000010)

projectgus, 2018-06-19:
Hi @rojer & @chegewara,

These issues are caused by Bluetooth modem sleep, which was enabled by default in this commit. Currently there are known issues with Bluetooth modem sleep, and it's been disabled by default in master since ae6b47a60a341c2544ec370b3cd2f12ef0279fd2

Disabling Bluetooth modem sleep, or updating to latest master branch and keeping the default config should resolve this issue.

Will keep the issue open until the problems with Bluetooth modem sleep are resolved.

chegewara, 2018-06-19:
Good to know where the issue is, bad it cant be fixed in arduino until master commit get recompiled or use older arduino commit. Thanks for update.

rojer, 2018-06-19:
@projectgus thanks Angus!

Weijian-Espressif, 2018-07-04:
hi all:
    we fix some bugs in bluetooth modem sleep, please update to the latest IDF and try, thanks.

FayeY, 2018-07-12:
Closing due to lack of response, please feel free to reopen if this is still unsolved on the latest ESP-IDF.
Thanks.

ANDnXOR, 2018-07-12:
I am still getting this occasionally. It's far less often, but happens after 1 to 7 hours of execution. I have the latest release/3.1 branch and modem sleep enabled.

projectgus, 2018-07-13:
@ANDnXOR Do you have any crash dumps or other logs from the failures you're getting with the current release/v3.1 branch?

ANDnXOR, 2018-07-19:
I'm on the June 29th Release/3.1 branch and just got

 "ASSERT_PARAM(32 0), in rwble.c at line 222"

0BackDoor1, 2018-07-25:
Hi,
I have IDF updated at commit f198339ec09e90666150672884535802304d23ec , blt modem sleep **disabled** on menuconfig, but I get this assert too:
`ASSERT_PARAM(32768 0), in rwble.c at line 222`
`Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)`

Weijian-Espressif, 2018-07-26:
@all please update to the latest IDF and try. if the issue is still exist, please provide the commit id and how to reproduce it, thanks.

0BackDoor1, 2018-07-26:
my idf was at latest commit available on branch release/v3.1 yesterday, anyway I tried switching to latest commit on master branch ( be81d2c16d7f4caeea9ceb29fece01510664caf3 ) and it still occur.
Actually I don't know how to reproduce it because it happens very randomly to me, from 0 to 3-4 per day, at the moment I didn't find any recurrent details between each crash, sometimes seems that happen during bluetooth connection, othertimes without any active bluetooth connection, simply keeping board on.
Since I don't know at all what can be the source of issue, I don't know which part of code I should provide for the example, I can't share the entire project due to privacy reason.
Seems that the same issue occurs to many people here, is there anybody that have a small project where it happen and that can share it entirely?

ANDnXOR, 2018-07-28:
Just got ASSERT_PARAM(131072 0), in rwble.c at line 222

This is with latest IDF,  561884b

davctv, 2018-08-09:
Hi,

I have also problem with rwble.c

`ASSERT_PARAM(1024 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084abd  PS      : 0x00060034  A0      : 0x8008846e  A1      : 0x3ffc0820  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc01dc  A7      : 0x60008050  A8      : 0x80084abd  A9      : 0x3ffc0800  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400849f5  LEND    : 0x400849fc  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4016a3ec  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084abd

Backtrace: 0x40084abd:0x3ffc0820 0x4008846b:0x3ffc0840 0x4008933f:0x3ffc0860 0x40082721:0x3ffc0880 0x4016a3e9:0x00000000

Core 1 register dump:
PC      : 0x401a45ae  PS      : 0x00060134  A0      : 0x80092872  A1      : 0x3ffd9900  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffd9428  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc8b70  A9      : 0x3ffc8b34  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80093402  A13     : 0x3ffd9810  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x401a45ae:0x3ffd9900 0x4009286f:0x3ffd9920

Entering gdb stub now.
$T06#ba`

My scenario:
-WiFi turned on in STA mode with SSID and password wrong (no AP available so it disconnect and then try to reconnect again)
-BLE in advertizing mode ("eddystone beacon style")

Disabling the BLE advertizing doesn't trigger the crash.

The complete project is quite big and involves a lot of stuff (I/O, I2C, SPIFFS, UART,ecc..) and it can't be shared. I have created a test project with the above scenario, in order to reproduce the error. I'm using release 3.0.3-rc (commit 054e82b [https://github.com/espressif/esp-idf/commit/054e82b4b32d231065be5af2f97f94148152fe6b](url)).

If the test project crash i will share it on github.

Thanks.

chegewara, 2018-08-09:
Looks like this issue persist over a year now and "has been fixed" few times already:
https://github.com/espressif/esp-idf/issues/880

davctv, 2018-08-09:
Hi @chegewara,

If the issue is related to "bluetooth modem sleep", do you know a method to disable this feature. For our application at moment we don't care about low-power so even if ble modem doesn't sleep it is ok.

The test I prepared is here: [https://github.com/davctv/test_rwble_crash](url)
At moment it didn't crash . Looking forward for a solution or some hint by espressif-guys.

Thanks

Pyponou, 2018-08-10:
Hi, I am having the same issue after 8 hours WITHOUT the Bluetooth modem sleep : 

> ASSERT_PARAM(131072 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084011  PS      : 0x00060434  A0      : 0x800879be  A1      : 0x3ffc07b0  
0x40084011: r_assert_param at ??:?
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc081c  A7      : 0x60008050  A8      : 0x80084011  A9      : 0x3ffc0790  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40083f49  LEND    : 0x40083f50  LCOUNT  : 0x00000000  
0x40083f49: r_assert_param at ??:?
0x40083f50: r_assert_param at ??:?
Core 0 was running in ISR context:
EPC1    : 0x40086869  EPC2    : 0x4000bff0  EPC3    : 0x00000000  EPC4    : 0x40084011
0x40086869: r_lld_evt_end_isr at ??:?
0x40084011: r_assert_param at ??:?
Backtrace: 0x40084011:0x3ffc07b0 0x400879bb:0x3ffc07d0 0x4008888f:0x3ffc07f0 0x40082256:0x3ffc0810 0x4000bfed:0x00000000
0x40084011: r_assert_param at ??:?
0x400879bb: r_rwble_isr at ??:?
0x4008888f: r_rwbtdm_isr_wrapper at intc.c:?
0x40082256: _xt_lowint1 at /home/ivan/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

I also noticed some crashs with the same kind of backtrace when closing a gattc thanks to esp_ble_gattc_close while reading/writing a characteristic on the device.

Weijian-Espressif, 2018-08-10:
Hi all:  
    I test our BLE for a long time, but i can not reproduce the issues. we need your help. 
   we need some detailed information:
1. the commit id of IDF
2. detailed log
3. only BLE ? or BLE and wifi are both enable?
4. test code if you have

chegewara, 2018-08-10:
HI, my app has been running on arduino-ide so its not very helpful but here are full logs:
https://github.com/espressif/esp-idf/issues/2073

I didnt test it since then, app is BLE + WiFi (mqtt).

Pyponou, 2018-08-10:
esp-idf commit id : 4225b9794fe70f2f4b86fc961f23d103c94a3277

The only detailed log that could help you are the ones appearing when closing a gattc thanks to esp_ble_gattc_close while reading/writing a characteristic on the device but I am not sure it is due to the same issue : 
<details>
  <summary>Click to see logs</summary>

> E (109642) BT_GATT: GATTC_ConfigureMTU GATT_BUSY conn_id = 1
E (109642) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109642) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109642) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109642) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109642) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109642) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109642) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109652) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT_BUSY conn_id = 1
E (109662)
=============STACK OVERFLOW
=============STACK OVERFLOW
  OVERFLOW
ATTC_Read GATT_BUSY conn_id = 1
E (109662) BT_GATT: GATTC_Read GATT
=============STACK OVERFLOW
=============STACK OVERFLOW
_BUSY conn_id = 1
tm_ble_scan_pf_cmpl_cback:4-Incorrect opcode: %d, %d, %d, %d, %d
E (%d) %s: btm_ble_scan_pf_cmpl_cback: unknown operation: %d
E (%d) %s: Local Name PF filter update failed
E (%d) %s: btm_ble_update_pf_manu_data - No manuf data
E (%d) %s: manufacturer data PF filter update failed
E (%d) %s: Broadcaster Address Filter Update failed
E (%d) %s: Illegal param for add/delete UUID filter
E (%d) %s: Update Address filter into controller failed.
E (%d) %s: illegal UUID length: %d
E (%d) %s: UUID filter udpating failed
E (%d) %s: unable to perform action:%d for generic adv filter type
E (%d) %s: Error: Can not clear filter, No PF filter has been configured!
E (%d) %s: BD Address not found!
W (%d) %s: condition type [%d] not supported currently.
@*@@@�@�@�@;@btm_ble_adv_filter_cleanupbtm_ble_adv_filter_initbtm_ble_scan_pf_cmpl_cbackW (%d) %s: btm_convert_uuid_to_uuid16 invalid uuid size
I (%d) %s: BTM_SetDiscoverability
I (%d) %s: BTM_SetDiscoverability: mode %d [NonDisc-0, Lim-1, Gen-2], window 0x%04x, interval 0x%04x
I (%d) %s: BTM_SetInquiryScanType
I (%d) %s: BTM_SetPageScanType
I (%d) %s: BTM_SetInquiryMode
I (%d) %s: BTM_ReadDiscoverability
I (%d) %s: BTM_SetPeriodicInquiryMode: mode: %d, dur: %d, rsps: %d, flt: %d, min: %d, max: %d
I (%d) %s: BTM_CancelPeriodicInquiry called
I (%d) %s: BTM_SetConnectability
I (%d) %s: BTM_SetConnectability: mode %d [NonConn-0, Conn-1], window 0x%04x, interval 0x%04x
[... 63 lines cut ...]
design max limit.
E (%d) %s: L2CAP - no buffer ASSERT_PARAM(16384 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400842f1  PS      : 0x00060034  A0      : 0x8008769e  A1      : 0x3ffc07b0  
0x400842f1: r_assert_param at ??:?
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc081c  A7      : 0x60008050  A8      : 0x800842f1  A9      : 0x3ffc0790  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40084229  LEND    : 0x40084230  LCOUNT  : 0x00000000  
0x40084229: r_assert_param at ??:?
0x40084230: r_assert_param at ??:?
Core 0 was running in ISR context:
EPC1    : 0x40100478  EPC2    : 0x4000bff0  EPC3    : 0x00000000  EPC4    : 0x400842f1
0x40100478: uart_tx_char at /home/ivan/esp/esp-idf/components/vfs/vfs_uart.c:139
0x400842f1: r_assert_param at ??:?
Backtrace: 0x400842f1:0x3ffc07b0 0x4008769b:0x3ffc07d0 0x40088573:0x3ffc07f0 0x400821c2:0x3ffc0810 0x40100475:0x00000000
0x400842f1: r_assert_param at ??:?
0x4008769b: r_rwble_isr at ??:?
0x40088573: r_rwbtdm_isr_wrapper at intc.c:?
0x400821c2: _xt_lowint1 at /home/ivan/esp/esp-idf/components/freertos/xtensa_vectors.S:1105
0x40100475: uart_tx_char at /home/ivan/esp/esp-idf/components/vfs/vfs_uart.c:139

</details>

Weijian-Espressif, 2018-08-10:
@Pyponou  do you init WIFI? Can you provide your test code?

jakkra, 2018-08-24:
@Weijian-Espressif We have major stability problems with BLE Modem sleep on latest esp-idf master. 
Our application is running BLE and Classic. The crash happens just after like 5 seconds sending data over BLE, even at low speeds. We noticed that turning the CPU Freq from 160 to 240MHz did help alot and we could send data for maybe 15 minutes before crashing. If we turn off Bluetooth modem sleep we have no problems.

Below is a coredump from the crash that we see. The address` 0x40015c4a` which coredump fails to decode, seems to be in the ROM function `AddBigHex256` when I checked it in the .map file.

A note is that we are not running Bluedroid stack, but Bluekitchen instead. However this is probably not a stack related.

I can't share any code, but will update if I manage to reproduce this with some example application.
```
==================== CURRENT THREAD STACK =====================
#-1 0x40015c4a in ?? ()
Backtrace stopped: Cannot access memory at address 0x40015c4a

======================== THREADS INFO =========================
  Id   Target Id         Frame
  11   process 10        _frxt_int_enter () at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos\\portasm.S:135
  10   process 9         0x40082332 in _xt_user_exc () at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos\\xtensa_vectors.S:630
  9    process 8         _frxt_int_enter () at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos\\portasm.S:135
  8    process 7         _frxt_int_enter () at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos\\portasm.S:135
  7    process 6         _frxt_int_enter () at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos\\portasm.S:135
  6    process 5         0x4013ffda in mbedtls_mpi_get_bit (X=0x8, pos=0) at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/mbedtls/mbedtls/library/bignum.c:319
  5    process 4         0x4013ffda in mbedtls_mpi_get_bit (X=0x8, pos=1) at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/mbedtls/mbedtls/library/bignum.c:319
  4    process 3         0x4000bff0 in ?? ()
  3    process 2         soc_get_available_memory_regions (regions=0x3ffc5efc <eventDataRingBuf+1636>) at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/soc/src/memory_layout_utils.c:177
  2    process 1         _frxt_int_enter () at C:/git/nina-esp/component/cb_rtsl_esp32/esp-idf/components/freertos\\portasm.S:135
* 1    <main task>       0x40015c4a in ?? ()

======================= ALL MEMORY REGIONS ========================
Name   Address   Size   Attrs
.rtc.text 0x400c0000 0x0 RW
.rtc_noinit 0x50000000 0x0 RW
.iram0.vectors 0x40080000 0x400 R XA
.iram0.text 0x40080400 0x12898 R XA
.dram0.data 0x3ffc0000 0x3968 RW A
.flash.rodata 0x3f400020 0x218f8 RW A
.flash.text 0x400d0018 0x70218 R XA
.coredump.tasks.data 0x3ffd616c 0x164 RW
.coredump.tasks.data 0x3ffd5a90 0x6d4 RW
.coredump.tasks.data 0x3ffcf5c8 0x164 RW
.coredump.tasks.data 0x3ffcf3d0 0x1f0 RW
.coredump.tasks.data 0x3ffd47f0 0x164 RW
.coredump.tasks.data 0x3ffd4550 0x298 RW
.coredump.tasks.data 0x3ffd0b54 0x164 RW
.coredump.tasks.data 0x3ffd0910 0x23c RW
.coredump.tasks.data 0x3ffd00f4 0x164 RW
.coredump.tasks.data 0x3ffcff80 0x16c RW
.coredump.tasks.data 0x3ffcfb88 0x164 RW
.coredump.tasks.data 0x3ffcfa20 0x160 RW
.coredump.tasks.data 0x3ffd2470 0x164 RW
.coredump.tasks.data 0x3ffd22d0 0x198 RW
.coredump.tasks.data 0x3ffccf00 0x164 RW
.coredump.tasks.data 0x3ffccd70 0x188 RW
.coredump.tasks.data 0x3ffcc940 0x164 RW
.coredump.tasks.data 0x3ffcc7b0 0x188 RW
.coredump.tasks.data 0x3ffcd344 0x164 RW
.coredump.tasks.data 0x3ffd1d50 0x1a0 RW
.coredump.tasks.data 0x3ffcc22c 0x164 RW
.coredump.tasks.data 0x3ffcc0a0 0x184 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

jakkra, 2018-08-31:
@Weijian-Espressif Yes we have it disabled while waiting for a fix. Have you identified the problem or do you need more information? Anything I can do to help?

Thank you

Weijian-Espressif, 2018-08-31:
Hi jakkra:
    I don't know what happened, I can't reproduce this issue.

jakkra, 2018-08-31:
Hi @Weijian-Espressif I will try to get more information. Our application works very bad when I enable BT power save, pretty much unusable. I'll see if I can get you better stack traces. Any idea what I can try?

Weijian-Espressif, 2018-08-31:
@jakkra  Thanks. we fix some bugs about BLE sleep before, if you use other stack, please provide your controller version.

jakkra, 2018-08-31:
@Weijian-Espressif We are on esp-idf release/v3.1 and controller commit d3c834d66153b1acd6b4e5e744a66aaa55ae5c40 so not the last 3 commits, however they doesn't seem relevant. I tested also on esp-idf master just a few days ago, and same problem.

jakkra, 2018-08-31:
@Weijian-Espressif I can not, it's a large application. I have not been able to reproduce it with any of the examples. I'm going to see if I can reproduce it running our application on windows instead on the esp32 and using esp32 as hci_controller over UART. If that is possible I might be able to send you a windows executable which uses the controller_hci_uart_demo example.

jakkra, 2018-08-31:
@Weijian-Espressif I was not able to reproduce it using the hci controller and windows setup previously mentioned. It seems as it's the TIMERG1 watchdog which triggers a reset. I will see if I can find out where it's stuck. Might be related to our other issue https://github.com/espressif/esp-idf/issues/2337

jakkra, 2018-09-03:
This patch seems to have solves this issue aswell: https://github.com/espressif/esp-idf/issues/2337#issuecomment-418038595

I will do some more testing, but it looks very promising.

Weijian-Espressif, 2018-09-04:
@jakkra thanks for your help

projectgus, 2018-10-07:
@sjoteppagol 

> " So by disabling this option is enough to over come this issue right!?
> CONFIG_BTDM_CONTROLLER_MODEM_SLEEP=y

Yes, if you're having problems which may be related to Bluetooth modem sleep then this config item will disable that feature

dipankarpanda, 2018-11-27:
Hello,
We are also facing the similar issue. Can anybody please let me know, how to configure this option in esp-idf version 3.0, as I didn't find any option to configure this in 3.0.

Thanks
Dipankar

bill1davis, 2019-11-19:
esp-idf v3.2 90af68bd3c1941d9f5513bee22a12de7250208d7  also seen in 3.3.  BLE is not connected and advertising, wifi is in client mode serving a web server at maybe 6 rest queries per second.  Happens once or twice in 24 hours.

ASSERT_PARAM(512 0), in rwble.c at line 235
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40089725  PS      : 0x00060034  A0      : 0x8008d3e0  A1      : 0x3ffbe2a0  
0x40089725: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdcc8  A7      : 0x60008050  A8      : 0x80089725  A9      : 0x3ffbe280  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4008965d  LEND    : 0x40089664  LCOUNT  : 0x00000000  
0x4008965d: r_assert_param at ??:?

0x40089664: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40081e88  EPC2    : 0x400824e7  EPC3    : 0x00000000  EPC4    : 0x40089725
0x40081e88: timer_overflow_happened at /home/dev/esp/esp-idf/components/esp32/esp_timer_esp32.c:162

0x400824e7: esp_intr_noniram_enable at /home/dev/esp/esp-idf/components/esp32/intr_alloc.c:776

0x40089725: r_assert_param at ??:?

Backtrace: 0x40089725:0x3ffbe2a0 0x4008d3dd:0x3ffbe2c0 0x4008dfcb:0x3ffbe2e0 0x4008772d:0x3ffbe300 0x40081e85:0x3ffd1b00 0x40081f67:0x3ffd1b20 0x40081e0f:0x3ffd1b40 0x400858b3:0x3ffd1b60 0x400856d1:0x3ffd1b80 0x40084eaf:0x3ffd1bb0 0x40084f96:0x3ffd1bd0 0x40085191:0x3ffd1bf0 0x400fccce:0x3ffd1c30 0x400d1401:0x3ffd1c50 0x400f0755:0x3ffd1c70 0x400f35b9:0x3ffd1c90 0x400f3645:0x3ffd1cb0 0x4009644d:0x3ffd1cd0
0x40089725: r_assert_param at ??:?

0x4008d3dd: r_rwble_isr at ??:?

0x4008dfcb: r_rwbtdm_isr_wrapper at intc.c:?

0x4008772d: _xt_lowint1 at /home/dev/esp/esp-idf/components/freertos/xtensa_vectors.S:1154

0x40081e85: timer_overflow_happened at /home/dev/esp/esp-idf/components/esp32/esp_timer_esp32.c:159

0x40081f67: esp_timer_impl_get_time at /home/dev/esp/esp-idf/components/esp32/esp_timer_esp32.c:200

0x40081e0f: esp_timer_get_time at /home/dev/esp/esp-idf/components/esp32/esp_timer.c:515

0x400858b3: coex_time_now at ??:?

0x400856d1: coex_force_wifi_mode at ??:?

0x40084eaf: coex_core_ts_start at ??:?

0x40084f96: coex_core_ts_start at ??:?

0x40085191: coex_core_request at ??:?

0x400fccce: coex_wifi_request at ??:?

0x400d1401: coex_wifi_request_wrapper at /home/dev/esp/esp-idf/components/esp32/esp_adapter.c:447

0x400f0755: pm_tbtt_process at ??:?

0x400f35b9: dbg_lmac_init at ??:?

0x400f3645: pp_timer_do_process at ??:?

0x4009644d: ppTask at ??:?

Core 1 register dump:
PC      : 0x40090f47  PS 0x40090f47: vPortCPUAcquireMutexIntsDisabledInternal at /home/dev/esp/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/dev/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/dev/esp/esp-idf/components/freertos/tasks.c:4201

     : 0x00060134  A0      : 0x80091684  A1      : 0x3ffbb350  
A2      : 0x3ffbf6e0  A3      : 0x00060120  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x000000fe  A7      : 0x00060023  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0x0000abab  A11     : 0x00000000  A12     : 0x3f40b9a1  A13     : 0x3f40b65a  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400916aa  LEND    : 0x400916b1  LCOUNT  : 0x00000001  
0x400916aa: xHaveReadyTasks at /home/dev/esp/esp-idf/components/freertos/tasks.c:3507
 (inlined by) prvGetExpectedIdleTime at /home/dev/esp/esp-idf/components/freertos/tasks.c:2186

0x400916b1: xHaveReadyTasks at /home/dev/esp/esp-idf/components/freertos/tasks.c:3507
 (inlined by) prvGetExpectedIdleTime at /home/dev/esp/esp-idf/components/freertos/tasks.c:2186

Backtrace: 0x40090f47:0x3ffbb350 0x40091681:0x3ffbb380 0x40091760:0x3ffbb3a0
0x40090f47: vPortCPUAcquireMutexIntsDisabledInternal at /home/dev/esp/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/dev/esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/dev/esp/esp-idf/components/freertos/tasks.c:4201

0x40091681: prvGetExpectedIdleTime at /home/dev/esp/esp-idf/components/freertos/tasks.c:3507

0x40091760: prvIdleTask at /home/dev/esp/esp-idf/components/freertos/tasks.c:3507

jack0c, 2022-05-17:
idfv3.2 has gone end of live. 
This issue is fixed since IDFv3.3.

closed 2022-05-17 by jack0c