espressif/esp-idf #2071: ESP_ERR_TIMEOUT during i2c_Read

hvishu, 2018-06-18:
Below is the wrapper function fro I2C_Read we have done in our product.  We have kept our product for duration testing.  In this, we saw error 0x107 returned twice(out of about 48 hrs).  Most importantly, along with this error 0x107, the board also gets reset without any additional print messages.  Could you provide us some pointer for this?  
We are using IDF release 3.0

bool esphal_I2CRead (i2c_port_t port, uint8_t devAddr, uint8_t regAddr, uint8_t* data)
{
    i2c_cmd_handle_t cmd;
    esp_err_t ret;
    cmd = i2c_cmd_link_create();
    ret = i2c_master_start(cmd);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_write_byte(cmd, devAddr << 1 | I2C_MASTER_WRITE, true);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_write_byte(cmd, regAddr , true);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_stop(cmd);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_cmd_begin(port, cmd, 1000 / portTICK_RATE_MS);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    i2c_cmd_link_delete(cmd);

    cmd = i2c_cmd_link_create();
    ret = i2c_master_start(cmd);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_write_byte(cmd, devAddr << 1 | I2C_MASTER_READ, true);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_read_byte(cmd, data, 0x1);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_stop(cmd);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    ret = i2c_master_cmd_begin(port, cmd, 1000 / portTICK_RATE_MS);
    if (ret != ESP_OK)
        goto EXIT_FAIL;
    i2c_cmd_link_delete(cmd);
    return true;
EXIT_FAIL:
    ESP_LOGI(DEBUG_MSGS,"Error %s (%d): 0x%x",__FUNCTION__,__LINE__,ret);
    return false;
}

koobest, 2018-06-19:
Hi,@hvishu
Thanks for your feedback. we are trying to find the cause, and we need more information.

1. Did you mean that 'time_out' happened twice in 48 hrs, and it caused a system reset every time?

2. How many slaves in your I2S bus？

3. Is there only one i2c task in your application？

4. Can you kindly provide the boot log when ESP32 is abnormally reset?

By the way,  here is a suggestion for your code ：

````
ret = i2c_master_cmd_begin(port, cmd, 1000 / portTICK_RATE_MS);
if (ret != ESP_OK)
    goto EXIT_FAIL;
i2c_cmd_link_delete(cmd);
````
can write like that:
```
ret = i2c_master_cmd_begin(port, cmd, 1000 / portTICK_RATE_MS);
i2c_cmd_link_delete(cmd);
if (ret != ESP_OK)
    goto EXIT_FAIL;
```

thank you   :-)

hvishu, 2018-06-20:
Did you mean that 'time_out' happened twice in 48 hrs, and it caused a system reset every time?
[hvishu] Yes twice in 48 hrs, in both cases system reset happened

How many slaves in your I2S bus？
[hvishu] We have one i2c slave device (IO expander)

Is there only one i2c task in your application？
[hvishu] i2c read/write are not part of the particular task.  They are like utility functions which are used by different tasks.  There is a possibility that two tasks tried to access the i2c calls simultaneously.  Would that be a problem?  I thought this was taken care in the SDK, would increasing the wait for semaphore  help? 

By the way, here is a suggestion for your code ：
[hvishu]I will change the code as per your suggestion.

Can you kindly provide the boot log when ESP32 is abnormally reset?
[hvishu] Below is the log, primary bootloader prints are disabled by using bootstrap option.
I (10) boot: ESP-IDF v3.0 2nd stage bootloader
I (10) boot: compile time 10:22:23
I (10) boot: Enabling RNG early entropy source...
I (13) boot: SPI Speed      : 40MHz
I (17) boot: SPI Mode       : DIO
I (21) boot: SPI Flash Size : 16MB
I (26) boot: Partition Table:
I (29) boot: ## Label            Usage          Type ST Offset   Length
I (36) boot:  0 phy_init         RF data          01 01 00009000 00001000
I (44) boot:  1 aws_config       unknown          41 00 0000a000 00001000
I (51) boot:  2 aws_cert         unknown          40 00 0000b000 00002000
I (59) boot:  3 tlc_bin          unknown          42 00 0000d000 00002000
I (66) boot:  4 nvs              WiFi data        01 02 0000f000 00009000
I (74) boot:  5 factory          factory app      00 00 00020000 00150000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x4c200 (311808) map
I (204) esp_image: segment 1: paddr=0x0006c228 vaddr=0x3ffc0000 size=0x03de8 ( 15848) load
I (210) esp_image: segment 2: paddr=0x00070018 vaddr=0x400d0018 size=0xde7d4 (911316) map
I (529) esp_image: segment 3: paddr=0x0014e7f4 vaddr=0x3ffc3de8 size=0x00bb4 (  2996) load
I (531) esp_image: segment 4: paddr=0x0014f3b0 vaddr=0x40080000 size=0x00400 (  1024) load
I (537) esp_image: segment 5: paddr=0x0014f7b8 vaddr=0x40080400 size=0x177c0 ( 96192) load
I (585) esp_image: segment 6: paddr=0x00166f80 vaddr=0x400c0000 size=0x00064 (   100) load
I (585) esp_image: segment 7: paddr=0x00166fec vaddr=0x50000000 size=0x00008 (     8) load
I (606) boot: Loaded app from partition at offset 0x20000
I (606) boot: Disabling RNG early entropy source...
I (607) cpu_start: Pro cpu up.
I (610) cpu_start: Starting app cpu, entry point is 0x40081108
I (0) cpu_start: App cpu up.
I (621) heap_init: Initializing. RAM available for dynamic allocation:
I (627) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (633) heap_init: At 3FFD5378 len 0000AC88 (43 KiB): DRAM
I (639) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (646) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (652) heap_init: At 40097BC0 len 00008440 (33 KiB): IRAM

hvishu, 2018-06-25:
@koobest Any updates for me?

koobest, 2018-06-26:
@hvishu 
Sorry for the late reply. I wrote an example to reproduce the problem you described, it run about 20hrs, but no restart occurred (two task, one task read, one task write， the slave is 24c02 ).
 in ` i2c_master_cmd_begin`, we use semaphores to keep in sync, so it should not be caused by simultaneous reading and writing. we will appreciate it if you could provide the code.

igrr, 2018-06-26:
Since in the original code posted by @hvishu there are
two i2c_master_cmd_begin calls, and no mutex around them, is it possible
that two tasks call i2c_master_cmd_begin in wrong order?

I.e. task 1 sends device and register address, task 2 sends device and
register address, task 1 sends device address and reads data, task 2 sends
device address and tries to read data.

There should probably be a mutex to guarantee that two i2c_master_cmd_begin
calls follow each other.

koobest, 2018-06-26:
Yes, @igrr ,
In this order will lead to read and write device error.  maybe we can use a mutex in the application layer to solve this problem, but it should not cause the system to restart .
@hvishu 

Can you try using semaphores to keep the read-write synchronization in your application?

```
read(...)//read function
{
   xSemaphoreTake(sem, portMAX_DELAY);
  ...
  xSemaphoreGive(sem);
}

write(...)//write function
{
   xSemaphoreTake(sem, portMAX_DELAY);
  ...//write
  xSemaphoreGive(sem);
}

```

UncleRus, 2018-07-01:
@hvishu You can use `i2cdev` component from https://github.com/UncleRus/esp-idf-lib

closed 2018-07-12 by FayeY