espressif/esp-idf #2072: can't connect to BLE when WiFi STA connection is active

Milan991, 2018-06-18:
When ESP is in STA mode and connected to my WiFi router (WiFi connection is always working fine), there is an issue with BLE connection. I can see BLE device but I can't connect to it. Every time I try connect I get "ESP_GATTS_DISCONNECT_EVT" log or connection just break without any log (error/warning/info) on ESP serial.
Interesting thing is that when ESP is not connected to WiFi router BLE connection is working great. 

I've already tried solutions from [url]https://www.esp32.com/viewtopic.php?f=13&t=3139[/url] and [url]https://www.esp32.com/viewtopic.php?f=13&t=6008[/url].

With ESP-IDF master branch commit from beginning of May 2018. I didn't have this issue, but with newest one I have.

chegewara, 2018-06-19:
Maybe its not the case, but i lost last few days working on wifi + ble project. Its arduino project: aws IoT + BLE POS printer. I knew that IoT code works good, ble code works good but together it was a crap. I lost 3 days to finally find a solution. In my case ive found that ble issue is caused by this error `BLE_HCI_INSTANT_PASSED`. What my app is doing (before i found solution):
- in loop every 1000ms is sending message on mqtt topic and getting back response on onther topic,
- when message is expected to be true then print on printer,
- issue: ble is disconnecting all the time.
And here is the issue, because when i changed pooling every 1000ms to send request on demand with push button all works fine. Maybe i will try with different pooling time in the future, but for now this works for me.

I wouldnt find this if i wouldnt clone my ble printer onto android device and simulate ble printer with it.
What i am trying to say is: esp32 is sharing radio between wifi and ble, so you need to find proper settings in radio usage by wifi and ble or it wont work together.

chegewara, 2018-06-19:
Now we have questions:
what commit are you using now?
what is the older commit that is working?

Milan991, 2018-06-20:
Commit that I'm using now is: 9a55b42 (Merge branch 'bugfix/tw22828_fix_tcp_bug_when_wifi_disconnect' into '… ).

Older commit: 0d65f3b (Merge branch 'bugfix/light_sleep_fixes' into 'master'). 

Also commit that isn't working: 8955db6 (Merge branch 'bugfix/btdm_fix_no_gatt_server_req_callback_warning' in… ).

In the meantime I've combined "gatt_server" and "simple_wifi" examples in one project, and same problem is present.

andrewvoznytsa, 2018-06-24:
I'd also join this discussion - have exactly the same issues.

If WiFi is active then BLE is not stable (peripheral - ESP32 4b91c82cc447640e5b61407e810f1d6f3eabd233, central iOS 11.4, WiFi/BT 'balanced'). At the same time A2DP works as expected (well - it could be more stable but at least I'm able to get it working in some cases, I'd discuss A2DP/BT classic issues after fixing BLE).

I'm sure that it is not 4b91c82cc447640e5b61407e810f1d6f3eabd233 issue - the same problem happens  with some 4-6 weeks older code.

I'm sure that this is iOS-specific issue - Nordic nRF51 Dongle (see https://www.nordicsemi.com/eng/Products/nRF51-Dongle) and Nordic Master Control Panel (nRF-MCP-x64 https://www.nordicsemi.com/eng/nordic/download_resource/38907/29/67944881/19422 or nRF-MCP-x86 https://www.nordicsemi.com/eng/nordic/download_resource/38909/36/59954738/19421) connects to ESP32 and is able to read characteristics (a bit slower due to radio sharing, maybe, but works).

I'm attaching my ESP32 and iOS applications - as simple as possible. ESP32 is combination of BT/BLE coexist + HTTP request demos. Hope that will help you to replicate and fix problems easily.

[coex.zip](https://github.com/espressif/esp-idf/files/2130939/coex.zip)
[BleScanner.zip](https://github.com/espressif/esp-idf/files/2130940/BleScanner.zip)

Milan991, 2018-06-24:
Thank a lot for sharing all of this. 

My opinion is that it is not iOS-specific issue. I have tested your coex project with ESP32 (9a55b42 ESP-IDF commit). There is no way to connect to BLE when ESP is connected to WiFi, I've tested it with Cypess BLE Dongle http://www.cypress.com/documentation/development-kitsboards/cy5670-cysmart-usb-dongle and with Android 6.0 using nRF Connect application. 

I agree with you, last commit that I know that works fine is 0d65f3b (May 2 2018.). And after that with same code BLE just simply don't work with active WiFi connection.

iosix, 2018-06-24:
I will just add that we successfully use SoftAP mode coincident with BTLE without issues, on a wide range of version of esp-idf, in case that helps narrow down the problem. 

Robert Vogt IV
CEO
IOSiX, LLC
389 Airport Industrial Drive
Ypsilanti, MI  48198
P: +1-855-OBD-1939
C: +1-734-730-9690
robert@iosix.com

UaHeadHunter, 2018-06-25:
Same issue.
BLE is working fine until WiFi connects to the AP.
After that BLE is not discoverable and works only if WiFi disconnects AP.
Tested 10 days ago with latest (for that time) master branch.
Roll back after testing to previous work state.

Milan991, 2018-07-09:
I would like to inform you all that issue is no longer present in newest commit from ESP-IDF master branch, I've tested it and everything is working fine. BLE is now working while WiFi is connected to the AP, without configuration/source code changes in my project.
Maybe this issue is solved in some of previous commits, but I'm sure for commit: a3c4325.

andrewvoznytsa, 2018-07-10:
They 'fixed' that in cost of WiFi throughput - it is slow and unacceptable for media streaming (fdee0b64a0b26cb522df6ceeb171f1930fa84d27). I'm unable to stream 128kbit audio smoothly over HTTP. HTTPS is even worse - throughput drop is just huge.

Milan991, 2018-07-10:
@andrewvoznytsa I've tested it with small WiFi throughput so I wasn't aware of this problem, so this issue will remain open.

mark-legend, 2018-07-30:
Keep an eye on this issue

closed 2018-10-23 by Milan991