espressif/esp-idf #2073: Unknown crash in rwble.c

chegewara, 2018-06-19:
I know this should be on arduino-esp32 repo, but crash is in `rwble.c at line 222', so i decided to post it here:
```
ASSERT_PARAM(512 0), in rwble.c at line 222
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40085344  PS      : 0x00060034  A0      : 0x80088116  A1      : 0x3ffc05b0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x80085341  A9      : 0x3ffc0590  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40085279  LEND    : 0x40085280  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4015bb8e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40085344

Backtrace: 0x40085344:0x3ffc05b0 0x40088113:0x3ffc05d0 0x40088a17:0x3ffc05f0 0x400816a5:0x3ffc0610 0x4015bb8b:0x00000000
```
```
0x40085344: r_assert_param at ?? line ?
0x40088113: r_rwble_isr at ?? line ?
0x40088a17: r_rwbtdm_isr_wrapper at intc.c line ?
0x400816a5: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x4015bb8b: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85

```

```
Core 1 register dump:
PC      : 0x4015bb8e  PS      : 0x00060934  A0      : 0x8008fc74  A1      : 0x3ffd6460  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ffd5f94  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc7274  A9      : 0x3ffc7238  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x3ffd6120  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4015bb8e:0x3ffd6460 0x4008fc71:0x3ffd6480
```
```
0x4015bb8e: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85
0x4008fc71: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3529
```

I just realized this is caused by WDT and if this is my app fault then feel free to close it. 
In addition:
- app has been working for some time (over an hour i think),
- all my loops have at least 100ms delays.

projectgus, 2018-06-19:
This looks like the same root cause as #2070.

chegewara, 2018-06-19:
I didnt see it, sorry. I will close this issue then.

closed 2018-06-19 by chegewara