espressif/esp-idf #2074: wifi sokcet works not well

NOA-Vernon, 2018-06-19:
Hello,

I ran the demo esp-idf\examples\protocols\http_request to my esp32 board, and found that the socket work not well and unstable. Below capture is the communication details, the source ip 192.168.8.3.4 is from esp32, destination ip 192.168.8.3 is from my PC(the http server). Is there anyone who has experienced this problem? Any suggestions  is appreciated.
Thanks a lot

 

[image]

liuzfesp, 2018-06-27:
@NOA-Vernon, I guess you captured the packets on PC's ethernet port, if I'm right, above captured packets only indicate that the packets 11~14 is successfully sent from PC ethernet port to AP, but they don't mean the AP has forward the packets to esp32 successfully. Could you help to capture the WiFi packets between AP to ESP32?  Then I can analyze them.

NOA-Vernon, 2018-06-28:
@liuzfesp  thank you for your information, do you have any testing tools or suggestion for capturing the WIFI packets between AP to ESP32?

zhangyanjiaoesp, 2018-07-02:
@NOA-Vernon you can use OmniPeek to capture the WiFi packets

NOA-Vernon, 2018-07-04:
@zhangyanjiaoesp Thank you for you reply, it seems OmniPeek is not a free capture packet software tool.

liuzfesp, 2018-07-17:
@NOA-Vernon could you help to provide the captured packets?

NOA-Vernon, 2018-07-18:
@liuzfesp Thank you for you reply, I updated to the newest IDF and not happened again by now, it might be solved in the new IDF. If it happened again, I will captured the packets and publish again.

closed 2018-07-24 by FayeY