espressif/esp-idf #2076: gatt_security_server Error BT_BTM:Device not found

redengin, 2018-06-19:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v1
- Core: ESP-WROOM32
- IDF version: 9a55b42 
- Development Env: Make
- Operating System: Ubuntu
- Power Supply: USB
- Android device: Fire(5th Generation) FireOS 5.6.1.0
- Android App: https://github.com/googlesamples/android-BluetoothLeGatt 

## Problem Description
Android application connection to the ESP32 gatt_security_example is lost.

### Expected Behavior
The gatt_security_example should stay connected.

### Actual Behavior
The gatt_security_example disconnects.
```
I (153) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (463) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (463) SEC_GATTS_DEMO: app_main init bluetooth
I (503) SEC_GATTS_DEMO: The number handle = 8
I (503) SEC_GATTS_DEMO: advertising start success
I (4343) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
W (34283) BT_L2CAP: L2CA_EnableUpdateBleConnParams - unknown BD_ADDR 70075f18fa48
W (34293) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 70075f18fa48 not connected
E (34293) BT_BTM: Device not found

I (34303) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
I (34313) SEC_GATTS_DEMO: remote BD_ADDR: 70075f18fa48
I (34313) SEC_GATTS_DEMO: address type = 1
I (34313) SEC_GATTS_DEMO: pair status = fail
I (34323) SEC_GATTS_DEMO: Bonded devices number : 0

I (34323) SEC_GATTS_DEMO: Bonded devices list : 0

I (34333) SEC_GATTS_DEMO: advertising start success
```

### Steps to reproduce
1. start the ESP32 gatt_security_example
```
make flash monitor
```
2. start the Android BLE GATT application
3. Observe connection
  * ESP32 reports ESP_GATTS_CONNECT_EVT
  * Android BLE GATT application reports connection/services/characteristics
4. Wait (approximately 30 seconds)
5. Observe the connection loss
  * ESP32 reports error (see above)
  * Android BLE GATT application reports disconnected.

redengin, 2018-07-03:
attaching logs with CONFIG_BTM_TRACE_LEVEL_EVENT=y
```
I (7764) BT_L2CAP: L2CA_ConnectFixedChnl() CID: 0x0006  BDA: 5fc799280db7
D (7774) BT_L2CAP: CID:0x0004 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
D (7784) BT_L2CAP: CID:0x0041 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Quota:200
I (7784) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 5fc799280db7
I (7794) BT_L2CAP: L2CA_EnableUpdateBleConnParams - BD_ADDR 5fc799280db7 enable 0 current upd state 0x08
I (7844) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (7994) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8104) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
D (8104) BT_BTM: btm_ble_read_remote_features_complete 
I (8114) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8134) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8144) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8164) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8174) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8194) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8204) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8224) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8234) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
I (8254) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 5fc799280db7
W (37164) BT_L2CAP: L2CA_EnableUpdateBleConnParams - unknown BD_ADDR 5fc799280db7
W (37164) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 5fc799280db7 not connected
I (37164) BT_BTM: BTM_InqDbRead: bd addr [5fc799280db7]

I (37174) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 5fc799280db7

E (37184) BT_BTM: Device not found
```

Weijian-Espressif, 2018-07-04:
hi @redengin, **l2cu_find_lcb_by_bd_addr** will return NULL in **L2CA_EnableUpdateBleConnParams** when Bluetooth is disconnected.  so it will printf "unknown BD_ADDR XXX " and "Device not found".  Bluetooth may be  disconnected due to a pairing failure.

Weijian-Espressif, 2018-07-04:
Does this problem happen every connection?

redengin, 2018-07-10:
Yes, this happens for every connection.

Weijian-Espressif, 2018-07-10:
Sorry, we do not have the Fire(5th Generation), Can you help me to do some test ? please test our  gatt_server demo  to check BLE connection without SMP.

redengin, 2018-07-12:
The gatt_security_server will not compile with the SMP module disabled.  The gatt_server example does compile with SMP disabled and it does not have the issue (i.e. the bluetooth connection is solid).

Unfortunately, for my use-case, I want to be able to pair the ESP32 which requires the Bluetooth security module (SMP).

Weijian-Espressif, 2018-07-16:
@redengin I mean to test if a non-encrypted connection can be established.

redengin, 2018-07-16:
@Weijian-Espressif  yes, non-encrypted (example gatt-server) works just fine.

Weijian-Espressif, 2018-07-16:
please open SMP log in menuconfig , and provide me the log, thanks.
[image]

redengin, 2018-07-16:
Here is the log with SMP:DEBUG
```
I (0) cpu_start: App cpu up.
I (404) heap_init: Initializing. RAM available for dynamic allocation:
I (411) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (417) heap_init: At 3FFCCD58 len 000132A8 (76 KiB): DRAM
I (423) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (429) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (436) heap_init: At 4009133C len 0000ECC4 (59 KiB): IRAM
I (442) cpu_start: Pro cpu start user code
I (124) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (171) BTDM_INIT: BT controller compile version [9525061]

I (171) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (481) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (481) SEC_GATTS_DEMO: app_main init bluetooth
D (481) BT_SMP: SMP_Init
D (481) BT_SMP: SMDBG l2c smp_l2cap_if_init
D (511) BT_SMP: SMP_Register state=0
D (521) BT_SMP: smp_encrypt_data

I (521) SEC_GATTS_DEMO: The number handle = 8
I (521) SEC_GATTS_DEMO: advertising start success
D (18841) BT_SMP: SMDBG l2c smp_connect_callback

I (18841) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
D (18851) BT_SMP: SMP_Pair state=0 br_state=0 flag=0x0 

D (18851) BT_SMP: SMDBG l2c smp_connect_callback

D (18851) BT_SMP: smp_connect_callback()  for pairing BDA: 400e1df2e29d  Event: connected

D (18861) BT_SMP: main smp_sm_event

D (18861) BT_SMP: SMP Role: Slave State: [SMP_STATE_IDLE (0)], Event: [L2CAP_CONN_EVT (18)]
D (18871) BT_SMP: State change: SMP_STATE_IDLE(0) ==> SMP_STATE_WAIT_APP_RSP(1)
D (18881) BT_SMP: smp_send_app_cback p_cb->cb_evt=1

D (18891) BT_SMP: io_cap = 4
D (18891) BT_SMP: callback_rc=0  p_cb->cb_evt=1

D (18891) BT_SMP: rcvd auth_req: 0x01, io_cap: 3                         loc_oob_flag: 0 loc_enc_size: 16,local_i_key: 0x03, local_r_key: 0x03

D (18911) BT_SMP: set auth_req: 0x01, local_i_key: 0x03, local_r_key: 0x03

D (18911) BT_SMP: main smp_sm_event

D (18921) BT_SMP: SMP Role: Slave State: [SMP_STATE_WAIT_APP_RSP (1)], Event: [API_IO_RSP_EVT (20)]
D (18931) BT_SMP: State change: SMP_STATE_WAIT_APP_RSP(1) ==> SMP_STATE_PAIR_REQ_RSP(3)
D (18931) BT_SMP: smp_process_io_response

D (18941) BT_SMP: State change: SMP_STATE_PAIR_REQ_RSP(3) ==> SMP_STATE_SEC_REQ_PENDING(2)
D (18951) BT_SMP: smp_send_cmd on l2cap cmd_code=0xb

D (18951) BT_SMP: smp_build_security_request

D (18961) BT_SMP: opcode=11 auth_req=0x1
D (18961) BT_SMP: smp_send_msg_to_L2CAP
D (18971) BT_SMP: result state = SMP_STATE_SEC_REQ_PENDING

D (18971) BT_SMP: smp_send_app_cback return

D (18971) BT_SMP: result state = SMP_STATE_SEC_REQ_PENDING

D (48281) BT_SMP: SMDBG l2c smp_connect_callback

D (48281) BT_SMP: smp_connect_callback()  for pairing BDA: 400e1df2e29d  Event: disconnected

D (48281) BT_SMP: main smp_sm_event

D (48281) BT_SMP: SMP Role: Slave State: [SMP_STATE_SEC_REQ_PENDING (2)], Event: [L2CAP_DISCONN_EVT (19)]
D (48291) BT_SMP: State change: SMP_STATE_SEC_REQ_PENDING(2) ==> SMP_STATE_IDLE(0)
D (48301) BT_SMP: smp_pair_terminate

D (48301) BT_SMP: smp_proc_pairing_cmpl 

D (48311) BT_SMP: send SMP_COMPLT_EVT reason=0x19 sec_level=0x0

W (48311) BT_L2CAP: L2CA_EnableUpdateBleConnParams - unknown BD_ADDR 400e1df2e29d
D (48321) BT_SMP: smp_reset_control_value

D (48331) BT_SMP: smp_remove_fixed_channel

W (48331) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 400e1df2e29d not connected
D (48341) BT_SMP: smp_cb_cleanup

D (48341) BT_SMP: result state = SMP_STATE_IDLE

E (48351) BT_BTM: Device not found

W (48351) BT_APPL: BTA got unregistered event id 31

W (48361) BT_APPL: BTA got unregistered event id 31

I (48361) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
I (48381) SEC_GATTS_DEMO: remote BD_ADDR: 400e1df2e29d
I (48381) SEC_GATTS_DEMO: address type = 1
I (48381) SEC_GATTS_DEMO: pair status = fail
I (48391) SEC_GATTS_DEMO: Bonded devices number : 0

I (48391) SEC_GATTS_DEMO: Bonded devices list : 0

I (48401) SEC_GATTS_DEMO: advertising start success
```

redengin, 2018-07-16:
@Weijian-Espressif please provide a link to the 'NRF connect APP'

(if you are on freenode I can also interact with you there)

Weijian-Espressif, 2018-07-16:
https://github.com/NordicSemiconductor/Android-nRF-Connect

redengin, 2018-07-17:
I get the same result using the Android-nRF-Connect application (connection is lost after ~ 40s)

Weijian-Espressif, 2018-07-20:
sorry for the late reply. 
[nrfconnect_downcc.apk.zip](https://github.com/espressif/esp-idf/files/2212313/nrfconnect_downcc.apk.zip)

 Is there a pairing box popping up during the connection?

redengin, 2018-07-28:
I've tested with both the ESP32 paired and unpaired using the Gatt Security Server example, no pairing pop-up ever appears.  To pair, I manually went to bluetooth settings and paired with the ESP32 running the Gatt Secrurity Server example.

Weijian-Espressif, 2018-07-30:
@redengin  it is wrong to manually went to bluetooth settings and paired with the ESP32.
Some android phones do not have a pairing pop-up, there will be a system notification, click the notification will pop up the paired box, if the phone does not have any respond, indicating that your phone has BLE encryption issues,  you can used other mobile phone to try .

FayeY, 2018-08-06:
Any update about this issue? Is it solved now?

redengin, 2018-08-25:
I've updated to the latest ESP-IDF and gatt_security_server example and still get the same disconnects.  At this point I'm going to concentrate on implementation of a gatt server and debug from that perspective.

I used the default project configuration.  Let me know if you'd like me to retest with changes to the configuration.

Weijian-Espressif, 2018-09-07:
@redengin ,  from the log, ESP32 have sended security_request, but your device don't send response. that is the point, I don't have Fire(5th Generation), I don't know what happened in your device. please capture the BLE packets with a BLE sniffer.

closed 2018-09-25 by FayeY

redengin, 2018-10-01:
Understood.  If the issue persists I'll make another ticket with BLE sniffer recordings.  This is still the most responsive hardware groups I've worked with and I'm very happy.