espressif/esp-idf #2078: How do i do https_request with self signed certificates

Kasuhikdas, 2018-06-19:
I have created python https server on 443 port. 
I have attached ca certificates in the https_request example. 
But it is not connecting  to it. 
Error is
I (3109) event: sta ip: 192.168.43.114, mask: 255.255.255.0, gw: 192.168.43.1
I (3109) example: Connected to AP
E (20639) esp-tls: Failed to connnect to host (errno 113)
E (20639) example: Connection failed...
this is the error. 

I have traced the packets. There is no client hello sent to server. 
51998	4257.563786986	192.168.43.114	192.168.43.187	TCP	58	60978 → 443 [SYN] Seq=0 Win=5744 Len=0 MSS=1436

52001	4260.430663109	192.168.43.114	192.168.43.187	TCP	58	[TCP Retransmission] 60978 → 443 [SYN] Seq=0 Win=5744 Len=0 MSS=1436

Second message continues for three or four times then serial generates error.

negativekelvin, 2018-06-19:
Are you sure there is no firewall? Where is ACK packet?

projectgus, 2018-06-20:
Hi @Kasuhikdas ,

As @negativekelvin points out, the TCP connection is failing to establish. Without an underlying TCP connection no TLS data can be exchanged. If you have openssl, you can test the connection yourself with

> openssl s_client -connect 192.168.43.114:443

If you get a TCP connection error, it's the same problem. If you get a certificate error, the TCP connection is succeeding but it's due to the self-signed certificate (you can add arguments to s_client to pass in the same cert file, but at this point you can probably go back to testing on the ESP32 as well.)

Kasuhikdas, 2018-06-20:
Certificate is working now. Issue was with server. But its not reading data. I want to do OTA update using https. OTA working with HTTP. 
Https is not reading data.
Any idea how to integrate both will be helpful.
Thanks for the help.

closed 2018-06-21 by Kasuhikdas