espressif/esp-idf #2080: Unable to flash SPIFFS

thisboyiscrazy, 2018-06-20:
## Environment

- Development Kit:      ESP-32S
- Kit version (for WroverKit/PicoKit/DevKitC): v1.1
- Core (if using chip or module): ESP32-WROOM-32U
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  9a55b42f
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         USB

## Problem Description

Can not make SPIFFS image

[http://esp-idf.readthedocs.io/en/latest/api-reference/storage/spiffs.html?highlight=mkspiffs](http://esp-idf.readthedocs.io/en/latest/api-reference/storage/spiffs.html?highlight=mkspiffs)

States you can make a SPIFFS image with [https://github.com/igrr/mkspiffs](https://github.com/igrr/mkspiffs) and flash it by doing:

```
mkspiffs -c [src_folder] -b 4096 -p 256 -s 0x100000 spiffs.bin
python esptool.py --chip esp32 --port [port] --baud [baud] write_flash -z 0x110000 spiffs.bin
```
I modified the example spiffs progam to read a file I put in the image and it reads ''.

Also if I do a read_flash of the partition area and do ```mkspiffs -u [dest] spiffs.bin``` I get files but not with the correct bytes.

### Expected Behavior

Able to create an image in development computer and read the files from the ESP-32 and dump the image from the ESP-32 and read the files from the development computer.

### Actual Behavior

Files created on the ESP-32 can be read on the ESP-32, files created on the development computer can be read on the development computer but not read file created on the other.

### Steps to repropduce

1. create a spiffs image using mkspiffs upload to spiffs partition (like in examples/storage/spiffs)
2. write image to flash
2. modify example program to read a file in image created.

```
I (284) example: Initializing SPIFFS
I (374) example: Partition size: total: 956561, used: 1506
I (374) example: Opening file
I (444) example: File written
I (454) example: Renaming file
I (484) example: Reading file
I (524) example: Read from file: ''
I (524) example: SPIFFS unmounted
```

The file contains ```ABCDEFGHIJKLMNOPQRSTUVWXYZ```

igrr, 2018-06-20:
Hi @thisboyiscrazy, please attach your `sdkconfig` file, and the output of `mkspiffs --version`.

thisboyiscrazy, 2018-06-20:
```
mkspiffs ver. 0.2.3-1-g011c110
Build configuration name: generic
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: (none)
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 0
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0
```

thisboyiscrazy, 2018-06-20:
```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=

#
# Security features
#
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
CONFIG_ESPTOOLPY_AFTER_NORESET=
CONFIG_ESPTOOLPY_AFTER="hard_reset"
CONFIG_MONITOR_BAUD_9600B=
[... 469 lines cut ...]
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

igrr, 2018-06-20:
There is a mismatch in the metadata length between your copy of `mkspiffs` and your sdkconfig:

```
  SPIFFS_OBJ_META_LEN: 0
```

```
CONFIG_SPIFFS_META_LENGTH=4
```

To resolve the issue you need to do either of:

- Download a copy of `mkspiffs` which is built with META_LEN=4 (https://github.com/igrr/mkspiffs/releases, look for `mkspiffs-x.y.z-esp-idf` archive for your platform)

- Build `mkspiffs` with META_LEN=4, see https://github.com/igrr/mkspiffs#spiffs-configuration

- Change `CONFIG_SPIFFS_META_LENGTH` to 0 in menuconfig. Note that you will need to disable timestamp support (CONFIG_SPIFFS_USE_MTIME) as well.

thisboyiscrazy, 2018-06-20:
Thanks, I went with the last option and it works now. It seems the documentation should be a little more clear about that.

closed 2018-06-20 by thisboyiscrazy

hawres, 2018-07-06:
@igrr Hi, i had very similar problem.
I generated filesystem using  command: 
`./mkspiffs -c files/ -b 4096 -p 256 -s 0x100000 spiffs.bin`
my partition csv is:

    nvs,      data, nvs,     0x9000,  0x6000,
    phy_init, data, phy,     0xf000,  0x1000,
    factory,  app,  factory, 0x10000, 1M,
    storage,  data, spiffs,  ,        2M

my partition from flashing message:
```c
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot:  3 storage          Unknown data     01 82 00110000 00200000
```
i flashed with command
`python2 /home/grzegorz/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 write_flash -z 0x110000 spriffs.bin`
and everything looks ok, but when i tried to get my prepared file 
```c
void test_partition()
{
    size_t total = 0, used = 0;
    uint8_t ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK) {
        ESP_LOGE("INFO", "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
    } else {
        ESP_LOGI("INFO", "Partition size: total: %d, used: %d", total, used);
    }

    ESP_LOGI("INFO", "Reading file");
    FILE* f = fopen("/spiffs/TrainingData.txt", "r");
    if (f == NULL) {
        ESP_LOGE("INFO", "Failed to open file for reading");
        return;
    }
```
i got 
```c
I (0) cpu_start: Starting scheduler on APP CPU.
I (817) INFO: Partition size: total: 1920401, used: 0
I (817) INFO: Reading file
E (887) INFO: Failed to open file for reading
```

i checked my configurations for mkspiffs 
```
./mkspiffs --version
mkspiffs ver. 0.2.3
Build configuration name: esp-idf
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: -DSPIFFS_OBJ_META_LEN=4
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 4
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0
```
and sdkconfig
```
#
# SPIFFS Cache Configuration
#
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
CONFIG_SPIFFS_CACHE_STATS=
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=y
```
as I understand, SPIFFS_OBJ_META_LEN = CONFIG_SPIFFS_META_LENGTH and its correct. 

I can create new files from my  c code, write and read to and from it, but i cannot see my prepared file from mkspiffs. 
What I am doing wrong? Could You help me, please?
Im using ESP-WROOM-32.

igrr, 2018-07-06:
There seems to be a mismatch in partition size: SPIFFS partition size is
2MB, but you pass -s 0x100000 to mkspiffs?

hawres, 2018-07-09:
@igrr yes, it was it! i don't know how i didn't find it... thanks!