espressif/esp-idf #2081: no output from make monitor, but platformio's monitor works.

ritualz, 2018-06-20:
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [dfrobot firebeetle esp32]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version: 9a55b42f
- Development Env:		[Make]
- Operating System:		[Linux]
- Power Supply:         [USB | Battery]

## Problem Description
I just got a new board, which connect to /dev/ttyACMx, in the past I've only used esp32 modules that uses ttyUSBx.  Uploading and flashing works fine, but when trying to monitor with `make monitor` the only output is:
`
MONITOR
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
` 
This didn't happen when I'm using the other boards. I'm has chmod the device and are in the dialout group. The strange thing is that the platformio monitor works, using the same baudrate. 

I've added the relevant section of the sdkconfig file.  
`
CONFIG_ESPTOOLPY_PORT="/dev/ttyACM0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_RESET=y
CONFIG_ESPTOOLPY_AFTER_NORESET=
CONFIG_ESPTOOLPY_AFTER="hard_reset"
CONFIG_MONITOR_BAUD_9600B=
CONFIG_MONITOR_BAUD_57600B=
CONFIG_MONITOR_BAUD_115200B=y
CONFIG_MONITOR_BAUD_230400B=
CONFIG_MONITOR_BAUD_921600B=
CONFIG_MONITOR_BAUD_2MB=
CONFIG_MONITOR_BAUD_OTHER=
CONFIG_MONITOR_BAUD_OTHER_VAL=115200
CONFIG_MONITOR_BAUD=115200
`

igrr, 2018-06-22:
I don't think we have a dfrobot firebeetle board here to check this, but my guess is that idf_monitor sets both DTR and RTS low (active), which might cause the USB MCU on the firebeetle board to keep ESP32 in reset. If you have some terminal program which allows controlling the state of DTR and RTS separately, please try different combinations (DTR/RTS on/off) and see which one allows the ESP32 to run. Then perhaps a configuration option for idf_monitor can be added to set such state of DTR and RTS.

ritualz, 2018-06-22:
The esp module is active and not resetting (basically displaying the boot-count on a external display).  I tried the different combinations with pio (platformIOs monitor):

- DRT off / RST off -> **no output**
- DRT on/ RST off -> **output**
- DRT off / RST on ->  **no output**
- DRT on/ RST on -> **crash**

igrr, 2018-06-25:
DTR off / RST off -> no output
DTR on/ RST off -> output
DTR off / RST on -> no output
DTR on/ RST on -> crash

projectgus, 2018-06-25:
I found a [schematic for the Firebeetle](https://raw.githubusercontent.com/Robert-MARKII/Document/master/%5BDFR0478%5DFireBeetle%20Board-ESP32(V1.0).pdf) and see that it's a STM32F042 with two GPIOs wired directly to GPIO0 and EN.

There are some soft USB/serial firmwares around which use DTR as the "enable serial" signal, so my guess is that's also what is happening here.

Out of interest, is esptool.py able to flash this board successfully? Does it auto-reset the board into flashing more successfully? Reason I ask is that DTR & RTS should both also be de-asserted (set False) during flashing, so I can't explain why this would work in esptool but fail in idf_monitor.

> DTR on/ RST on -> crash

What crashes, the ESP32 or idf_monitor.py? (or both!)

Angus

ritualz, 2018-06-25:
@projectgus 
I can flash the board using esptool.py, and it does reset the board after completion.
Regarding the crash, it seemed like it was the platformIO's (the monitored i used to test the options), which was at fault. When trying with both DTR and RTS option enabled, idf_monitor.py operate normally. So it seems like the board only need the DTR to be pulled up for it to work.

projectgus, 2019-03-01:
Sorry for the lack of update in so long.

I'm going to Close as this is a problem with the fireBeetle CDC firmware behaviour, not ESP-IDF. Please raise the issue with DFRobot.

closed 2019-03-01 by projectgus

Micrified, 2019-09-20:
I am experiencing this issue as well, but I am only using `idf.py`. I have disabled RST and DTR as recommended and implemented the changes found in this [pull request](https://github.com/espressif/esp-idf/pull/2090/files) by another user experiencing the problem. The fix worked one-time and then I am again faced with zero output from the monitor.

AHafsy, 2019-12-12:
Hello,
I recently started working on ESP-LAUNCHER (ESP8266) and I have the same porblem described before. I tried to solve it using the recommended solutions, but unfortunately I still have no output when running idf.py monitor. 
Do anyone have an idea about this error ?