espressif/esp-idf #2082: ledc_set_fade_with_time() does not fade from min..max when executed immediately after updating the duty cycle to 0

pantaluna, 2018-06-20:
## Environment
- Development Kit:      [Adafruit HUZZAH32]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (commit id): 4b91c82c Jun19,2018
- Development Env:      [Eclipse]
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
ledc_set_fade_with_time() does not fade from duty 0..255 when it is executed immediately after setting and updating the duty cycle to 0 using ledc_set_duty() and ledc_update_duty().

### Expected Behavior
ledc_set_fade_with_time() does fade from duty 0..255 in 5 seconds.

### Actual Behavior
ledc_set_fade_with_time() does not fade and ends immediately.

### Workaround
Adding a delay of 1 second after setting and updating the duty cycle to 0 using ledc_set_duty() and ledc_update_duty(), and before ledc_set_fade_with_time().

### Steps to repropduce
No wiring needed.

```
cd ~
git clone --recursive https://github.com/pantaluna/support_esp_espidf.git
cd  support_esp_espidf
make flash monitor
```

### Code to reproduce this issue
https://github.com/pantaluna/support_esp_espidf

## Debug Logs
Note that the first two ledc_set_fade_with_time() section does nothing (it ends immediately).

```
D (424) myapp: app_main()
I (424) myapp: === LEDC: ledc_timer_config(LEDC_TIMER_8_BIT) ledc_channel_config() freq=60 | duty = 255
D (434) ledc: LEDC_PWM CHANNEL 0|GPIO 14|Duty 0255|Time 0
V (444) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (444) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (444) intr_alloc: Connected src 43 to int 13 (cpu 0)
I (464) myapp: === freq=60 | vary duty 0..255 using ledc_set_fade_with_time(5 seconds)
I (474) myapp:   Set duty := 0
I (474) myapp:   ***PROBLEM ledc_set_fade_with_time() does not fade. It starts at duty 255 instead of duty 0 and so e
nds right away
D (484) ledc: Set to target duty: 255
I (504) myapp: === freq=60 | vary duty 0..255 using ledc_set_fade_with_time(5 seconds)
I (504) myapp:   Set duty := 0
I (504) myapp:   ***PROBLEM ledc_set_fade_with_time() does not fade. It starts at duty 255 instead of duty 0 and so e
nds right away
D (514) ledc: Set to target duty: 255
I (534) myapp: === freq=60 | vary duty 0..255 using ledc_set_fade_with_time(5 seconds)
I (534) myapp:   Set duty := 0
I (534) myapp:   ***OK when using a delay of 1 second after updating duty...***
I (544) myapp:    FIX: wait 1 second after updating duty...
D (1554) ledc: cur duty: 0; target: 255, step: 255, cycle: 1; scale: 1; dir: 1

I (5804) myapp: === freq=60 | vary duty 0..255 using ledc_set_fade_with_time(5 seconds)
I (5804) myapp:   Set duty := 0
I (5804) myapp:   ***OK when using a delay of 1 second after updating duty...***
I (5814) myapp:    FIX: wait 1 second after updating duty...
D (6814) ledc: cur duty: 0; target: 255, step: 255, cycle: 1; scale: 1; dir: 1
```

pantaluna, 2018-06-30:
Thanks.
Yes, the example already shows that you need to wait for some time to update the hardware.

Questions:
1. How long exactly must it wait after changing the duty value?
2. We already have to call the func ledc_update_duty() explicitly to activate the change of the duty cycle. I wonder if the logic of waiting for the hardware to update can be implemented (nicely hidden as such) in that func ledc_update_duty(). That would be a perfect implementation. Alternatively a flag can be set in an event group to indicate that the hardware has updated correctly and the calling program can proceed. As it is now, the full responsibility is moved to the calling program; it needs to guess how long it must wait after changing the duty cycle; and surely often forget to do so as it is not imposed or reported somehow that the calling program forgot to add a delay after it?

What do you think of this proposal?

costaud, 2018-07-09:
We will modify the function of ledc_set_duty_and_update which will return after the duty is actually updated. The code will be merged soon.

costaud, 2018-07-25:
The fix should be merged since 8809fab. @pantaluna

closed 2018-07-28 by igrr

pantaluna, 2018-07-28:
The retest failed using the project https://github.com/pantaluna/support_esp_espidf

ledc_set_fade_with_time() still does not fade from duty 0..255 when it is executed immediately after setting and updating the duty cycle to 0 using ledc_set_duty() and ledc_update_duty().