espressif/esp-idf #2084: https_request is not reading response from localhost website.

Kasuhikdas, 2018-06-21:
- Development Kit:ESP32-Wrover-Kit
- Development Env:Make
- Operating System:Ubuntu

I am trying to read response from a local web page. But after connecting it is not reading the data. 
But it reads data from https://www.howsmyssl.com/a/check this. Both headers are same of these pages. 
`I (862694) example: Connected to AP
I (863514) example: Connection established...
I (863514) example: 88 bytes written
I (863514) example: Reading HTTP response...
I (983584) example: connection closed
`
I am using example program https_request in protocols. 
Why this is happening?

negativekelvin, 2018-06-21:
Is the URL valid? What does the webserver log say?

Kasuhikdas, 2018-06-21:
URL is valid. Showing the packet logs
[image]

negativekelvin, 2018-06-21:
Don't see `I (863514) example: 88 bytes written` in wireshark

Kasuhikdas, 2018-06-21:
then what might be the problem behind this? I have not changed any code from example.

Kasuhikdas, 2018-06-21:
BTW i am using self signed certificate. I tried to access the page from another machine.I am getting security warning for certificate and then i have to click on proceed and then only server log is coming.  
So,is it not accessing because of self signed certificate? the same certificate is working with mqtts.

Kasuhikdas, 2018-06-22:
The problem was the default request it uses http 1.0 it has to be 1.1 to work.

closed 2018-06-22 by Kasuhikdas