espressif/esp-idf #2085: spi slave use dma，lost last 4 bytes

BigSmileFace, 2018-06-21:
In spi slave mode，if i do not use dma,just set`spi_slave_initialize(spi, &buscfg, &slvcfg, 0);`My code works well.
And if i use dma:`spi_slave_initialize(spi, &buscfg, &slvcfg, 1);` esp will lost last 4 bytes.

i found that:
`Warning: Due to a design peculiarity in the ESP32, if the amount of bytes sent by the master or the length of the transmission queues in the slave driver, in bytes, is not both larger than eight and dividable by four, the SPI hardware can fail to write the last one to seven bytes to the receive buffer.`
So I send 12,16 bytes each time,

For 12 bytes,esp get 8 bytes  `a5,a5,a5,a5,a5,a5,a5,a5,0,0,0,0:`
For 16 bytes,esp get 12 bytes  `a5,a5,a5,a5,a5,a5,a5,a5,a5,a5,a5,a5,0,0,0,0:`
I use the latest idf.
So what shall i do with the setting of spi module?

ginkgm, 2018-06-27:
hi @BigSmileFace ,
I wonder whether the timing of SPI bus does not meet the requirement of ESP32 slave. Could you please provide the following to let us help you?

1. your IDF commit
2. your bus LA capture
3. frequency of SCLK
4. your working mode
5. your code for testing
6. did you modify the slave driver code?

The issue may be:
1. currently we have issues with DMA in mode 0/2, which is being solved
2. there should be a hold time between the last clock edge and the CS inactive edge, or the last bytes will not be written into the memory.

ginkgm, 2018-06-27:
To check whether is from reason 1, just config both your master and slave to mode 1

trancefreak77, 2018-09-25:
Is there any outcome of the investigations regarding this issue? I face the same problems when using DMA ans SPI slave driver. I use the arduino-esp32 module.

The last 4 bytes are always missing. Frequency doesn't matter. I've tried all modes and the problem is in all modes. I checked the output of the SPI master (raspberry pi 3) with a logic anylizer and the master definitely sends all bytes. The signals also look nice and clean.

For now, I use a workaround and send 4 dummy bytes to make sure the slave (esp32) does get all needed bytes but I'm wondering if someone already nailed down the problem.

trancefreak77, 2019-01-10:
Sorry, I have no solution. I still send 4 dummy bytes to make sure the ESP does get all bytes.

squonk11, 2019-01-10:
Unfortunately I don't have any possibility to modify the data coming from the master. The master sends data of arbitrary length. Do you have any idea how to workaround this problem?

trancefreak77, 2019-01-13:
Sorry, I don't have an idea how to solve this issue. I think Espressif needs to fix this.

ghazel, 2019-01-16:
I also encountered this issue.

Bogdancev, 2019-07-12:
I also have this problem. Fortunately, I can repeat last 4 bytes as I control the master.

ginkgm, 2020-07-15:
Hello @BigSmileFace @trancefreak77 @squonk11 @ghazel @Bogdancev ,
our SPI slave DMA has limitation that it can only receive in multiples of 4 bytes, see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#restrictions-and-known-issues.

We also have tested for it, and it seems working well when receiving 4 bytes only. Could you try adding some delay between the cs active edge and the first clock edge, and between the last clock edge and the cs inactive edge on the master side (about 3 SPI clock will be totally enough), and run SPI slave *on the latest master branch*? .

If this works, but the same issue happen on earlier versions, we can backport some patches to no earlier than 3.3 for you.

JosuGZ, 2021-01-01:
I'm experiencing a similar issue, but I'm losing the last byte. If I try to read 256 bytes, I get 255, if I try to read 257, I get 256.

It does not always happen, though. I also have two SPI devices on the same bus, and only one of them is doing this.

The other one only misbehaves when it's being used on an HTTP handler; for storing data on an external SPI flash. If I try to test it on, for example, "main()", it works properly.

niklasekstrom, 2021-09-09:
I see this problem as well. The problem being: using ESP32 as a SPI slave, in DMA mode, and loosing the last 4 bytes.

I have an ESP32-WROOM-32D module, and have tried with ESP-IDF 4.2 and 4.4 (the master branch at this time). I have tried different SPI frequencies, and the problem is the same. The time until CS goes high must certainly be long enough.

@ginkgm Can you explain the setup for the tests in which you say that you do not see this problem? Can you try with the same ESP module I'm using? I would think that after this many bug reports saying the same thing you can conclude the the problem is real, and that you have to try harder to replicate the problem.

darioia, 2023-05-24:
Hi, 
   I have implemented a SPI master (TX) and spi Slave (RX) multithread application and send 4124 bytes in a frame (in 4 subframe for memory problem), the latest byte (not word) was 0 for every subframe. 
Yesterday I have seen that if the t.lenght field was len*9 bit (instead of 8 in TX) the frame was sent and received correctly. Obviously the frame was incompatible with the requirements. But why works if 9 bit for byte ?

pgordinho, 2024-03-05:
Any news on this? This is a real deal breaker for me. I don't have access to the SPI master and I really need those 4 bytes.

squonk11, 2024-03-05:
I did not follow this for a long time. Maybe the newer versions of the ESP32 have fixed this HW bug?
I solved this by polling the port pins by SW. This was possible because the baudrate of my master was low enough.

wanckl, 2024-03-07:
@pgordinho Use later chips for example esp32s3 don't have this issue

cawald3, 2024-05-31:
@wanckl do you have a source on the fact that the esp32-s3 doesn't have this bug? The documentation for ESP-IDF still lists this restriction but the technical manual for the s3 doesn't mention the 4 byte issue.