espressif/esp-idf #2087: Undefined reference to NVS if a component is named "core"

nick-atmosphereiot, 2018-06-21:
## Environment

- Development Kit:      [AWS Lanyard]
- Core (if using chip or module): [ESP-WROOM-32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    4b91c82 (also in release 3.1)
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
If a project includes a component named `core`, you get undefined reference errors during linking. On the ESP32 forums I was told that there are no built in components named core, so it shouldn't be overwriting anything.

### Expected Behavior
It should build

### Actual Behavior
```
LD build/scan.elf
/home/nick/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_api.o):(.text.wifi_init_in_caller_task+0x14): undefined reference to `misc_nvs_init'
/home/nick/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_api.o): In function `wifi_init_in_caller_task':
(.text.wifi_init_in_caller_task+0x50): undefined reference to `misc_nvs_init'
/home/nick/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_deinit_process+0x0): undefined reference to `misc_nvs_deinit'
/home/nick/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_deinit_process+0x16): undefined reference to `misc_nvs_deinit'
/home/nick/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_restore_process+0x0): undefined reference to `misc_nvs_restore'
/home/nick/esp-idf/components/esp32/lib/libnet80211.a(ieee80211_ioctl.o):(.text.wifi_restore_process+0x7): undefined reference to `misc_nvs_restore'
/home/nick/esp-idf/components/esp32/lib/libnet80211.a(ieee80211.o):(.text.wifi_mode_set+0x0): undefined reference to `g_misc_nvs'
collect2: error: ld returned 1 exit status
```

### Steps to reproduce
Add a component named `core`

### Code to reproduce this issue
See attached project. If you do not use the `core` component, it builds just fine. If you rename it to anything other than `core`, it also builds just fine.

## Debug Logs
[scan_test.zip](https://github.com/espressif/esp-idf/files/2125280/scan_test.zip)

igrr, 2018-06-22:
Hi @nick-atmosphereiot, such issue will happen when a component's name results in the static library name which matches one of the pre-built libraries shipped with IDF:

```
libc-psram-workaround.a
libm-psram-workaround.a
libc_nano.a
libg-psram-workaround.a
libm.a
libg.a
libc.a
libhal.a
libwps.a
libcoexist.a
libpp.a
libmesh.a
libsmartconfig.a
libwpa.a
libphy.a
libcore.a
librtc.a
libwpa2.a
libespnow.a
libnet80211.a
libbtdm_app.a
```

So the following component names will cause such issue:
```
c-psram-workaround
m-psram-workaround
c_nano
g-psram-workaround
m
g
c
hal
wps
coexist
pp
mesh
smartconfig
wpa
phy
core
rtc
wpa2
espnow
net80211
btdm_app
```

We will update build system documentation to mention this.

negativekelvin, 2018-06-22:
Maybe core libs with common name could adopt esp prefix

tom-borcin, 2021-02-25:
Hello @nick-atmosphereiot,

thank you for reporting this issue. At Espressif we really appreciate it when our users report issues as it helps us make our products better. Unfortunately there hasn't been any progress with this issue for the last 2 years and we are closing it to keep our backlog manageable.

If you don't agree with this decision, please feel free to reopen the issue or contact me directly.

Regards,

Tomas

closed 2021-02-25 by tom-borcin