espressif/esp-idf #2088: sdspi_host_deinit

Masi2017, 2018-06-21:
Hi, This function remove all three spi devices and if user registered another spi device like LCD it removes that one as well!, I think it should have slot as an input.

ginkgm, 2018-06-27:
Hi @Masi2017 ,
As far as I can see from the source code of sdspi_host.c, it only deinit the device which is enabled.
```
esp_err_t sdspi_host_deinit()
{
    for (size_t i = 0; i < sizeof(s_slots)/sizeof(s_slots[0]); ++i) {
        if (s_slots[i].handle) {
            spi_bus_remove_device(s_slots[i].handle);
            free(s_slots[i].block_buf);
            s_slots[i].block_buf = NULL;
            free(s_slots[i].transactions);
            s_slots[i].transactions = NULL;
            spi_bus_free((spi_host_device_t) i);
            s_slots[i].handle = NULL;
        }
    }
    return ESP_OK;
}
```

If you do meet such problem , please come with more detailed information for us to help you. The issue template may be a good start. Thanks.

Masi2017, 2018-07-03:
Hi @ginkgm , Thanks for replying .
Yes you're right, the problem was because of “spicommon_dma_chan_free “ in spi_common.c which is called by “spi_bus_free” in “sdspi_host_deinit”. In this function DMA module gets disabled (PERIPH_SPI_DMA_MODULE) if only the dedicated DMA channel is freed. I am using two different DMA channeles one for LCD (channel 2) and another for SD card (channel 1). Once SD card is unmounted, DMA module gets disabled and this causes problem in the LCD.

ginkgm, 2018-07-04:
Hi @Masi2017 
I am sorry but you might be wrong.
according to 
https://github.com/espressif/esp-idf/blob/master/components/driver/spi_common.c#L102-L116

the driver carefully check whether all the channels are disabled, only when none of the channels are disabled will it disable the DMA peripheral.

closed 2018-07-05 by FayeY