espressif/esp-idf #2091: Esptool tries to write app at offset 0x0 when no factory partition configured

vitaminwater, 2018-06-22:
After upgrading I realized that `make flash` fails because esptools tries to write at offset 0x0, resulting in the following error:

```
Building partitions from /Users/YYY/Documents/perso/XXX/firmware/partitions.csv...
Flashing binaries to serial port /dev/cu.usbserial-14301 (app at offset 0x0 )...              
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]           
                           [--flash_mode {keep,qio,qout,dio,dout}]   
                           [--flash_size FLASH_SIZE]              
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]          
                           <address> <filename> [<address> <filename> ...]      
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x1000 for file: /Users/YYY/Documents/perso/XXX/firmware/build/bootloader/bootloader.bin
make: *** [flash] Error 2
```

chegewara, 2018-06-22:
https://github.com/espressif/esp-idf/issues/2086

vitaminwater, 2018-06-22:
Sorry didn't see this one, confirm that the patch un the mentionned issue fixes it. Thanks:)

closed 2018-06-22 by vitaminwater