espressif/esp-idf #2095: heap trace crash when both core are traced

chegewara, 2018-06-24:
I dont know if this is bug or design, but heap memory debugging with:
```
heap_trace_start();
heap_trace_stop();
heap_trace_dump();
```
if traced operations are on both cores between start and stop in most cases causes crash. Switching to menuconfig->FreeRTOS->run on one core only allows trace memory leaks with no crash.

projectgus, 2018-07-03:
Hi @chegewara,

Heap tracing should run across both cores. Do you mean you're starting tracing inside a task on one core, then stopping on another? This should work as well, but it's possible that it's not fully tested.

Do you have an example of a crash log, and any more code/context about what you're doing when it crashes?

Thanks,

Angus

chegewara, 2018-07-03:
Hi @projectgus, i dont have simple example for testing. Im not even sure if this is heap_trace issue. The case is:
- im starting heap trace (`heap_trace_start()`),
- few line later on the same task and the same core i end tracing and print results with `heap_trace_stop()` and `heap_trace_print()`
- in meantime i have nghttp2 client request, which is causing crash (without heap trace all works fine)

I wish i could help more, but its beyond my skills.

projectgus, 2018-07-03:
The scenario you're describing should work.

Are you sure the heap trace record structure is correctly allocated and the size is being set correctly when you initialize heap tracing?

Heap tracing changes timing a lot when enabled, so maybe it's triggering some other bug which is timing dependent?

chegewara, 2018-07-03:
Hi @projectgus 
I have more info. I did try it with esp_http_client and it crash:
```
D (5885) HTTP_CLIENT: on_message_begin
D (5885) HTTP_CLIENT: HEADER=Date:Tue, 03 Jul 2018 22:40:47 GMT
D (5885) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Date, value=Tue, 03 Jul 2018 22:40:47 GMT
D (5895) HTTP_CLIENT: HEADER=Content-Type:text/html; charset=utf-8
D (5895) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Type, value=text/html; charset=utf-8
D (5905) HTTP_CLIENT: HEADER=Content-Length:12
D (5905) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Length, value=12
D (5915) HTTP_CLIENT: HEADER=Connection:keep-alive
D (5925) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
D (5925) HTTP_CLIENT: HEADER=ETag:W/"c-dAuDFQrdjS3hezqxDTNgW7AOlYk"
D (5935) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=ETag, value=W/"c-dAuDFQrdjS3hezqxDTNgW7AOlYk"
D (5945) HTTP_CLIENT: http_on_headers_complete, status=401, offset=191, nread=191
D (5955) HTTP_CLIENT: http_on_body 12
D (5955) HTTP_CLIENT: HTTP_EVENT_ON_DATA, len=12
D (5955) HTTP_CLIENT: http_on_message_complete, parser=3ffba328
D (5965) HTTP_CLIENT: content_length = 12
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4011b251  PS      : 0x00060c30  A0      : 0x8011ad0b  A1      : 0x3ffc8080
A2      : 0x3ffc80a0  A3      : 0x00000000  A4      : 0x3ff96354  A5      : 0x0000001f
A6      : 0x00000001  A7      : 0x00000005  A8      : 0x80082b98  A9      : 0x3ffc8030
A10     : 0x00000000  A11     : 0x3ffc5f90  A12     : 0x3ffc8070  A13     : 0x0000000c
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

Backtrace: 0x4011b251:0x3ffc8080 0x4011ad08:0x3ffc80a0 0x4011b0f4:0x3ffc80d0 0x400d2c83:0x3ffc80f0 0x400d2cfe:0x3ffc8170
```
And here is backtrace gdb printout:
```
(gdb) list *0x4011ad08
0x4011ad08 is in esp_http_check_response (Z:/msys32/home/imper/esp/esp-idf/components/esp_http_client/esp_http_client.c:555).
550                 client->redirect_counter ++;
551                 client->process_again = 1;
552                 break;
553             case HttpStatus_Unauthorized:
554                 auth_header = client->auth_header;
555                 http_utils_trim_whitespace(&auth_header);
556                 ESP_LOGI(TAG, "UNAUTHORIZED: %s", auth_header);
557                 client->redirect_counter ++;
558                 if (auth_header) {
559                     if (http_utils_str_starts_with(auth_header, "Digest") == 0) {
(gdb) list *0x4011b251
0x4011b251 is in http_utils_trim_whitespace (Z:/msys32/home/imper/esp/esp-idf/components/esp_http_client/lib/http_utils.c:69).
64      void http_utils_trim_whitespace(char **str)
65      {
66          char *end;
67          char *start = *str;
68          // Trim leading space
69          while (isspace((unsigned char)*start)) start ++;
70
71          if (*start == 0) {  // All spaces?
72              **str = 0;
73              return;
(gdb)
```
Only changes in esp_http_client code i did is here:
```
#include "esp_heap_trace.h"

static void http_test_task(void *pvParameters)
{
    app_wifi_wait_connected();
    ESP_LOGI(TAG, "Connected to AP, begin http example");
    // http_rest();
    // http_auth_basic();
    // http_auth_basic_redirect();
    // http_auth_digest();
    // http_relative_redirect();
    heap_trace_start(HEAP_TRACE_ALL);
    http_absolute_redirect();
    heap_trace_stop();
    heap_trace_dump();

    // https();
    // http_redirect_to_https();
    // http_download_chunk();
    // http_perform_as_stream_reader();
    ESP_LOGI(TAG, "Finish http example");
    vTaskDelete(NULL);
}
#define NUM_RECORDS 100
heap_trace_record_t trace_record;

void app_main()
{
 heap_trace_init_standalone(&trace_record, NUM_RECORDS) ;

    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    app_wifi_initialise();

    xTaskCreate(&http_test_task, "http_test_task", 8192, NULL, 5, NULL);
}
```

chegewara, 2018-07-03:
ok, i thought thats not important, but after testing i see this change in code is leading to crash:
```
esp_http_client_config_t config = {
        .url = "https://maestro.emfcdn.com/stream_for/sa008/tunein",
        .event_handler = _http_event_handler,
    };
```

projectgus, 2018-07-05:
For benefit of anyone who sees this, there's a short piece of heap tracing example code under this section of the docs: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks

> The point is that it worked with normal code but crashed with nghttp2.

Memory corruption can be weird like this, especially static memory corruption (which heap checking can't detect). If the static memory which is corrupted is never read, you'll never know it was corrupted at all. And even a small change can reorder the way static memory is linked, so something gets moved from a "clean" memory location to one which is corrupted.

closed 2018-07-12 by chegewara