espressif/esp-idf #2099: Slight inconsistency in pad naming in TRM

nkolban, 2018-06-24:
Very small stuff ... 

In the Technical Reference Manual (TRM) on page 56 in 4.10 in Table 19 we see the list of pad names in column 2.  Pay attention to the pad names associated with GPIOs 32 through 39.

Now look at the names of the registers beginning IO_MUX_* on page 59 (4.12).  We see that the names of the registers are `IO_MUX_<PadName>_REG` ... except that the names associated GPIOs 32-39 are called 32-39 and not by their pad names.

For example.

We have a register called `IO_MUX_GPIO36_REG` as opposed to a register called `IO_MUX_SENSOR_VP_REG`.  It feels like we are not consistent.

negativekelvin, 2018-06-25:
The problem is that SENSOR_VP function is in the analog/rtc subsystem so has nothing to do with iomux.  Wouldn't it be more confusing if the register was called IO_MUX_SENSOR_VP_REG since no value of that register is going to cause it to have the SENSOR_VP function?

negativekelvin, 2018-06-25:
Well it is better to stick with the pad name _when possible_ since in schematics and docs you are more likely to see the pad name.

xiongyumail, 2018-09-25:
Hi @nkolban 
Thank you for your advice.

In `io_mux_reg.h` we use the `Function 1` to express registers.
Instead of the PAD name.

closed 2018-10-18 by Alvin1Zhang