espressif/esp-idf #2100: LED blink controlled my wifi

AKROF, 2018-06-25:
----------------------------- Delete below -----------------------------

Hi, 
I am fairly new to programming in general. I have done very basic arduino programming earlier and currently working with esp idf, which has proven to be very complicated for me in  terms of chosing which function we should choose. I wanted connect an LED to esp32 and set it so that when it is trying to connect to wifi, I want to led to blink. I donot want to use any fade option. All I need is a basic blink.
Can any one suggest how do I proceed.

Spritetm, 2018-06-25:
That's not really an issue with ESP-IDF. Suggest you use the forums at http://esp32.com for this instead.

Curclamas, 2018-06-25:
Maybe https://github.com/espressif/arduino-esp32 is also a good place to start in order to get acquainted with the ESP32 and they have a fair selection of examples where you could start. If you have any questions regarding these, there is also a chat: https://gitter.im/espressif/arduino-esp32
But let's agree there are better places/options to answer such a question than opening an issue in the IDF.

AmPalRed, 2018-06-26:
Sorry. I will close the issue now

closed 2018-06-26 by AmPalRed