espressif/esp-idf #2102: esp_http_client MEMORY LEAKS

DavidMora, 2018-06-25:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Core (if using chip or module): ESP32-Wrover
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 4b91c82cc447640e5b61407e810f1d6f3eabd233
- Development Env:		Make
- Operating System:		Ubuntu
- Power Supply:         USB
## Problem Description

esp_http_client module has memory leaks after perform. After taking a look at the code i think it is related to headers or body data. Since i was able to perform against custom server with no response packages and no leaks. 

### Expected Behavior

No memory leaks

### Actual Behavior

TEST CASE:

Running **HTTPREQUEST - send MEMORY LEAKS ON SIMPLE PERFORM  ...
MALLOC_CAP_8BIT potential leak: Before 227324 bytes free, After 226620 bytes free (delta 704)
MALLOC_CAP_32BIT potential leak: Before 289596 bytes free, After 288892 bytes free (delta 704)
./test_httprequest.c:205:**HTTPREQUEST - send MEMORY LEAKS ON SIMPLE PERFORM  :PASS
Test ran in 518ms

### Steps to repropduce

Just write the test below, different servers have different deltas.

### Code to reproduce this issue

```cpp
TEST_CASE("**HTTPREQUEST - send MEMORY LEAKS ON SIMPLE PERFORM  ", "[httprequest]")
{
    esp_http_client_config_t config = {
        .url="http://inmote.api.insite.com.co:3100/",        
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_http_client_perform(client);
    esp_http_client_cleanup(client);
}
```

DavidMora, 2018-06-26:
Hi Curclamas, 

I'm going to perform some test about this, but do you know what is the main reason for this?. 

DM

Curclamas, 2018-06-26:
AFAIK it has to do with the way the memory is allocated for the internal network/TCP/LWIP-stacks. I think some of this only happens once and the other memories are cleared regularly in other tasks (for example the network task), therefor the freeing of all memory might be delayed. But unless your example shows an "endless" leak it would not actually be considered a memory leak at all.

DavidMora, 2018-06-26:
How much time is it needed to confirm or discard memory leaks?

Curclamas, 2018-06-26:
I can not nail it down completely. From my experience: I had some tests where I checked http(s) every 10 seconds and about after a minute I had no more leaks with the memory being freed after a few seconds after each request was finished.

DavidMora, 2018-06-26:
Ok, i think you can close this issue now. I will run some tests and let you
know if i find something different.

Thank you so much!

VallariTrivedi, 2018-12-19:
Hi,

1. I am experiencing similar issue on esp32
2. I have enabled logs to print heap memory

- Free HEAP : 20608

after http post

- Free HEAP : 20384
224 bytes less

I am calling ` esp_http_client_cleanup(client`)

> it is returing ESP_OK

Here Is the code snipped I am using 
`void http_post_request(void)
{
    ESP_LOGD(TAG,"http_post_request-start -----> heap_caps_check_integrity_all : %s ", heap_caps_check_integrity_all(true) ? "true" : "false"); 
    ESP_LOGD(TAG,"http_post_request-start -----> free DRAM %u IRAM %u",heap_caps_get_free_size(MALLOC_CAP_8BIT),heap_caps_get_free_size(MALLOC_CAP_32BIT));
    esp_http_client_config_t config = 
    {
    .url = SomePath,
    .event_handler = _http_event_handler,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    if(client == NULL)
    {
       ESP_LOGE(TAG, "POST fail");
       Err_Flag = eADM_HTTP_FAIL;
    }
    else
    {
        const char *post_data = msgString;
        esp_http_client_set_url(client, SignonPath);
        esp_http_client_set_method(client, HTTP_METHOD_POST);
        esp_http_client_set_post_field(client, post_data, strlen(post_data));
        esp_err_t err = esp_http_client_perform(client);
        if (err == ESP_OK) {
            ESP_LOGD(TAG, "HTTP POST Status = %d",
                    esp_http_client_get_status_code(client));
        } 
        else 
        {
            ESP_LOGE(TAG, "POST fail: %s", esp_err_to_name(err));
            Err_Flag = eADM_HTTP_FAIL;
        }

        if(esp_http_client_cleanup(client)==ESP_OK)
            ESP_LOGI(TAG,"http connection closed clean");
        else
            ESP_LOGI(TAG,"http connection closed unclean");

ESP_LOGD(TAG,"http_post_request-start -----> heap_caps_check_integrity_all : %s ", heap_caps_check_integrity_all(true) ? "true" : "false"); 
ESP_LOGD(TAG,"http_post_request-start -----> free DRAM %u IRAM %u",heap_caps_get_free_size(MALLOC_CAP_8BIT),heap_caps_get_free_size(MALLOC_CAP_32BIT));
  
    }
}`

JuanValerio89, 2022-06-21:
Hi @VallariTrivedi 

I have the same issue, did you find how to avoid the leak ?

JuanValerio89, 2022-06-22:
Hi @projectgus i hope you can help me with my code, 

I have the next code 

>         esp_http_client_config_t config = {
>             .url = bufURL,
>             .method = HTTP_METHOD_POST,
>             .event_handler = _http_event_handler,
>         };
>         printf("Memory POST before: %u \n", esp_get_free_heap_size());
>         esp_http_client_handle_t clientePost = esp_http_client_init(&config);
>         esp_http_client_set_method(clientePost, HTTP_METHOD_POST);     // Set the current method to POST
>         esp_http_client_set_post_field(clientePost, out, strlen(out)); // Packet used when setting POST
>         esp_http_client_set_header(clientePost, "Content-Type", "application/json");
>         esp_http_client_set_header(clientePost, "x-api-key", "a90a9af7ec610ca4752c4dacd86086170fe11111");
>         esp_http_client_perform(clientePost);
> 
>         esp_http_client_cleanup(clientePost);
>         printf("Memory POST after: %u \n", esp_get_free_heap_size());

I call esp_get_free_heap_size(), after and before client perfom. I found that this part of code leak memory.
I tried do in this part of code in a cycle of 30 seconds, 3 minutes and 10 minutes. Everytime get the same issue. 

Memory POST after: `125464`
Memory POST after: `123652`
Memory POST after: `121416`
Memory POST after: `119560`

ESP-YJM, 2022-06-28:
@JuanValerio89 Do you test this cycle with a large times? Such as 100 times. And then check whether the memory has run out.

JuanValerio89, 2022-06-28:
Hello @ESP-YJM 

Yes I did it. Also I try changed some certificates but nothing. Always get the same result.

sajit7991, 2022-06-30:
Hi, similar issue is facing by me also, i guess it mbedtls or tls problem(not 100% sure). Is there any one who figure out the solution.

ESP-YJM, 2022-07-01:
> Hello @ESP-YJM
> 
> Yes I did it. Also I try changed some certificates but nothing. Always get the same result.

What IDF version you used? I need depend on your version and test it.

ESP-YJM, 2022-07-01:
> Hi, similar issue is facing by me also, i guess it mbedtls or tls problem(not 100% sure). Is there any one who figure out the solution.

Also What IDF version you used?

JuanValerio89, 2022-07-01:
I try with Visual Studio Espressif IDF v1.4.0 and IDF V4.4 and V4.3

sajit7991, 2022-07-01:
> > Hi, similar issue is facing by me also, i guess it mbedtls or tls problem(not 100% sure). Is there any one who figure out the solution.
> 
> Also What IDF version you used?

I'm using ESP-IDF v4.2.2 in eclipse.

ESP-YJM, 2022-07-01:
@sajit7991 @JuanValerio89 Sorry that i use esp_http_client example and change the code to POST data with TLS, and re-cycle about 200+ times. And i view no memory leak. I think you can refer to https://github.com/espressif/esp-idf/issues/2102#issuecomment-400503906 to debug where memory leak.

sajit7991, 2022-07-01:
> @sajit7991 @JuanValerio89 Sorry that i use esp_http_client example and change the code to POST data with TLS, and re-cycle about 200+ times. And i view no memory leak. I think you can refer to [#2102 (comment)](https://github.com/espressif/esp-idf/issues/2102#issuecomment-400503906) to debug where memory leak.

 Hi, for Json i'm using cJson lib. json is getting free by using the free() after the call of httpPost(). so from that side, i don't think  there  is any probelm. I'm sharing  the piece of code .   down there is the problem.txt 
[problem.txt](https://github.com/espressif/esp-idf/files/9028106/problem.txt)

ESP-YJM, 2022-07-20:
@sajit7991 It seems no memory leak in your posted code. I think you can follow https://github.com/espressif/esp-idf/issues/2102#issuecomment-400503906 to trace the memory usage.