espressif/esp-idf #2103: BT Classic (SPP profile) Discovery crash

louisthetall, 2018-06-25:
## Environment

- Development Kit: none
- Core: ESP-WROOM32
- IDF version: 4b91c82
- Development Env: Make
- Operating System: Windows 10
- Power Supply: external 5V

## Problem Description

I am currently working on a project where I am using a BTClassic (SPP profile) connection between two devices to transfer information, having used the esp-idf bt_spp_initiator and bt_ssp_acceptor examples as templates. The Initiator Device also have the a series of console commands I have implemented, which allow me to connect/disconnect from different Acceptor Devices more easily.

The problem occurs on the Initiator Device when I call the discovery function that looks for a Bluetooth network to connect to. The Initiator Device crashes after receiving a few ESP_BT_GAP_DISC_RES_EVT.

In the examples, the esp_bt_gap_start_discovery() function is called right after ESP_SPP_INIT_EVT. When I do this with my program, the crash does not occur and afterwards, I can use my console commands to  disconnect and reconnect as usual. However, when I change the Initiator Device to not start looking for bluetooth networks right away, and then input the "connect" command, which calls esp_bt_gap_start_discovery(), the device crashes.

## Coredump file
[bt_spp_coredump.txt](https://github.com/espressif/esp-idf/files/2134051/bt_spp_coredump.txt)

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x40106fe4  PS      : 0x00060630  A0      : 0x800fab66  A1      : 0x3ffddd10
0x40106fe4: BTM_CheckEirData at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/stack/btm/btm_inq.c:2445

A2      : 0x400067d0  A3      : 0x00000009  A4      : 0x3ffddd50  A5      : 0x00000000
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000002  A9      : 0x3ffddc90
A10     : 0x3ffd3bbc  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3f403272
A14     : 0x00000002  A15     : 0x3ffe2350  SAR     : 0x00000018  EXCCAUSE: 0x00000003
EXCVADDR: 0x400067d0  LBEG    : 0x400ff371  LEND    : 0x400ff37d  LCOUNT  : 0x00000000

0x400ff371: bdcpy at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_bt.c:610 (inlined by) btc_gap_bt_search_devices_evt at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_bt.c:354

0x400ff37d: btc_gap_bt_search_devices_evt at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_bt.c:610

Backtrace: 0x40106fe4:0x3ffddd10 0x400fab63:0x3ffddd30 0x400e72c6:0x3ffddd50 0x400ff0f9:0x3ffddd80 0x400ff3a5:0x3ffddda0 0x400ff58a:0x3ffddef0 0x400fdf52:0x3ffddf20
0x40106fe4: BTM_CheckEirData at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/stack/btm/btm_inq.c:2445

0x400fab63: esp_bt_gap_resolve_eir_data at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/api/esp_gap_bt_api.c:143

0x400e72c6: get_name_from_eir at C:/Users/Louis/Documents/Troo/AgYield/bluetooth_api_tester/components/btSppConn/BtSppConnInitiator.c:98
 (inlined by) esp_bt_gap_cb at C:/Users/Louis/Documents/Troo/AgYield/bluetooth_api_tester/components/btSppConn/BtSppConnInitiator.c:206

0x400ff0f9: btc_gap_bt_cb_to_app at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_bt.c:610

0x400ff3a5: btc_gap_bt_search_devices_evt at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_bt.c:610

0x400ff58a: btc_gap_bt_cb_handler at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_bt.c:730

0x400fdf52: btc_task at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:110
```

blueMoodBHD, 2018-06-27:
Hi @louisthetall 
Could you please give me `.elf` file,  Thanks.

daveatpanic, 2018-06-28:
I'm having the same problem with A2DP. The LoadStoreError and the line it's crashing on (BTM_CheckEirData calling STREAM_TO_UINT8) suggest that whatever that p_eir pointer is, it's not 8-bit accessible. I logged the pointer address and it's always somewhere around 0x40002xxx. That's in "Internal ROM 0", which doesn't make any sense.

Next step is tracing that pointer back to where it's being set. (Slow going using printf() debugging..) I'll let you know what I find!

blueMoodBHD, 2018-06-28:
Hi @louisthetall 
The message in your logs is different with the message in coredump. So, could you please test again, then sent the log, coredump and .elf (which are both in the same test)to me. Thanks.

daveatpanic, 2018-06-29:
@blueMoodBHD It looks like you introduced that in commit 174501f on 5/16. :) There are a couple more bad pointer calculations in there, lines 197 and 569 (in commit e3d3ed5).

blueMoodBHD, 2018-06-29:
@daveatpanic  Thanks very much for your help. I have commit this bug fix, and it will be merged into ESP-IDF soon.

blueMoodBHD, 2018-07-03:
Hi @daveatpanic  @louisthetall Please check that is there any problems now, Thanks.

FayeY, 2018-07-03:
Please have a try, and feel free to reopen if the issue persists. Thanks.

closed 2018-07-03 by FayeY

louisthetall, 2018-07-05:
that has resolved the issue, thanks for everyone's help on this!