espressif/esp-idf #2107: bluetooth LE indicate is missing?

xufeipyxis, 2018-06-26:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    aaf12390
- Development Env:		[other]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
I'm using a DevKitC to write/read from another BLE module (Microchip RN4020) which supports `ESP_GATT_CHAR_PROP_BIT_INDICATE`. But current esp apis seems to only supports `ESP_GATT_CHAR_PROP_BIT_NOTIFY`
//Detailed problem description goes here.

### Expected Behavior
`esp_ble_gattc_register_for_indicate` should be available
### Actual Behavior
only `esp_ble_gattc_register_for_notify` is available

### Code to reproduce this issue
this is my current workaround but I don't think it's the right way to go
```cpp
if (char_elem_result[0].properties & ESP_GATT_CHAR_PROP_BIT_NOTIFY){
    gl_profile_tab[PROFILE_A_APP_ID].char_handle = char_elem_result[0].char_handle;
    esp_ble_gattc_register_for_notify (gattc_if, gl_profile_tab[PROFILE_A_APP_ID].remote_bda, char_elem_result[0].char_handle);
}else if (char_elem_result[0].properties & ESP_GATT_CHAR_PROP_BIT_INDICATE){
    ESP_LOGI(GATTC_TAG, "ESP_GATT_CHAR_PROP_BIT_INDICATE");
    gl_profile_tab[PROFILE_A_APP_ID].char_handle = char_elem_result[0].char_handle;                            
    esp_ble_gattc_register_for_notify (gattc_if, gl_profile_tab[PROFILE_A_APP_ID].remote_bda, char_elem_result[0].char_handle);
}

my `char_elem_result[0].properties` is 0x28 which is Write & Indicate

xufeipyxis, 2018-06-27:
then should the gattc_demo.c be update to:

```cpp
if (char_elem_result[0].properties & (ESP_GATT_CHAR_PROP_BIT_NOTIFY|ESP_GATT_CHAR_PROP_BIT_INDICATE)){
    gl_profile_tab[PROFILE_A_APP_ID].char_handle = char_elem_result[0].char_handle;
    esp_ble_gattc_register_for_notify (gattc_if, gl_profile_tab[PROFILE_A_APP_ID].remote_bda, char_elem_result[0].char_handle);
}
```

Weijian-Espressif, 2018-07-05:
@xufeipyxis please refer to #1628 
@negativekelvin  thanks for your help.

closed 2018-07-06 by xufeipyxis