espressif/esp-idf #2110: kconfig build is kept in IDF_PATH/tools/kconfig

mamoit, 2018-06-26:
## Environment

- Development Kit:      none
- IDF version: 
    c2b39f4
- Development Env:		Make
- Operating System:		Ubuntu

## Problem Description
When building a project such as the example hello world using the provided Makefile, some of the generated objects will be kept in `$IDF_PATH/tools/kconfig`.
This makes it difficult to recompile a build with an ephemeral idf installation (such as a docker container), since it takes a while to relink everything instead of reusing the already compiled objects.

### Expected Behavior
All the objects resulting from compilation should be kept in the build folder.
This would allow make to not have to rebuild the whole project if it has the build folder.

### Actual Behavior
Some of the files generated by kconfig are kept in the esp-idf/tools/kconfig folder.

### Steps to reproduce
I have a docker image to reproduce this [here](https://github.com/mamoit/docker-esp32-idf).
But you can basically:
1. compile a project;
2. remove the esp-idf and re-clone it;
3. compile the project again.

Nothing should be recompiled, but it is.

### Code to reproduce this issue

```
git clone https://github.com/mamoit/docker-esp32-idf
cd docker-esp32-idf
make shell
# now you're in a container
cp -r ../esp-idf/examples/get-started/hello_world/* ./
make # the project is built
make # nothing is rebuilt
exit # get out of the container
make shell # and we're back in a new container
make # Entering directory '/srv/esp-idf/tools/kconfig' and recompiling everything
```

projectgus, 2018-06-26:
Thanks for reporting this, @mamoit . For what it's worth, this is fixed in the [CMake build system branch](https://esp32.com/viewtopic.php?f=13&t=5559&p=24109) (currently in preview release), which will be supported alongside GNU Make in IDF V3.2 and made the default in V4.0.

We may also be able to fix it in the GNU Make build system before this, though.

mamoit, 2018-06-27:
Just gave the CMake branch a run for its money, and yes, it does not suffer from the issue I reported.
Thank you @projectgus for your swift reply and work, this is a very well structured project and it has been a pleasure using it :+1:
You can close this issue, or keep it open if you think a GNU Make solution will be implemented.

projectgus, 2018-12-04:
Thanks @mamoit for confirming CMake fixed this issue (sorry I didn't reply earlier!)

Closing as I think we'll probably live with this limitation in GNU Make, and fix it by adopting CMake.

closed 2018-12-04 by projectgus

mamoit, 2018-12-04:
> we'll probably live with this limitation in GNU Make, and fix it by adopting CMake.

Sounds pretty reasonable :+1: