espressif/esp-idf #2113: Crash in LWIP

jeffsenn, 2018-06-27:
## Environment

- Development Kit:     
   custom board
- Core (if using chip or module): 
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    aaf12390
- Development Env:
make
- Operating System:
MacOS
- Power Supply: 
External

## Problem Description

Guru Mediation crash in LWIP. During normal operation. (Making api requests to website at a rate of about 1 per 10 seconds)

### Expected Behavior

Not crashing.

### Actual Behavior

Crashing:  (seems to be around dealing with segment ack in LWIP)

Reading symbols from build/isxwifi.elf...done.
(gdb) Line 1099 of "/Users/Shared/src/co/embeddedcode/tools/esp32/esp-idf/components/lwip/core/tcp_out.c" starts at address 0x401235ca <tcp_output+554>
   and ends at 0x4012361a <tcp_output+634>.
(gdb) Line 1065 of "/Users/Shared/src/co/embeddedcode/tools/esp32/esp-idf/components/lwip/core/tcp.c" starts at address 0x4011fcd7 <tcp_slowtmr+843>
   and ends at 0x4011fcdc <tcp_slowtmr+848>.
(gdb) Line 148 of "/Users/Shared/src/co/embeddedcode/tools/esp32/esp-idf/components/lwip/core/tcp.c" starts at address 0x4012015f <tcp_tmr+23>
   and ends at 0x40120164 <tcp_shutdown>.
(gdb) Line 86 of "/Users/Shared/src/co/embeddedcode/tools/esp32/esp-idf/components/lwip/core/timers.c" starts at address 0x40123bdb <tcpip_tcp_timer+3>
   and ends at 0x40123bde <tcpip_tcp_timer+6>.
(gdb) Line 575 of "/Users/Shared/src/co/embeddedcode/tools/esp32/esp-idf/components/lwip/core/timers.c"
   starts at address 0x40123d53 <sys_timeouts_mbox_fetch+71>
   and ends at 0x40123d5c <sys_timeouts_mbox_fetch+80>.
(gdb) Line 102 of "/Users/Shared/src/co/embeddedcode/tools/esp32/esp-idf/components/lwip/api/tcpip.c" starts at address 0x4011c646 <tcpip_thread+18>
   and ends at 0x4011c64f <tcpip_thread+27>.

### Steps to repropduce

Happens randomly 5 mins to an hour of running small connections(open,write,read,close) against remote server on a busy LAN.
Note: This does NOT seem to happen with exactly the same code running on nearly identical hardware with an ethernet PHY rather than Wifi.

## Debug Logs

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.\0d
Core 1 register dump:\0d
PC      : 0x401235f1  PS      : 0x00060f30  A0      : 0x8011fcdc  A1      : 0x3ffd0db0  \0d
A2      : 0x3ffccb74  A3      : 0x3ffd038c  A4      : 0x004e0000  A5      : 0x00004e89  \0d
A6      : 0x00000000  A7      : 0x00001653  A8      : 0x801235ee  A9      : 0x3ffd0d90  \0d
A10     : 0x00004e89  A11     : 0x3ffccb74  A12     : 0x00000000  A13     : 0x3ffcc058  \0d
A14     : 0x3ffcc4e0  A15     : 0x3ffd8d20  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  \0d
EXCVADDR: 0x00000010  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  \0d
\0d
Backtrace: 0x401235f1:0x3ffd0db0 0x4011fcd9:0x3ffd0dd0 0x4012015f:0x3ffd0df0 0x40123bdb:0x3ffd0e10 0x40123d55:0x3ffd0e30 0x4011c64c:0x3ffd0e50\0d
\0d
Rebooting...\0d

jeffsenn, 2018-06-27:
Further info: 'useg' is null. Probably a bug in LWIP.  What are the odds of upgrading the version of LWIP to be more current in ESP-IDF?

jeffsenn, 2018-06-27:
Not sure what is going wrong - the LWIP code for this has moved on quite a bit after the version included in ESP-IDF currently.  I have temporarily patched the crash with this (very hurried and possibly wrong code):

diff --git a/components/lwip/core/tcp_out.c b/components/lwip/core/tcp_out.c
index 8eee0fee..c3c8901a 100644
--- a/components/lwip/core/tcp_out.c
+++ b/components/lwip/core/tcp_out.c
@@ -1093,6 +1093,10 @@ tcp_output(struct tcp_pcb *pcb)
         useg = seg;
       /* unacked list is not empty? */
       } else {
+        //fix for possible bug in LWIP - not sure what is going wrong... 
+        if(!useg && seg == pcb->unacked) {
+          useg = seg;
+        } else 
         /* In the case of fast retransmit, the packet should not go to the tail
          * of the unacked queue, but rather somewhere before it. We need to check for
          * this case. -STJ Jul 27, 2004 */

zhangyanjiaoesp, 2018-07-16:
@jeffsenn how to reproduce this problem? Can you provide your demo?

jeffsenn, 2018-07-16:
I can't provide an easy test case (it's part of a multi- processor IoT device - just the ESP32 code would not help as the ESP32 is only doing TCP stream processing).  There is nothing exotic about the code, it is simply making a single TCP socket connection (using LWIP raw API) once every few seconds and making a small HTTPS request and closing the connection. (Note: it is handling the HTTPS at a higher level, NOT in the ESP32 code - the ESP32 is just forwarding the binary stream to another processor).

I suspect that it has to do with packet ack/re-transmit as it appears to happen (randomly) more often on a busy LAN.

The above fix prevents the crash - however I cannot figure out why useg == NULL && pcb->unacked == seg.  Unless the retransmit  (tcp_slowtmr()?) is somehow interrupting tcp_output().

jeffsenn, 2018-07-17:
Actually (now that I'm looking how the ESP port invokes the timer functions) - perhaps what I'm trying to do is not supported. I do have calls into the LWIP raw API that do not originate from the dedicated tcp thread.  Is the "raw" lwip API not supported?  Is the netconn API supported? (I happen to be using existing code that uses the raw API, but I suppose I could convert it).

jeffsenn, 2018-07-17:
Ok - I've switched my few calls that were not in the TCP thread into it (using: 'tcpip_callback_with_block'  - for anyone finding this thread, this is not that well documented, but reading the code in 'lwip/api/tcpip.c' will be instructive ). I'll test for awhile and see if the problem is gone.