espressif/esp-idf #2114: using ccache with esp-idf

nicola-lunghi, 2018-06-27:
Hi all, 

is possible to use ccache to speed up the build?

Thanks
Nicola Lunghi

me-no-dev, 2018-06-27:
totally you can ;) many of us already use it :)

nicola-lunghi, 2018-06-28:
ok i managed to create a directory called bin-ccache in the xtensa installation directory and link all the executables in bin to ccache

then I've added the xtensa bin-ccache folder to path before the bin folder itself.

and ccache works perfectly

closed 2018-06-28 by nicola-lunghi

projectgus, 2018-07-09:
Hi @nicola-lunghi ,

Just FYI, if you're still using the cmake branch then you should get ccache enabled automatically if it's on your PATH:
https://github.com/espressif/esp-idf/blob/feature/cmake/tools/cmake/idf_functions.cmake#L101

(But setting up links in the way you mention will also work - I have this set on my local system as well.)

Angus