espressif/esp-idf #2115: How do I export/copy project settings in eclipse from one project to another

SantoshVema1, 2018-06-27:
Hello,

I followed the instructions under http://esp-idf.readthedocs.io/en/latest/get-started/eclipse-setup.html#eclipse-install-steps for setting up eclipse in linux and I was able build and flash successful. But how do I copy/export these **_Project Settings_** from one project to another without manually configuring these settings again?

Thanks,

Santosh

krzychb, 2018-06-27:
Hi @SantoshVema1 this is described under Eclipse window picture in https://github.com/espressif/esp-idf/pull/157#issuecomment-270134045

If the issue is a general question, starts similar to "How do I..", please discuss this on our community forum at esp32.com instead.

SantoshVema1, 2018-06-27:
Hello @krzychb,

Thanks that did the trick. For anyone else who need this. The only thing that this doesn't do is copy the Indexer settings below take from the instructions page. 
Navigate to “C/C++ General” -> “Indexer” property page:

Check “Enable project specific settings” to enable the rest of the settings on this page.
Uncheck “Allow heuristic resolution of includes”. When this option is enabled Eclipse sometimes fails to find correct header directories.

So to save these as well, just click "Store settings with project" under “C/C++ General” -> “Indexer” property page, before you copy the project and paste as a new one.

Thanks,

Santosh

closed 2018-07-05 by FayeY

manojsai17, 2019-09-20:
Hello Santosh ,
I able to copy the paths of .xml to antother project .

But how to copy all settings of project i.e IDF_PATH,indexer settings . to another project instead of manually typing 

#So to save these as well, just click "Store settings with project" under “C/C++ General” -> “Indexer” property page, before you copy the project and paste as a new one.

Unable to understand what to do in this step