espressif/esp-idf #2116: a2dp source bluetooth example not connecting to phone speaker or pc.

yoelrc88, 2018-06-27:
## Environment

- Development Kit:      [HUZZAH32 - ESP32 Feather | MATRIX Voice ESP32 version ]
- Kit version (for WroverKit/PicoKit/DevKitC): [ not sure ]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version [4b91c82cc447640e5b61407e810f1d6f3eabd233 Jun 19 2018]
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB|external 5V|]

## Problem Description
I am trying the a2dp_source example to connect the ESP-WROOM32 to stream audio data from the ESP32 to a PC acting as a bluetooth a2dp sink. I know the example was built to talk to the sink example in the esp-idf but I supposed can be also used as a reference for what I am trying to do.  

Basically connection fails at some part of the process 

//Detailed problem description goes here.

### Expected Behavior
I am trying to connect the a2dp source example to an Android phone or to a PC runing Ubuntu or simply to a bluetooth speaker.

### Actual Behavior
Not been able to connect.

### Steps to repropduce
I am pretty much using the same code in the a2dp source example. The only change so far was in the `filter_inquiry_scan_result` method where I am looking now for a specific address (my phone address):
### Code to reproduce this issue
```cpp
static void filter_inquiry_scan_result(esp_bt_gap_cb_param_t *param)
{
    char bda_str[18];
    uint32_t cod = 0;
    int32_t rssi = -129; /* invalid value */
    uint8_t *eir = NULL;
    uint8_t *bdname = NULL;
    esp_bt_gap_dev_prop_t *p;

    ESP_LOGI(BT_AV_TAG, "*********************************************");
    ESP_LOGI(BT_AV_TAG, "Scanned device: %s", bda2str(param->disc_res.bda, bda_str, 18));
    for (int i = 0; i < param->disc_res.num_prop; i++) {
        p = param->disc_res.prop + i;
        switch (p->type) {
        case ESP_BT_GAP_DEV_PROP_COD:
            cod = *(uint32_t *)(p->val);
            ESP_LOGI(BT_AV_TAG, "--Class of Device: 0x%x", cod);
            break;
        case ESP_BT_GAP_DEV_PROP_RSSI:
            rssi = *(int8_t *)(p->val);
            ESP_LOGI(BT_AV_TAG, "--RSSI: %d", rssi);
            break;
        case ESP_BT_GAP_DEV_PROP_EIR:
            eir = (uint8_t *)(p->val);
            ESP_LOGI(BT_AV_TAG, "--EIR: %d", (int32_t)eir);
            break;
        case ESP_BT_GAP_DEV_PROP_BDNAME:
            bdname = (uint8_t *)(p->val);
            ESP_LOGI(BT_AV_TAG, "-- >>>>>>>>>>>>>>>>>>>>>> BDNAME: %s", (char *)bdname);
        default:
            break;
        }
    }

    if (strcmp((char *)bda_str, "60:a4:d0:c6:9b:c2") == 0) {
      ESP_LOGI(BT_AV_TAG,
               ">>>>>>>>>>>>>>>> Found Galaxy S7 address %s, name %s",
               bda_str, peer_bdname);
      m_a2d_state = APP_AV_STATE_DISCOVERED;
      memcpy(peer_bda, param->disc_res.bda, ESP_BD_ADDR_LEN);
      ESP_LOGI(BT_AV_TAG, "Cancel device discovery ...");
      esp_bt_gap_cancel_discovery();
    }
```
So what happen when I run this is that it founds the phone and tries to connect to it, but eventually it failed and the esp32 restarts. See below the UART output :

## Debug Logs

```log
I (8783) BT_AV: *********************************************
I (8783) BT_AV: Scanned device: 60:a4:d0:c6:9b:c2
I (8783) BT_AV: --Class of Device: 0x5a020c
I (8783) BT_AV: --RSSI: -46
I (8793) BT_AV: --EIR: 1073690736
I (8793) BT_AV: >>>>>>>>>>>>>>>> Found Galaxy S7 address 60:a4:d0:c6:9b:c2, name yo7
I (8803) BT_AV: Cancel device discovery ...
I (8813) BT_AV: Device discovery stopped.
I (8813) BT_AV: a2dp connecting to peer: yoelrc88S7
I (8813) BT_AV: esp_a2d_source_connect(): ESP_OK
E (8823) BT_APPL: reset flags
E (9513) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (9653) BT_APPL: bta_dm_act no entry for connected service cbs
W (9653) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 3

I (10523) BT_AV: a2dp connecting to peer: 60:a4:d0:c6:9b:c2
E (10523) BT_APPL: reset flags
E (11113) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (11323) BT_APPL: bta_dm_act no entry for connected service cbs
W (11323) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 3

E (14733) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
/home/yoelrc88/esp/esp-idf/components/freertos/queue.c:718 (xQueueGenericSend)- ass!
abort() was called at PC 0x4008b54f on core 0

Backtrace: 0x4008e934:0x3ffddca0 0x4008eb0b:0x3ffddcc0 0x4008b54f:0x3ffddce0 0x40110

Rebooting...
```
I did successfully tested to connect the sink example and this source example .. all good.

## Questions:
* Is this demo able to connect to other a2dp sink devices (other than the sink a2dp example provided) without any mayor modification?
* In case that any modification is needed can you provide some guidance where to start.

Let me know if you need more infromation about the my setup.

Thank you

rob-bits, 2018-07-02:
Hi there,

I do not have any suggestion for your problem but I can answer one of your question. 

Yes this example code can be used to connect the ESP32 to a commercial available Bluetooth Sink device (BT speaker). And the esp32 can stream audio to the connected sink device with good audio quality.

Regards,

Rob

yoelrc88, 2018-07-03:
Thank you @rob-bits, for taking the time to respond. I guess the speaker I tried is, in some way, incompatible with the example. I also tried to connect it to an Ubuntu pc using the standard Bluetooth software in Ubuntu. 

I will test again and try with different speakers. 

Any ideas or suggestions from anyone else would be much appreciated.

Thank you

-Yoel

zenmanenergy, 2018-09-10:
@yoelrc88  have you had any luck connecting off the shelf BT speakers? I'm just starting my project, so I'm just researching at the moment, I'm personally interested in BT headphones. @rob-bits Is there a specific BT version this works with?

blueMoodBHD, 2018-09-25:
Hi @yoelrc88 , First, according to your code, your changes is OK.
Then according your log, the COD of your phone is  `0x5a020c`, **Pendering** bit is not set but **Capturing** bit is set, which means it is a device act source instead of sink.
Please make sure there is a sink server in the device you want connect, using **sdptool** or other ways.
Thanks.

Alvin1Zhang, 2018-10-15:
@yoelrc88 Could you help share some updates of this issue? Thanks.

Alvin1Zhang, 2019-06-14:
@yoelrc88 Thanks for reporting the issue. Would you help share if any updates for this issue? Thanks.

yunling-cn, 2019-06-23:
Hello, how do you implement the connection to the Bluetooth headset? I use ESP32 to connect to the Bluetooth headset and always stuck in "esp_a2d_source_connect". I just modified the official example a2dp_source. But connecting a computer can be, that is, connecting a Bluetooth headset does not work. Sometimes it suddenly crashes and then restarts and suddenly connects.

AbnerFederer, 2019-12-03:
Hi @yunling-cn 

Please pay attention to the official a2dp source example runs as follow:

search --> connect --> play for a while --> disconnect --> restart

Thank you.

Alvin1Zhang, 2019-12-12:
@yunling-cn Thanks for the updates. @yoelrc88 Thanks for reporting, will close this issue due to short of feedback. Feel free to reopen if the issue still happens. Thanks.

closed 2019-12-12 by Alvin1Zhang