espressif/esp-idf #2117: ubuntu Eclipse neon header file issue

ahmedawais, 2018-06-28:
I am using Eclipse neon and trying to build a BLE project which send and receive data from andriod phone. for this i am using neil kolban libary of BLE       https://github.com/nkolban/ESP32_BLE_Arduino   
i have added all the necessay header path into the main folder of project and also into C/C++ general path and symbols but still i am getting this error. BLEDevice.h No such file or directory. Bluetooth is enable in menuconfig and paths have been added in both GNU C  and GNU C++. 
[image]
[image]
[image]
[image]

please suggest i am a newbie 
thanks

Dazza0, 2018-07-04:
Did you also copy the ``src`` directory of the repository?

ahmedawais, 2018-07-04:
I copied the  whole tool chain of esp 32 from github repository. Can
explain a bit ?

Dazza0, 2018-07-04:
I'm referring to the [src](https://github.com/nkolban/ESP32_BLE_Arduino/tree/master/src) directory in the repo you linked above. ``BLEDevice.h`` doesn't exist in ESP-IDF

ahmedawais, 2018-07-04:
yes  i copied src directory path and also added its path in Path ans symbols

Dazza0, 2018-07-04:
Adding the paths to Eclipse has no effect on the build. I suggest you either make those sources a separate component by adding a ``component.mk`` file into the ``src`` directory (see ESP-IDF components for examples), or directly putting those source files in the same directory as your ``main.cpp``

FayeY, 2018-07-12:
Any update? Is this issue solved now?

closed 2018-08-08 by FayeY