espressif/esp-idf #2118: Brownout detector was triggered with multimeter

mojgithub, 2018-06-28:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Chip version ( v0 and v1]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (latest)
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]
- Power Supply:         [USB|external 5V|external 3.3V|]

## Problem Description

I am getting error "Brownout detector was triggered" when powering up my ESP32 and measuring the current with my multimeter.

ESP32 is working without measuring with the multimeter.

I read that this is probably power related error.

1) ESP32 is connected with USB. I have open serial monitor.
2) ESP32 has external power supply 3.3V.
3) I have strapped a 104uF cap across 3.3v and Gnd.

4) I have also tried with external 5v with 3.3v voltage regulator.

Any ideas?

## Debug Logs

```
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90

DEVICE - Wakeup was not caused by deep sleep

E (20) wifi: esp_wifi_sta_get_ap_info 1277 wifi is not init

Brownout detector was triggered
```

igrr, 2018-06-28:
This indeed looks like a power supply related issue. Usually putting an ordinary multimeter in line with ESP32 when WiFi is used causes such issues. If the multimeter is an auto-ranging one, then it usually can not switch ranges fast enough as the current is increasing. For an non auto-ranging multimeter, set the range to 10A initially (or just short the multimeter input with a jumper) and switch to 200mA after WiFi has been enabled.

mojgithub, 2018-06-28:
i dont have ordinary multimeter(only auto ranging) :( so guess i cant measure it. Sad panda.

closed 2018-06-28 by mojgithub

hal9000-x, 2018-12-13:
i have the same problem. so i attached an oscilloscope and saw it is drawing too much current and the voltage goes down below 2.8V (sometimes below 2.5-2.6V) with a spike. so i disabled the brown out detection by WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0) before WiFi.begin() and seems working.

btw, you have to include two libraries:  "soc/soc.h" and "soc/rtc_cntl_reg.h"

mrcmarco90, 2026-10-06:
Filed a DSI-side report with register-level forensics (SAR correct, master port correct, PSRAM content verified correct via read-back, no underrun — yet wrong data delivered): espressif/esp-idf#19188