espressif/esp-idf #2119: Which GPIO pins are unusable when Wifi is active

cspwcspw, 2018-06-28:
There are numerous posts about failure of AnalogRead when WiFi is active.  The docs https://github.com/espressif/esp-idf/blob/4b91c82cc447640e5b61407e810f1d6f3eabd233/docs/en/api-reference/peripherals/adc.rst  (url)  tell us that WiFi uses ADC2, and the ten GPIO pins 0, 2, 4, 12-15, and 25-27 are off limits.  (Off limits just for ADC conversions, or off limits for everything?)

I'm interfacing an OV7670 (no fifo) camera, which works fine until I turn on WiFi.   So my pin assignments are probably wrong / conflicting with WiFi.    But all the sample code that I find, and my WROVER board with a standard camera connector seem to quite happily use some pins from ADC2.    

So does the advice that ADC2 and WiFi don't play well together mean that 
a) I cannot do ADC conversions on any of those pins while WiFi is active, but I could use the pins for digital I/O,  or does it mean 
b) "These pins are strictly off limits for any kind of usage while WiFi is active." , or 
c) you will get away with it provided WiFi is not trying to set the pin mode (input/output) different from the what the camera is trying to do.    

And then, how do samples like https://github.com/bitluni/ESP32CameraI2S work? (It uses many of the ADC2 pins)   Or do WiFi and the camera play nicely together on the WROVER kit, even though the camera header breaks out quite a few ADC2 pins. 

Thanks
Peter

negativekelvin, 2018-06-28:
(a)

FayeY, 2018-07-05:
Is this issue solved now?

closed 2018-07-12 by FayeY