espressif/esp-idf #2120: touch_pad_filter intrude auto_light_sleep

infotrino, 2018-06-28:
## Environment

- Development Kit:       custom board from infotron (ESP32-Wrover)
- Core (if using chip or module): ESP32-Wrover
- IDF version : esp-idf v3.1-beta
- Development Env:		Make
- Operating System:		Windows
- Power Supply:         USB

## Problem Description
Problem 1:
If I use touch interface with touch_pad_filter_start(FILTER_PERIOD),  then I cannot go into auto light sleep. 
Without filter mode, it looks like normal.
   1. without  touch_pad_filter_start(10) + auto light sleep code -->10mA
   2. without  touch_pad_filter_start(10) + esp_pm_lock_acquire(pm_lock_handle_no_light_sleep); -->22mA
   3. with        touch_pad_filter_start(10) + auto light sleep code -->50mA
   4. with       touch_pad_filter_start(10) + esp_pm_lock_acquire(pm_lock_handle_no_light_sleep); -->50mA
If I use touch_pad_filter_start, current consumption is same regardless auto light sleep.
So, I think touch filter mode intrude auto light sleep, 

Problem 2:
In the auto light sleep environment,  the value from touch_pad_read_filtered(..) is very different from that of touch_pad_read_raw_data(). variance of filtered value is much larger than variance of raw value.  This cannot occur if filtering is low pass filtering.

Problem 3:
In the auto light sleep environment,  touch interrupt occurs too frequently (with filter mode).  I cannot understand this face, since I thought interrupt is determined by raw value not filtered value.  If I don't use filter mode, all is fine.

## CODE

void touch_pad_init()
{
	touch_pad_init();
    touch_pad_filter_start(10);   // <-------------------------------------------
    touch_pad_set_meas_time(TOUCH_SLEEP_TIME, 0xffff);
    touch_pad_set_voltage(TOUCH_HVOLT_2V7, TOUCH_LVOLT_0V5, TOUCH_HVOLT_ATTEN_1V5);
    touch_pad_config(0, TOUCH_THRESH_NO_USE);
    tp_set_threshold();
    touch_pad_set_trigger_mode(TOUCH_TRIGGER_BELOW);
    touch_pad_isr_register(tp_rtc_intr, NULL);
    touch_pad_intr_enable();
}

igrr, 2018-06-29:
FreeRTOS Tickless Idle feature which automatic light sleep is based on has a threshold for the minimal amount of time the system can spend in sleep mode. This threshold can be configured using [CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#envvar-CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP) option in menuconfig. By default this is set to 3 ticks, and the lowest value is 2 ticks. If you set filter period (argument of touch_pad_filter_start) to `portTICK_PERIOD_MS * (CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP + 1)`, then the system should be able to go into sleep mode in between timer task wakeups.

infotrino, 2018-07-02:
I understand.

Thank you very much igrr.

closed 2018-07-05 by FayeY