espressif/esp-idf #2122: SPI Flash not working

epicface2304, 2018-06-28:
Hello, I am having some issues with my ESP32's SPI flash.

I am trying to use it with a 7.5" waveshare ePaper/eInk display and it is not working with the example code, I have tried the code with the 2 boards I have and both get the same error during boot. The code shows debug messages like it is working, but the display is not updating so I think this error has to do with it.

`load:0x40080000,len:6696
entry 0x400802e4
E (129) spiram: SPI RAM enabled but initialization failed. Bailing out.`

I am using the ESP32 DOIT DevKit v1 with the Arduino IDE on Windows.

negativekelvin, 2018-06-28:
You don't have spi **ram** on that board so don't enable it in menuconfig

epicface2304, 2018-06-28:
How do I disable SPI ram? I can't seem to find menuconfig.

dhrishi, 2018-06-28:
@epicface2304 You can disable it through menuconfig.
make menuconfig -> Component config -> ESP32-specific -> Support for external, SPI-connected RAM

epicface2304, 2018-06-28:
For the Arduino IDE is it the sdkconfig file within the sdk folder?

dhrishi, 2018-06-28:
Oops. Sorry missed that you are using Arduino IDE. I am not really sure if this can be done as the libs are prebuilt from IDF

epicface2304, 2018-06-28:
oh... is there any other way I can work around this?

negativekelvin, 2018-06-28:
It should not matter unless your app is trying to use spi ram specifically.  Anyway you can close this issue it doesn't belong here.

closed 2018-06-28 by epicface2304