espressif/esp-idf #2123: Feature Request: mDNS Service Query callback

paulreimer, 2018-06-28:
It would be very handy to have a e.g. a new event type in the event loop handler, which is triggered whenever an mDNS service is returned in response to a query from the network.

As an example use-case, using this hook would make it possible to start a service on demand, whenever a specific mDNS service is queried. (So you could have functionality for e.g. an HTTP server, that is always published to mDNS, but the actual server is only activated when it is scanned by another device).

liuzfesp, 2018-07-08:
Hi @paulreimer, we have no plan to support this event in short time, such as IDF v3.2. We will consider it in future release plan.

david-cermak, 2019-10-15:
@paulreimer I agree that this feature might be quite useful for some usecases, but since this is pretty old request let me check with you first if this is still something you would find meaningful feature for IDF to support?

Alvin1Zhang, 2019-11-12:
@paulreimer Thanks for reporting, feel free to reopen if needed, thanks.

closed 2019-11-12 by Alvin1Zhang