espressif/esp-idf #2127: Guru Meditation Error with release version (wifi power save example)

infotrino, 2018-06-29:
## Environment

- Development Kit:    custom board from infotron (Wrover)
- Core (if using chip or module): ESP32-Wrover
- IDF version: v3.1-beta
- Development Env:		Make
- Operating System:		Windows
- Power Supply:         USB

## Problem Description
If I compile "wifi power save example" as release version, I meet guru meditation error.
(No problem with debug version)

### Steps to repropduce
1.  copy power_save from example folder
2. make menu config: debug --> release

### monitor error report
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5616
ho 0 tail 12 room 4
load:0x40078000,len:6568
ho 0 tail 12 room 4
load:0x40080000,len:7056
0x40080000: _iram_start at D:/ghlee/18.esp32/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080384
0x40080384: _UserExceptionVector at ??:?

I (32) boot: ESP-IDF v3.1-beta1-dirty 2nd stage bootloader
I (32) boot: compile time 12:03:18
I (38) boot: Enabling RNG early entropy source...
I (38) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (68) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (75) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  3 factory          factory app      00 00 00010000 001a0000
I (90) boot:  4 ota_0            OTA app          00 10 001b0000 000b0000
I (98) boot:  5 ota_1            OTA app          00 11 00260000 000b0000
I (105) boot: End of partition table
I (109) boot: Defaulting to factory image
I (114) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x10a64 ( 68196) map
I (147) esp_image: segment 1: paddr=0x00020a8c vaddr=0x3ffb0000 size=0x0375c ( 14172) load
I (153) esp_image: segment 2: paddr=0x000241f0 vaddr=0x3ffb375c size=0x00000 (     0) load
I (154) esp_image: segment 3: paddr=0x000241f8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at D:/ghlee/18.esp32/esp-idf/components/freertos/xtensa_vectors.S:1685

I (163) esp_image: segment 4: paddr=0x00024600 vaddr=0x40080400 size=0x0ba10 ( 47632) load
I (191) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x5d9cc (383436) map
0x400d0018: _stext at ??:?

I (325) esp_image: segment 6: paddr=0x0008d9ec vaddr=0x4008be10 size=0x046f4 ( 18164) load
0x4008be10: vTaskPlaceOnUnorderedEventList at D:/ghlee/18.esp32/esp-idf/components/freertos/tasks.c:3564

I (333) esp_image: segment 7: paddr=0x000920e8 vaddr=0x400c0000 size=0x00064 (   100) load
I (333) esp_image: segment 8: paddr=0x00092154 vaddr=0x50000000 size=0x00000 (     0) load
I (351) boot: Loaded app from partition at offset 0x10000
I (351) boot: Disabling RNG early entropy source...
I (353) cpu_start: Pro cpu up.
I (356) cpu_start: Starting app cpu, entry point is 0x40081178
0x40081178: call_start_cpu1 at D:/ghlee/18.esp32/esp-idf/components/esp32/cpu_start.c:225

[... 290 lines cut ...]
#define CONFIG_MBEDTLS_TLS_SERVER_AND_CLIENT 1
#define CONFIG_FREERTOS_ISR_STACKSIZE 1536
#define CONFIG_OPENSSL_ASSERT_DO_NOTHING 1
#define CONFIG_WL_SECTOR_SIZE_4096 1
#define CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA 1
#define CONFIG_WIFI_PASSWORD "mypassword"
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE 32
#define CONFIG_ESP32_WIFI_TX_BUFFER_TYPE 1
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 1
#define CONFIG_LWIP_LOOPBACK_MAX_PBUFS 8
#define CONFIG_ESP32_DEFAULT_CPU_FREQ_80 1
#define CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP 3
#define CONFIG_SPI_FLASH_ROM_DRIVER_PATCH 1
#define CONFIG_SPIFFS_PAGE_SIZE 256
#define CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED 1
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 1
#define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 3072
#define CONFIG_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_NEWLIB_STDOUT_LINE_ENDING_CRLF 1
#define CONFIG_ESPTOOLPY_PORT "com3"
#define CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS 1
#define CONFIG_OPTIMIZATION_LEVEL_RELEASE 1

igrr, 2018-06-29:
Thanks for the comprehensive report @infotrino. We will have the fix for this soon. In the meantime, you may apply the following change locally to work around the issue:

```patch
From 8878a65bf04ec3925b814532ba91fa181609d36f Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Fri, 29 Jun 2018 22:30:32 +0800
Subject: [PATCH] xtensa: make XTHAL_GET_INTERRUPT volatile

---
 components/esp32/include/xtensa/core-macros.h | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/components/esp32/include/xtensa/core-macros.h b/components/esp32/include/xtensa/core-macros.h
index c8f7e3476..c2eb4a646 100644
--- a/components/esp32/include/xtensa/core-macros.h
+++ b/components/esp32/include/xtensa/core-macros.h
@@ -335,7 +335,7 @@
 			__asm__ __volatile__("wsr.intenable %0" :: "a"(__intenable):"memory"); \
 				} while(0)
 # define XTHAL_GET_INTERRUPT()	({ int __interrupt; \
-				__asm__("rsr.interrupt %0" : "=a"(__interrupt)); \
+				__asm__ __volatile__ ("rsr.interrupt %0" : "=a"(__interrupt)); \
 				__interrupt; })
 # define XTHAL_SET_INTSET(v)	do { int __interrupt = (int)(v); \
 			__asm__ __volatile__("wsr.intset %0" :: "a"(__interrupt):"memory"); \
-- 
2.16.0.windows.2

```

infotrino, 2018-07-02:
That works well.

Thank you igrr.

closed 2018-07-12 by igrr