espressif/esp-idf #2129: Problems Error on Build & Flash The ESP32 under Eclipse IDE

apriliadina, 2018-06-29:
Hi, 

I'm trying to build and flash the example project of hello world and many others in ESP32.
I'm using the Windows 8.1, the latest update of msys2 and eclipse IDE cpp Photon but when I tried to build and flash the program in Eclipse I keep getting this kind of error messages. But strangely, I can make flash from the msys32 built in command window and Arduino IDE.
I've followed all the step written in https://esp-idf.readthedocs.io/en/latest/get-started/eclipse-setup-windows.html#eclipse-windows-setup 

Here is the full trace that written in the console program:

13:23:29 **** Build of configuration Default for project hello_world_example_eclipse ****
python C:/pci_internship/msys32/home/lennovo/esp/esp-idf/tools/windows/eclipse_make.py flash 
Running make in 'C:/pci_internship/msys32/home/lennovo/esp/hello_world'
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/bootloader/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/bootloader_support/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/coap/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esp32/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/nghttp/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/partition_table/Makefile.projbuild...
including C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/ulp/Makefile.projbuild...
MAKEFLAGS="" CC=cc LD=ld \
make -C C:/pci_internship/msys32/home/lennovo/esp/esp-idf/tools/kconfig
make[1]: Entering directory 'C:/pci_internship/msys32/home/lennovo/esp/esp-idf/tools/kconfig'
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD -lncursesw -lintl
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/util.o lxdialog/util.c
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncursesw -lintl
cc  -IC:/pci_internship/msys32/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o conf.o conf.c
cc -o conf conf.o  zconf.tab.o -lncursesw -lintl
make[1]: Leaving directory 'C:/pci_internship/msys32/home/lennovo/esp/esp-idf/tools/kconfig'
mkdir -p /home/lennovo/esp/hello_world/build/include/config
cd /home/lennovo/esp/hello_world/build; KCONFIG_AUTOHEADER=/home/lennovo/esp/hello_world/build/include/sdkconfig.h COMPONENT_KCONFIGS="C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/app_trace/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/aws_iot/Kconfig   C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/bt/Kconfig    C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/driver/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esp32/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esp_adc_cal/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esp_http_client/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/ethernet/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/fatfs/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/freertos/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/heap/Kconfig    C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/libsodium/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/log/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/lwip/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/mbedtls/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/mdns/Kconfig     C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/openssl/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/pthread/Kconfig    C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/spi_flash/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/spiffs/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/tcpip_adapter/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/vfs/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/wear_levelling/Kconfig  " KCONFIG_CONFIG=C:/pci_internship/msys32/home/lennovo/esp/hello_world/sdkconfig COMPONENT_KCONFIGS_PROJBUILD="   C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/bootloader/Kconfig.projbuild           C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esptool_py/Kconfig.projbuild                    C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/partition_table/Kconfig.projbuild            " C:/pci_internship/msys32/home/lennovo/esp/esp-idf/tools/kconfig/conf.exe --olddefconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/Kconfig
#
# configuration written to C:/pci_internship/msys32/home/lennovo/esp/hello_world/sdkconfig
#
mkdir -p /home/lennovo/esp/hello_world/build/include/config
cd /home/lennovo/esp/hello_world/build; KCONFIG_AUTOHEADER=/home/lennovo/esp/hello_world/build/include/sdkconfig.h COMPONENT_KCONFIGS="C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/app_trace/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/aws_iot/Kconfig   C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/bt/Kconfig    C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/driver/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esp32/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esp_adc_cal/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esp_http_client/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/ethernet/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/fatfs/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/freertos/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/heap/Kconfig    C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/libsodium/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/log/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/lwip/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/mbedtls/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/mdns/Kconfig     C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/openssl/Kconfig  C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/pthread/Kconfig    C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/spi_flash/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/spiffs/Kconfig C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/tcpip_adapter/Kconfig  C:/pci_internship/msys32/home/len [... cut ...]
python C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode "dio" --flash_freq "40m" --flash_size "2MB"  -o /home/lennovo/esp/hello_world/build/bootloader/bootloader.bin C:/pci_internship/msys32/home/lennovo/esp/hello_world/build/bootloader/bootloader.elf
esptool.py v2.3.1
make[1]: Leaving directory 'C:/pci_internship/msys32/home/lennovo/esp/esp-idf/components/bootloader/subproject'

Anyone has the same problem like me? I really need your help to fix this, thank you.

closed 2018-07-03 by apriliadina

projectgus, 2018-07-03:
Hi @apriliadina ,

Sorry noone got back to you earlier. Are you closing the issue because you solved the problem? If so, could you please leave any tip for someone who finds this issue in the future?

Thanks,

Angus

apriliadina, 2018-07-03:
Hi, Angus

Yeah, It's okay. I'm not really solved the problems, and I've tried to do
anything with my Eclipse IDE, toolchain, and even the python since last
weeks. So, because of my lack time, and I really need to use my ESP32 in
Eclipse IDE for my project, I decided to reinstall my windows, and upgrade
it to windows 10. Suprisingly it is works. But, I still don't know the
right way to solve that problem.

Thanks,

Dina

asmitajha, 2019-03-18:
Hi
I was able to build and flash successfully using eclipse IDE. But the problem is that when I checked with blinking LED example, the LED was not blinking on the board. I also tried by changing its GPIO in menuconfig. But when i flashed using msys command prompt using same examples and menuconfig the led was blinking on the board. So i want to ask that why is it not blinking using when flashed with eclipse?﻿