espressif/esp-idf #2130: time(NULL) function returns bad value (2147s) after startup

zodianet, 2018-06-29:
2147s (=35mn) means in fact  0x7FFFFFFF in term of uS (as 64 bits  result of esp_timer_impl_get_time() in esp_timer_esp32.c)
in fact, esp_timer_impl_get_time() seems to return good value but the caller gets bad value. 
if esp_timer_impl_get_time() and esp_timer_get_time() report the result value in a global variable, and the callers read this global variable, all is ok. 
No explanation :-(
set 's_time_base_us = 2100000000LL;'  after startup  to avoid to wait 35mn to observe ...

igrr, 2018-06-29:
Hi @zodianet, could you please mention the commit ID of IDF?

I did a quick test, setting s_time_base_us after startup per your advice, and got what looks like expected result:

```
...
     2144            860        2144001521              7fcaddf1
     2145            861        2145001521              7fda2031
     2146            862        2146001521              7fe96271
     2147            863        2147001521              7ff8a4b1
     2148            864        2148001521              8007e6f1
     2149            865        2149001521              80172931
     2150            866        2150001521              80266b71
     2151            867        2151001520              8035adb0
     2152            868        2152001520              8044eff0
     2153            869        2153001520              80543230
     2154            86a        2154001520              80637470
     2155            86b        2155001520              8072b6b0
     2156            86c        2156001520              8081f8f0
     2157            86d        2157001520              80913b30
```

with the following code (s_time_base_us was made non-static in esp_timer_esp32.c):
```c++
extern uint64_t s_time_base_us;
void app_main()
{
    s_time_base_us = 2100000000LL;
    while (true) {
    	time_t now = time(NULL);
    	int64_t timer = esp_timer_get_time();
    	printf("%9ld\t%8lx\t%lld\t%16llx\n", now, now, timer, (uint64_t) timer);
    	vTaskDelay(1000/portTICK_PERIOD_MS);
    }
}
```

Can you give any other information which can help reproduce this? Do you have any other modifications to IDF? Can you attach your sdkconfig file?

zodianet, 2018-06-29:
Yout test is good to understand:  Some functions prototypes lack. 
Functions with 64 bits returns are considered as regular 32 bits return and in execution,  LSB 32 bits are extended to 64 bits (and 32 bits MSB ignored). 
thus,  I added :
uint64_t IRAM_ATTR esp_timer_impl_get_time() ;
int64_t IRAM_ATTR esp_timer_get_time() ;
at the beginning of time.c and esp_timer.c and 
now all is ok. 
these prototypes are apparently not found in the header .h 
remark:  "#pragma once" in esp_timer.h cannot be avoided  ?

igrr, 2018-07-01:
I've checked that commenting out `esp_timer_get_time` prototype in esp_timer.h leads to the following compilation error in time.c:

```
C:/Users/ivan/Documents/Espressif/dev/esp-idf/components/newlib/time.c: In function 'esp_set_time_from_rtc':
C:/Users/ivan/Documents/Espressif/dev/esp-idf/components/newlib/time.c:250:49: error: implicit declaration of function 'esp_timer_get_time' [-Werror=implicit-function-declaration]
     s_microseconds_offset = get_rtc_time_us() - esp_timer_get_time();
                                                 ^
```

So there should have been an error if the prototype was missing.

Could you please add the following information?

1. Attach full output of `make V=1` (as a file or using gist.github.com). Do `make clean` first to make sure you have a full rebuild.

2. Mention version of ESP-IDF which you are using. You can check it using `git describe --tags`.

FayeY, 2018-07-06:
Hi @zodianet,

Is there any update? Is this issue solved now?
Or could you provide above information for debugging?

Thanks.

zodianet, 2018-07-09:
Hi,
this issue is solved. 
'git describe --tags' returns:  v3.2-dev-57-ge75628b4.
The first compilation did not pass and I quickly made unclear things (in timer and mdns areas) to get a compilation ok. I am therefore responsible. Next time, I'll stop and contact you instead of doing weird things that waste your time later.
Thanks for your help.

closed 2018-07-12 by FayeY