espressif/esp-idf #2133: Error using non-output pin for SCL in slave mode

antonio-fiol, 2018-06-30:
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Core (if using chip or module): ESP-WROOM32
- IDF version b1fa1ba6
- Development Env:		Make
- Operating System:		Linux on raspberry pi
- Power Supply:         USB

## Problem Description
Commit 1ad68de (fixing #393) produces an error on the console when using a non-output pin in slave mode for SCL.
```
E (270) gpio: gpio_set_level(189): GPIO output gpio_num error
```
It seems that the message may not be critical, but it's really confusing.

### Expected Behavior
Able to use pin 34 for SCL in slave mode.

### Actual Behavior
The error stated in the description appears on the log.

### Steps to repropduce
Modify the i2c sample to use pin 34 on the slave for SCL.

## Suggested fix
Only call `gpio_set_level` if configuring MASTER mode, but not in SLAVE.

stickbreaker, 2018-06-30:
@loboris under what circumstances does slave mode drive SCL low? I am working on a SlaveMode driver for the Arduino branch, I need this functionality to match the Arduino Slave protocol.

Chuck.

antonio-fiol, 2018-06-30:
I agree that using a pin without output capability will stop me from possibly doing clock stretching from the ESP32 slave.
However, the rest of the code seems to accept the slave mode with SCL on a non-output pin as a valid combination.
In my case, I need a lot of output pins, nearly no inputs, and do not need to stretch the clock at all... so this combination would be really helpful.

stickbreaker, 2018-06-30:
@negativekelvin How do I make the ESP32 clock Stretch in Slave Mode? I understand the i2c protocol, I have found no situations under which I can cause the i2c peripheral to clock stretch.  

The Arduino i2c protocol, which was based on the capabilities of the AtMega168 automatically started clock stretching as soon as it ack'd the Slave Id, this paused the i2c Master while application code generated the data to be returned.  The AtMega168 held SCL low until the driver code released it.  During my investigation of the ESP32, there is no delay between the Slave Address Ack and data being set out.  If the TX Fifo is empty the ESP32 sends 0xff.  

If there is any known procedure to cause the i2c peripheral to instigate a SCL stretching operation, I WOULD LOVE to know it!

Chuck.

loboris, 2018-06-30:
The clock will be stretched while processing the i2c interrupt or if the interrupts are disabled.
Look at [my PR](https://github.com/espressif/esp-idf/pull/2096), in this modified driver more processing is done in interrupt routine and clock stretching may be required at higher speeds.
I think it newer happens in the current driver.

xiongyumail, 2018-08-29:
Hi @antonio-fiol 

A standard I2C Slaver requires the ability to output SCL, as well as configuring SCL as open-drain output in Master mode. If you don't need this capability, you can modify the I2C code. But it's impractical to make changes on the master branch.

antonio-fiol, 2018-08-29:
@xiongyumail I'm just saying:

- this is a **regression** introduced in commit b1fa1ba
- it won't stop anyone from using SCL on an input pin, but it's very confusing
- it is trivial to fix.

**A regression**
Besides the fact that the version before that commit does not give you the error, in the same function, you can read that it checks that SDA is a valid OUTPUT GPIO:
```
    I2C_CHECK(((GPIO_IS_VALID_OUTPUT_GPIO(sda_io_num))), I2C_SDA_IO_ERR_STR, ESP_ERR_INVALID_ARG);
```
But then it checks that SCL is either a valid OUTPUT GPIO or just a valid GPIO if we're requesting SLAVE mode.
```
    I2C_CHECK((GPIO_IS_VALID_OUTPUT_GPIO(scl_io_num)) ||
              (GPIO_IS_VALID_GPIO(scl_io_num) && mode == I2C_MODE_SLAVE),
              I2C_SCL_IO_ERR_STR,
              ESP_ERR_INVALID_ARG);
```

IMO, this proves that the original intention was to allow (restricted, maybe) SCL operation on an input-only pin.

Commit b1fa1ba breaks that scenario: the `I2C_CHECK` lines allow it, but then it outputs an error message.

**Confusing**
The message is at ERROR level, internally on `gpio_set_level`, and without any information to help you understand why. Oh, and actually it does not break anything that worked before.

**Trivial fix**
It is trivial to fix this issue, without reverting b1fa1ba and without causing the glitches described in the bug it fixes.
The fix is one line:
```
if( GPIO_IS_VALID_OUTPUT_GPIO(scl_io_num) ) gpio_set_level(scl_io_num, I2C_IO_INIT_LEVEL);
```
instead of
```
gpio_set_level(scl_io_num, I2C_IO_INIT_LEVEL);
```
The glitches can't appear simply because an input pin can not pull that line down.

igrr, 2022-03-14:
This issue was recently fixed via https://github.com/espressif/esp-idf/pull/8312 (thanks @jkearins!)

closed 2022-03-14 by igrr